espressif/vscode-esp-idf-extension #968: VScode 安装Python虚拟环境不成功

murarduino, 2023-05-11:
### OS

Windows

### Operating System version

Windows10

### Visual Studio Code version

1.78.1

### ESP-IDF version

4.4.4

### Python version

3.8.7 （idf4.4_py3.8_env）

### Doctor command output

重装电脑，安装VScode，安装ESP-IDF-4.4.4离线版本。ESP-IDF安装完成后，可以正确运行idf.py
多次安装ESP-IDF-extension，都卡在最后 的Python虚拟环境的配置上。

### Description

重装电脑，安装VScode，安装ESP-IDF-4.4.4离线版本。ESP-IDF安装完成后，可以正确运行idf.py
多次安装ESP-IDF-extension，都卡在最后 的Python虚拟环境的配置上。
科学上网，Chrome可以正常打开https://pypi.org/simple/，为防万了一，国内源
腾讯 http://mirrors.tencentyun.com/pypi/simple
阿里 https://mirrors.aliyun.com/pypi/simple
豆瓣 https://pypi.douban.com/simple
中科大 https://pypi.mirrors.ustc.edu.cn/simple/
清华 https://pypi.tuna.tsinghua.edu.cn/simple
都试了一遍，都卡在相同的There was a problem confirming the ssl certificate问题上

### Debug Message

```plain
idf_versions.txt has (1) KB
idf_versions.txt progress: 100.00% (0.08 / 0.08) KB
Installing ESP-IDF extension python packages in D:/Espressif/python_env/idf4.4_py3.8_env/Scripts/python.exe ...

Could not fetch URL https://pypi.org/simple/gcovr/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='pypi.org', port=443): Max retries exceeded with url: /simple/gcovr/ (Caused by SSLError(SSLEOFError(8, 'EOF occurred in violation of protocol (_ssl.c:1125)'))) - skipping
Could not fetch URL https://pypi.org/simple/pip/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='pypi.org', port=443): Max retries exceeded with url: /simple/pip/ (Caused by SSLError(SSLEOFError(8, 'EOF occurred in violation of protocol (_ssl.c:1125)'))) - skipping
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLEOFError(8, 'EOF occurred in violation of protocol (_ssl.c:1125)'))': /simple/gcovr/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLEOFError(8, 'EOF occurred in violation of protocol (_ssl.c:1125)'))': /simple/gcovr/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLEOFError(8, 'EOF occurred in violation of protocol (_ssl.c:1125)'))': /simple/gcovr/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLEOFError(8, 'EOF occurred in violation of protocol (_ssl.c:1125)'))': /simple/gcovr/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLEOFError(8, 'EOF occurred in violation of protocol (_ssl.c:1125)'))': /simple/gcovr/
ERROR: Could not find a version that satisfies the requirement gcovr
ERROR: No matching distribution found for gcovr
Command failed: "D:/Espressif/python_env/idf4.4_py3.8_env/Scripts/python.exe" -m pip install --upgrade --constraint "c:\Users\mura\.vscode\extensions\espressif.esp-idf-extension-1.6.2\espidf.constraints.txt" --no-warn-script-location  -r "c:\Users\mura\.vscode\extensions\espressif.esp-idf-extension-1.6.2\requirements.txt"
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLEOFError(8, 'EOF occurred in violation of protocol (_ssl.c:1125)'))': /simple/gcovr/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLEOFError(8, 'EOF occurred in violation of protocol (_ssl.c:1125)'))': /simple/gcovr/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLEOFError(8, 'EOF occurred in violation of protocol (_ssl.c:1125)'))': /simple/gcovr/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLEOFError(8, 'EOF occurred in violation of protocol (_ssl.c:1125)'))': /simple/gcovr/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLEOFError(8, 'EOF occurred in violation of protocol (_ssl.c:1125)'))': /simple/gcovr/
ERROR: Could not find a version that satisfies the requirement gcovr
ERROR: No matching distribution found for gcovr
idf_versions.txt has (1) KB
idf_versions.txt progress: 100.00% (0.08 / 0.08) KB
Using existing D:\Espressif\dist\idf-git-2.30.1-win64.zip
Installing D:\Espressif\tools\idf-git\2.30.1 ...
Warning File D:\Espressif\tools\idf-git\2.30.1\cmd\git.exe
                                        already exists and was not updated.
Warning File D:\Espressif\tools\idf-git\2.30.1\dev\_._
                                        already exists and was not updated.
Warning File D:\Espressif\tools\idf-git\2.30.1\etc\bash.bash_logout
[... 134 lines cut ...]
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLEOFError(8, 'EOF occurred in violation of protocol (_ssl.c:1125)'))': /simple/gcovr/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLEOFError(8, 'EOF occurred in violation of protocol (_ssl.c:1125)'))': /simple/gcovr/
ERROR: Could not find a version that satisfies the requirement gcovr
ERROR: No matching distribution found for gcovr
Command failed: "D:\Espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe" -m pip install --upgrade --constraint "c:\Users\mura\.vscode\extensions\espressif.esp-idf-extension-1.6.2\espidf.constraints.txt" --no-warn-script-location  -r "c:\Users\mura\.vscode\extensions\espressif.esp-idf-extension-1.6.2\requirements.txt"
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLEOFError(8, 'EOF occurred in violation of protocol (_ssl.c:1125)'))': /simple/gcovr/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLEOFError(8, 'EOF occurred in violation of protocol (_ssl.c:1125)'))': /simple/gcovr/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLEOFError(8, 'EOF occurred in violation of protocol (_ssl.c:1125)'))': /simple/gcovr/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLEOFError(8, 'EOF occurred in violation of protocol (_ssl.c:1125)'))': /simple/gcovr/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLEOFError(8, 'EOF occurred in violation of protocol (_ssl.c:1125)'))': /simple/gcovr/
ERROR: Could not find a version that satisfies the requirement gcovr
ERROR: No matching distribution found for gcovr
Command failed: "D:\Espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe" -m pip install --upgrade --constraint "c:\Users\mura\.vscode\extensions\espressif.esp-idf-extension-1.6.2\espidf.constraints.txt" --no-warn-script-location  -r "c:\Users\mura\.vscode\extensions\espressif.esp-idf-extension-1.6.2\requirements.txt"
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLEOFError(8, 'EOF occurred in violation of protocol (_ssl.c:1125)'))': /simple/gcovr/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLEOFError(8, 'EOF occurred in violation of protocol (_ssl.c:1125)'))': /simple/gcovr/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLEOFError(8, 'EOF occurred in violation of protocol (_ssl.c:1125)'))': /simple/gcovr/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLEOFError(8, 'EOF occurred in violation of protocol (_ssl.c:1125)'))': /simple/gcovr/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLEOFError(8, 'EOF occurred in violation of protocol (_ssl.c:1125)'))': /simple/gcovr/
ERROR: Could not find a version that satisfies the requirement gcovr
ERROR: No matching distribution found for gcovr
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

closed 2023-05-12 by murarduino

hayschan, 2023-06-19:
我是一个相似可以不一样的的问题。

```python
Installing ESP-IDF extension python packages in d:\.espressif\python_env\idf5.0_py3.8_env\Scripts\python.exe ...

Looking in indexes: https://pypi.mirrors.ustc.edu.cn/simple/
--- Logging error ---
Traceback (most recent call last):
  File "d:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages\pip\_internal\resolution\resolvelib\factory.py", line 158, in _make_candidate_from_dist
    base = self._installed_candidate_cache[dist.canonical_name]
KeyError: 'gcovr'

During handling of the above exception, another exception occurred:
```

## 全部 LOG

```python
Python 3.8.7
pip 23.1.2 from d:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages\pip (python 3.8)

Looking in indexes: https://pypi.mirrors.ustc.edu.cn/simple/
Requirement already satisfied: pip in d:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (23.1.2)
Requirement already satisfied: setuptools in d:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (67.8.0)
Looking in indexes: https://pypi.mirrors.ustc.edu.cn/simple/, https://dl.espressif.com/pypi
Requirement already satisfied: setuptools in d:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from -r d:\esp\esp-idf\tools\requirements\requirements.core.txt (line 3)) (67.8.0)
Requirement already satisfied: click in d:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from -r d:\esp\esp-idf\tools\requirements\requirements.core.txt (line 4)) (8.0.1)
Collecting click (from -r d:\esp\esp-idf\tools\requirements\requirements.core.txt (line 4))
  Using cached https://dl.espressif.cn/pypi/click/click-8.0.4-py3-none-any.whl (97 kB)
Requirement already satisfied: pyserial in d:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from -r d:\esp\esp-idf\tools\requirements\requirements.core.txt (line 5)) (3.5)
Collecting future (from -r d:\esp\esp-idf\tools\requirements\requirements.core.txt (line 6))
  Using cached https://dl.espressif.cn/pypi/future/future-0.18.2-py3-none-any.whl (491 kB)
Requirement already satisfied: cryptography in d:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from -r d:\esp\esp-idf\tools\requirements\requirements.core.txt (line 7)) (36.0.2)
Requirement already satisfied: pyparsing in d:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from -r d:\esp\esp-idf\tools\requirements\requirements.core.txt (line 8)) (3.0.9)
Requirement already satisfied: pyelftools in d:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from -r d:\esp\esp-idf\tools\requirements\requirements.core.txt (line 9)) (0.27)
Requirement already satisfied: idf-component-manager in d:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from -r d:\esp\esp-idf\tools\requirements\requirements.core.txt (line 10)) (1.2.3)
Requirement already satisfied: esp-coredump in d:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from -r d:\esp\esp-idf\tools\requirements\requirements.core.txt (line 11)) (1.5.0)
Requirement already satisfied: esptool in d:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from -r d:\esp\esp-idf\tools\requirements\requirements.core.txt (line 12)) (4.5.1)
Requirement already satisfied: kconfiglib in d:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from -r d:\esp\esp-idf\tools\requirements\requirements.core.txt (line 15)) (14.1.0)
Requirement already satisfied: windows-curses in d:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from -r d:\esp\esp-idf\tools\requirements\requirements.core.txt (line 16)) (2.3.1)
Requirement already satisfied: freertos_gdb in d:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from -r d:\esp\esp-idf\tools\requirements\requirements.core.txt (line 19)) (1.0.2)
Requirement already satisfied: colorama in d:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from click->-r d:\esp\esp-idf\tools\requirements\requirements.core.txt (line 4)) (0.4.6)
Requirement already satisfied: cffi>=1.12 in d:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from cryptography->-r d:\esp\esp-idf\tools\requirements\requirements.core.txt (line 7)) (1.15.1)
Requirement already satisfied: packaging in d:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from idf-component-manager->-r d:\esp\esp-idf\tools\requirements\requirements.core.txt (line 10)) (23.1)
Requirement already satisfied: requests<3 in d:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from idf-component-manager->-r d:\esp\esp-idf\tools\requirements\requirements.core.txt (line 10)) (2.31.0)
Requirement already satisfied: urllib3<2 in d:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from idf-component-manager->-r d:\esp\esp-idf\tools\requirements\requirements.core.txt (line 10)) (1.26.16)
Requirement already satisfied: requests-file in d:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from idf-component-manager->-r d:\esp\esp-idf\tools\requirements\requirements.core.txt (line 10)) (1.5.1)
Requirement already satisfied: requests-toolbelt in d:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from idf-component-manager->-r d:\esp\esp-idf\tools\requirements\requirements.core.txt (line 10)) (1.0.0)
Requirement already satisfied: schema in d:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from idf-component-manager->-r d:\esp\esp-idf\tools\requirements\requirements.core.txt (line 10)) (0.7.5)
[... cut ...]
  File "d:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages\pip\_internal\utils\logging.py", line 179, in emit
    self.handleError(record)
Message: 'Requirement %s: %s (%s)'
Arguments: ('already satisfied', <InstallRequirement object: MarkupSafe>=2.0 in d:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from jinja2->gcovr->-r c:\Users\\u9648\u5e0c\.vscode\extensions\espressif.esp-idf-extension-1.6.3\requirements.txt (line 1)) editable=False>, <Version('2.0.1')>)
Error during ESP-IDF setup

```

murarduino, 2023-06-19:
> 

建议改用4.4.4，我记得5.X的版本不稳定。

murarduino, 2023-06-19:
建议改用4.4.4版本，我记得好像5.0.X版本还不太稳定。

------------------&nbsp;原始邮件&nbsp;------------------
发件人:                                                                                                                        "espressif/vscode-esp-idf-extension"                                                                                    ***@***.***&gt;;
发送时间:&nbsp;2023年6月19日(星期一) 下午5:09
***@***.***&gt;;
***@***.***&gt;;"State ***@***.***&gt;;
主题:&nbsp;Re: [espressif/vscode-esp-idf-extension] VScode 安装Python虚拟环境不成功 (VSC-1107) (Issue #968)

 
我是一个相似可以不一样的的问题。
 Installing ESP-IDF extension python packages in d:\.espressif\python_env\idf5.0_py3.8_env\Scripts\python.exe ... Looking in indexes: https://pypi.mirrors.ustc.edu.cn/simple/ --- Logging error --- Traceback (most recent call last):   File "d:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages\pip\_internal\resolution\resolvelib\factory.py", line 158, in _make_candidate_from_dist     base = self._installed_candidate_cache[dist.canonical_name] KeyError: 'gcovr' During handling of the above exception, another exception occurred:
 
全部 LOG
 Python 3.8.7 pip 23.1.2 from d:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages\pip (python 3.8) Looking in indexes: https://pypi.mirrors.ustc.edu.cn/simple/ Requirement already satisfied: pip in d:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (23.1.2) Requirement already satisfied: setuptools in d:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (67.8.0) Looking in indexes: https://pypi.mirrors.ustc.edu.cn/simple/, https://dl.espressif.com/pypi Requirement already satisfied: setuptools in d:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from -r d:\esp\esp-idf\tools\requirements\requirements.core.txt (line 3)) (67.8.0) Requirement already satisfied: click in d:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from -r d:\esp\esp-idf\tools\requirements\requirements.core.txt (line 4)) (8.0.1) Collecting click (from -r d:\esp\esp-idf\tools\requirements\requirements.core.txt (line 4))   Using cached https://dl.espressif.cn/pypi/click/click-8.0.4-py3-none-any.whl (97 kB) Requirement already satisfied: pyserial in d:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from -r d:\esp\esp-idf\tools\requirements\requirements.core.txt (line 5)) (3.5) Collecting future (from -r d:\esp\esp-idf\tools\requirements\requirements.core.txt (line 6))   Using cached https://dl.espressif.cn/pypi/future/future-0.18.2-py3-none-any.whl (491 kB) Requirement already satisfied: cryptography in d:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from -r d:\esp\esp-idf\tools\requirements\requirements.core.txt (line 7)) (36.0.2) Requirement already satisfied: pyparsing in d:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from -r d:\esp\esp-idf\tools\requirements\requirements.core.txt (line 8)) (3.0.9) Requirement already satisfied: pyelftools in d:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from -r d:\esp\esp-idf\tools\requirements\requirements.core.txt (line 9)) (0.27) Requirement already satisfied: idf-component-manager in d:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from -r d:\esp\esp-idf\tools\requirements\requirements.core.txt (line 10)) (1.2.3) Requirement already satisfied: esp-coredump in d:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from -r d:\esp\esp-idf\tools\requirements\requirements.core.txt (line 11)) (1.5.0) Requirement already satisfied: esptool in d:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from -r d:\esp\esp-idf\tools\requirements\requirements.core.txt (line 12)) (4.5.1) Requirement already satisfied: kconfiglib in d:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from -r d:\esp\esp-idf\tools\requirements\requirements.core.txt (line 15)) (14.1.0) Requirement already satisfied: windows-curses in d:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from -r d:\esp\esp-idf\tools\requirements\requirements.core.txt (line 16)) (2.3.1) Requirement already satisfied: freertos_gdb in d:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from -r d:\esp\esp-idf\tools\requirements\requirements.core.txt (line 19)) (1.0.2) Requirement already satisfied: colorama in d:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from click-&gt;-r d:\esp\esp-idf\tools\requirements\requirements.core.txt (line 4)) (0.4.6) Requirement already satisfied: cffi&gt;=1.12 in d:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from cryptography-&gt;-r d:\esp\esp-idf\tools\requirements\requirements.core.txt (line 7)) (1.15.1) Requirement already satisfied: packaging in d:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from idf-component-manager-&gt;-r d:\esp\esp-idf\tools\requirements\requirements.core.txt (line 10)) (23.1) Requirement already satisfied: requests<3 in d:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from idf-component-manager-&gt;-r d:\esp\esp-idf\tools\requirements\requirements.core.txt (line 10)) (2.31.0) Requirement already satisfied: urllib3<2 in d:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from idf-component-manager-&gt;-r d:\esp\esp-idf\tools\requirements\requirements.core.txt (line 10)) (1.26.16) Requirement already satisfied: requests-file in d:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages (from idf-component-manager-&gt;-r d:\esp\esp-idf\tools\requirements\requirements.core.txt (line 10)) (1.5.1) Requirement alr [... cut ...] s.txt (line 2)) editable=False&gt;, <Version('2.31.0')&gt;) --- Logging error --- Traceback (most recent call last):   File "d:\.espressif\python_env\idf5.0_py3.8_env\lib\site-packages\pip\_internal\resolution\resolvelib\factory.py", line 158, in _make_candidate_from_dist     base = self._installed_candidate_cache[dist.canonical_name] KeyError: 'urllib3' During handling of the above exception, another exception occurred: Traceback (most recent call last):   File "d:\.espressif\python_env\idf5.0_

hayschan, 2023-06-19:
解决了，竟然是和 Windows 的非 Unicode 设定有关。

把 Windows 的非 Unicode 语言设置成中文就没问题了。原因应该是我的用户路径是中文的，造成了一些工具安装的时候无法识别。

[image]