espressif/vscode-esp-idf-extension #969: ESP-IDF Setup fails when using existing ESP-IDF setup

Kilill, 2023-05-11:
### OS

Linux

### Operating System version

Ubuntu 22.04.2 LTS

### Visual Studio Code version

Version: 1.77.3 (system setup) Commit: 704ed70d4fd1c6bd6342c436f1ede30d1cff4710 Date: 2023-04-12T09:16:02.548Z Electron: 19.1.11 Chromium: 102.0.5005.196 Node.js: 16.14.2 V8: 10.2.154.26-electron.0 OS: Windows_NT x64 10.0.19045 Sandboxed: No

### ESP-IDF version

ESP-IDF v5.2-dev-503-g17451f1fb3-dirty

### Python version

3.10.6

### Doctor command output

The what command ?

### Description

- vscode on windows remotly connected to Linux system
- Extension installed on linux system
- ESP-IDF installed for multiuser global access on Linux system
- ESP-IDF directories are not writeable by regular users
- Trying the **ESP-IDF Setup** and selecting **USE EXISTING SETUP**

setup correctly identifies the ESP-IDF directories but:

1. Incorrectly reports that it is downloading the Xtensa tools , then reports that is has found then already installed. Should say that it is _Checking for ...._ 
2. Tries to install ESP ROM ELFS but that ofc fails due to permissions and setup fails. 

In this case (existing installation) it is not a catastrophic error and should not result in an setup failure.

And it should definitely **not** try to erase the complete directory!

even if this is bypassed, it still tries to install python modules into the global IDF directory, even though the modules requested are already globally installed.

### Debug Message

```plain
/usr/bin/git

Downloading GDB for Xtensa
Found xtensa-esp-elf-gdb in /opt/esp32/xtensa_tools/dist
Downloading GDB for RISC-V
Found riscv32-esp-elf-gdb in /opt/esp32/xtensa_tools/dist
Downloading Toolchain for Xtensa (ESP32) based on GCC
Found xtensa-esp32-elf in /opt/esp32/xtensa_tools/dist
Downloading Toolchain for Xtensa (ESP32-S2) based on GCC
Found xtensa-esp32s2-elf in /opt/esp32/xtensa_tools/dist
Downloading Toolchain for Xtensa (ESP32-S3) based on GCC
Found xtensa-esp32s3-elf in /opt/esp32/xtensa_tools/dist
Downloading Toolchain for 32-bit RISC-V based on GCC
Found riscv32-esp-elf in /opt/esp32/xtensa_tools/dist
Downloading Toolchain for ESP32 ULP coprocessor
Found esp32ulp-elf in /opt/esp32/xtensa_tools/dist
Downloading OpenOCD for ESP32
Found openocd-esp32 in /opt/esp32/xtensa_tools/dist
Downloading ESP ROM ELFs
Found esp-rom-elfs in /opt/esp32/xtensa_tools/dist

Using existing GDB for Xtensa in /opt/esp32/xtensa_tools/tools/xtensa-esp-elf-gdb/12.1_20221002
Using existing GDB for RISC-V in /opt/esp32/xtensa_tools/tools/riscv32-esp-elf-gdb/12.1_20221002
Using existing Toolchain for Xtensa (ESP32) based on GCC in /opt/esp32/xtensa_tools/tools/xtensa-esp32-elf/esp-12.2.0_20230208
[... 71 lines cut ...]
Requirement already satisfied: requests>=2.21.0 in /opt/esp32/xtensa_tools/python_env/idf5.2_py3.10_env/lib/python3.10/site-packages (from -r /home/kill/.vscode-server/extensions/espressif.esp-idf-extension-1.6.2/esp_debug_adapter/requirements.txt (line 7)) (2.30.0)
Collecting xmlrunner>=1.7.7 (from -r /home/kill/.vscode-server/extensions/espressif.esp-idf-extension-1.6.2/esp_debug_adapter/requirements.txt (line 9))
  Using cached xmlrunner-1.7.7-py3-none-any.whl
Requirement already satisfied: charset-normalizer<4,>=2 in /opt/esp32/xtensa_tools/python_env/idf5.2_py3.10_env/lib/python3.10/site-packages (from requests>=2.21.0->-r /home/kill/.vscode-server/extensions/espressif.esp-idf-extension-1.6.2/esp_debug_adapter/requirements.txt (line 7)) (3.1.0)
Requirement already satisfied: idna<4,>=2.5 in /opt/esp32/xtensa_tools/python_env/idf5.2_py3.10_env/lib/python3.10/site-packages (from requests>=2.21.0->-r /home/kill/.vscode-server/extensions/espressif.esp-idf-extension-1.6.2/esp_debug_adapter/requirements.txt (line 7)) (3.4)
Requirement already satisfied: urllib3<3,>=1.21.1 in /opt/esp32/xtensa_tools/python_env/idf5.2_py3.10_env/lib/python3.10/site-packages (from requests>=2.21.0->-r /home/kill/.vscode-server/extensions/espressif.esp-idf-extension-1.6.2/esp_debug_adapter/requirements.txt (line 7)) (1.26.15)
Requirement already satisfied: certifi>=2017.4.17 in /opt/esp32/xtensa_tools/python_env/idf5.2_py3.10_env/lib/python3.10/site-packages (from requests>=2.21.0->-r /home/kill/.vscode-server/extensions/espressif.esp-idf-extension-1.6.2/esp_debug_adapter/requirements.txt (line 7)) (2023.5.7)
Installing collected packages: xmlrunner, psutil, click
ERROR: Could not install packages due to an OSError: [Errno 13] Permission denied: '/opt/esp32/xtensa_tools/python_env/idf5.2_py3.10_env/lib/python3.10/site-packages/xmlrunner'
Check the permissions.

Command failed: "/opt/esp32/xtensa_tools/python_env/idf5.2_py3.10_env/bin/python" -m pip install --upgrade --constraint "/home/kill/.vscode-server/extensions/espressif.esp-idf-extension-1.6.2/espidf.constraints.txt" --no-warn-script-location -r "/home/kill/.vscode-server/extensions/espressif.esp-idf-extension-1.6.2/esp_debug_adapter/requirements.txt"
ERROR: Could not install packages due to an OSError: [Errno 13] Permission denied: '/opt/esp32/xtensa_tools/python_env/idf5.2_py3.10_env/lib/python3.10/site-packages/xmlrunner'
Check the permissions.

Command failed: "/opt/esp32/xtensa_tools/python_env/idf5.2_py3.10_env/bin/python" -m pip install --upgrade --constraint "/home/kill/.vscode-server/extensions/espressif.esp-idf-extension-1.6.2/espidf.constraints.txt" --no-warn-script-location -r "/home/kill/.vscode-server/extensions/espressif.esp-idf-extension-1.6.2/esp_debug_adapter/requirements.txt"
ERROR: Could not install packages due to an OSError: [Errno 13] Permission denied: '/opt/esp32/xtensa_tools/python_env/idf5.2_py3.10_env/lib/python3.10/site-packages/xmlrunner'
Check the permissions.

````
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2023-05-15:
Some permission issues in your system ? Can you chmod this directory ?

Command failed: "/opt/esp32/xtensa_tools/python_env/idf5.2_py3.10_env/bin/python" -m pip install --upgrade --constraint "/home/kill/.vscode-server/extensions/espressif.esp-idf-extension-1.6.2/espidf.constraints.txt" --no-warn-script-location -r "/home/kill/.vscode-server/extensions/espressif.esp-idf-extension-1.6.2/esp_debug_adapter/requirements.txt"
ERROR: Could not install packages due to an OSError: [Errno 13] Permission denied: '/opt/esp32/xtensa_tools/python_env/idf5.2_py3.10_env/lib/python3.10/site-packages/xmlrunner'
Check the permissions.

brianignacio5, 2023-05-15:
Yeah but these are python packages used by the debug adapter of the extension. They should exist on the virtual env of ESP-IDF. Not sure how to proceed here, packages are still required by the extension to be installed and this is done using `pip install ...` with constraints.

Kilill, 2023-05-15:
I thought that it would be installed into the .vscode (or .vscode-server) / extensions/xyz directory ? but then i'm totally unfamiliar on how the extension system on vscode works ..

closed 2023-06-05 by github-actions[bot]