espressif/vscode-esp-idf-extension #970: Default baud rate is 460800 in monitor in extension v1.6.2

rtheil-growlink, 2023-05-12:
### OS

Windows

### Operating System version

Windows 10

### Visual Studio Code version

Latest

### ESP-IDF version

4.3.5

### Python version

3.8.7

### Doctor command output

N/A

### Description

Monitor no longer works because baud rate is set as 406800 instead of the necessary 115200

### Debug Message

```plain
See image
```

### Other Steps to Reproduce

[image]

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

phfbertoleti, 2023-05-13:
For reference, I'm experiencing the same error here.

dpslwk, 2023-05-15:
see #963

brianignacio5, 2023-05-15:
Closing this issue since it is a duplicate of #963

closed 2023-05-15 by brianignacio5