espressif/vscode-esp-idf-extension #978: Can not start debugging ESP32-C3 builtin USB-JTAG

walkiewicz125, 2023-05-27:
### OS

Windows

### Operating System version

Windows 10

### Visual Studio Code version

1.70.1

### ESP-IDF version

5.1

### Python version

3.8.7

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19044 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\Microsoft VS Code\bin;C:\msys64\mingw64\bin;C:\msys64\usr\bin;C:\Program Files\Git\cmd;C:\Program Files\Git\mingw64\bin;C:\Program Files\Git\usr\bin;C:\Program Files\Docker\Docker\resources\bin;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:\msys64\mingw64\bin; 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.70.1 
Visual Studio Code language en 
Visual Studio Code shell C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.6.3 
Workspace folder c:\esp32\projects\blink_test\blink 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) c:\esp32\esp_idf\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) c:\esp32\esp_idf\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin;c:\esp32\esp_idf\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin;c:\esp32\esp_idf\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin;c:\esp32\esp_idf\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin;c:\esp32\esp_idf\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin;c:\esp32\esp_idf\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin;c:\esp32\esp_idf\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;c:\esp32\esp_idf\tools\cmake\3.24.0\bin;c:\esp32\esp_idf\tools\openocd-esp32\v0.12.0-esp32-20230419\openocd-esp32\bin;c:\esp32\esp_idf\tools\ninja\1.10.2;c:\esp32\esp_idf\tools\idf-exe\1.0.3;c:\esp32\esp_idf\tools\ccache\4.8\ccache-4.8-windows-x86_64;c:\esp32\esp_idf\tools\dfu-util\0.11\dfu-util-0.11-win64;c:\esp32\esp_idf\tools\esp-rom-elfs\20230320
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: c:\esp32\esp_idf\tools\openocd-esp32\v0.12.0-esp32-20230419/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
    ESP_ROM_ELF_DIR: c:\esp32\esp_idf\tools\esp-rom-elfs\20230320/
Virtual env Python Path (idf.pythonBinPath) c:\esp32\esp_idf\python_env\idf5.1_py3.8_env\Scripts\python.exe
Serial port (idf.port) COM1
OpenOCD Configs (idf.openOcdConfigs) board/esp32c3-builtin.cfg
ESP-IDF Tools Path (idf.toolsPath) c:\esp32\esp_idf
Git Path (idf.gitPath) c:\esp32\esp_idf\tools\idf-git\2.30.1\cmd\git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to c:\esp32\esp_idf\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin: true
Access to c:\esp32\esp_idf\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin: true
Access to c:\esp32\esp_idf\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin: true
Access to c:\esp32\esp_idf\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin: true
Access to c:\esp32\esp_idf\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin: true
Access to c:\esp32\esp_idf\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin: true
Access to c:\esp32\esp_idf\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin: true
Access to c:\esp32\esp_idf\tools\cmake\3.24.0\bin: true
Access to c:\esp32\esp_idf\tools\openocd-esp32\v0.12.0-esp32-20230419\openocd-esp32\bin: true
Access to c:\esp32\esp_idf\tools\ninja\1.10.2: true
Access to c:\esp32\esp_idf\tools\idf-exe\1.0.3: true
Access to c:\esp32\esp_idf\tools\ccache\4.8\ccache-4.8-windows-x86_64: true
Access to c:\esp32\esp_idf\tools\dfu-util\0.11\dfu-util-0.11-win64: true
Access to c:\esp32\esp_idf\tools\esp-rom-elfs\20230320: true
[... 325 lines cut ...]
[Debug Adapter]
DEBUG_ADAPTER_STOPPED
2023-05-27 14:20:13,203 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STOPPED

[Debug Adapter]
OSError: [Errno 22] Invalid argument

[OpenOCD]
Info : dropped 'gdb'
[OpenOCD]
 connection

[Stopped] : ESP-IDF Debug Adapter
```

### Other Steps to Reproduce

Installed ESP_IDF extension in VS Code
Then F1: Configure ESP-IDF extension. 
Select release/5.1 version
Create example project "blink".

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2023-05-29:
Could you explain what is not working ? You say that setting breakpoints work right ?

Also you can increase the `logLevel` to 4 in launch.json to see more information about the error, but I'm not sure what is it that is not working ?

walkiewicz125, 2023-06-06:
Ok, I am sorry I was unclear. I will try to explain more clearly.

I added logLevel 4 to launch.json.
It looks like this:
```
{
  "version": "0.2.0",
  "configurations": [
    {
      "type": "espidf",
      "name": "Launch",
      "request": "launch",
      "logLevel": 4,
    }
  ]
}
```

### Result:
After launching debugging, it looks like it is trying to start debugging.
After a few seconds. Error pop-ups without message in it. Just two buttons: 
    - Open 'launch.json'
    - Cancel

### Debug Console:
```
Debug adapter -> Extension: DEBUG_ADAPTER_STARTED
Starting. Cmd: c:\Users\proxima\.vscode\extensions\espressif.esp-idf-extension-1.6.3\esp_debug_adapter\debug_adapter_main.py -d 4 -e c:\esp32\projects\blink_test\blink\build\blink.elf -l c:\esp32\projects\blink_test\blink\debug.log -p 43474 -dn esp32c3 -a 0x10000 -t riscv32-esp-elf-
Listening on port 43474
Debug adapter -> Extension: DEBUG_ADAPTER_READY2CONNECT
Got connection
Got json: {
    "arguments": {
        "adapterID": "espidf",
        "clientID": "vscode",
        "clientName": "Visual Studio Code",
        "columnsStartAt1": true,
        "linesStartAt1": true,
        "locale": "pl",
        "pathFormat": "path",
        "supportsArgsCanBeInterpretedByShell": true,
        "supportsInvalidatedEvent": true,
        "supportsMemoryReferences": true,
        "supportsProgressReporting": true,
        "supportsRunInTerminalRequest": true,
        "supportsVariablePaging": true,
        "supportsVariableType": true
    },
    "command": "initialize",
    "seq": 1,
    "type": "request"
}
Created gdb object of type: GdbEsp32c3
Debug Adapter initialized
The operation took too long! - Measured 2.551870 s ( from ~ 19:07:25,106207)
Processed command: initialize
Got json: {
    "arguments": {
        "__configurationTarget": 6,
        "__sessionId": "f83482c4-6e25-44a0-bdd7-1afd6a4fbe62",
        "elfFilePath": "c:\\esp32\\projects\\blink_test\\blink\\build\\blink.elf",
        "logLevel": 4,
        "name": "Launch",
        "request": "launch",
        "type": "espidf"
    },
    "command": "launch",
    "seq": 2,
    "type": "request"
}
[... 652 lines cut ...]
2023-06-06 19:07:40,150 - Debug Adapter (main) - DEBUG - list index out of range
2023-06-06 19:07:40,150 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "list index out of range\n", "source": {}}, "seq": 26}

2023-06-06 19:07:40,150 - Debug Adapter (main) - DEBUG - Closing the Socket
2023-06-06 19:07:40,150 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Closing the Socket\n", "source": {}}, "seq": 27}

2023-06-06 19:07:40,150 - Debug Adapter (main) - DEBUG - Stopping of the Writer thread
2023-06-06 19:07:40,150 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Stopping of the Writer thread\n", "source": {}}, "seq": 28}

2023-06-06 19:07:40,151 - Debug Adapter (main) - DEBUG - Stopping of the Reader thread

[Debug Adapter]
OSError: [Errno 22] Invalid argument

[Debug Adapter]
2023-06-06 19:07:40,151 - Debug Adapter (main) - DEBUG - Adapter is Stopped
DEBUG_ADAPTER_STOPPED
2023-06-06 19:07:40,151 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STOPPED
2023-06-06 19:07:40,151 - Debug Adapter(Command Processor) - DEBUG - Processed command: disconnect

[OpenOCD]
Info : dropped 'gdb' connection

[Stopped] : ESP-IDF Debug Adapter
```

walkiewicz125, 2023-06-06:
Ok, I don't know what happened.
I tried once more time.
First holding Boot button while connecting power. - Did not work
Without boot  button - it started working.

closed 2023-06-06 by walkiewicz125