espressif/vscode-esp-idf-extension #979: Release/v5.1: Python requirements are not satisfied

eos1d3, 2023-05-27:
### OS

MacOS

### Operating System version

macOS 13.4

### Visual Studio Code version

1.78.2

### ESP-IDF version

5.1

### Python version

3.11.2

### Doctor command output

na

### Description

After deleting ESP-IDF folder, I installed IDF 5.0.2 to 5.1.

It seems installation for 5.1 package has problem. Nothing can work again for every command.

/usr/local/bin/ninja

```

Install 5.0.2 again and it works again.

### Debug Message

```plain
When a project file is open:

/usr/local/bin/ninja

The following Python requirements are not satisfied:
esp-idf-monitor - The 'esp-idf-monitor' distribution was not found and is required by the application
esp-idf-kconfig - The 'esp-idf-kconfig' distribution was not found and is required by the application
esp-idf-size - The 'esp-idf-size' distribution was not found and is required by the application
To install the missing packages, please run "install.sh"
Diagnostic information:
    IDF_PYTHON_ENV_PATH: /Users/andy/.espressif/python_env/idf5.0_py3.11_env
    Python interpreter used: /Users/andy/.espressif/python_env/idf5.0_py3.11_env/bin/python
Command failed: "/Users/andy/.espressif/python_env/idf5.0_py3.11_env/bin/python" "/Users/andy/esp/esp-idf/tools/check_python_dependencies.py" -r "/Users/andy/esp/esp-idf/tools/requirements/requirements.core.txt"
idf_versions.txt has (1) KB
idf_versions.txt progress: 100.00% (0.10 / 0.10) KB
/usr/local/bin/cmake
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2023-05-29:
There is nothing related to the extension in the issue information you shared. Could you share the required doctor command output information here. Is hard to understand the issue if you don't share the information we can use to debug your problem.

It seems you might have some local settings (in your project's .vscode/settings.json) that could be overridden whatever ESP-IDF changes you are doing. Make sure you are saving the settings in the proper location.

eos1d3, 2023-05-29:
When installing with IDF 5.1

```
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS darwin x64 22.5.0 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 /usr/local/bin:/System/Cryptexes/App/usr/bin:/usr/bin:/bin:/usr/sbin:/sbin:/opt/ST/STM32CubeCLT:/opt/ST/STM32CubeCLT/GNU-tools-for-STM32/bin:/opt/ST/STM32CubeCLT/STM32CubeProgrammer/bin:/opt/ST/STM32CubeCLT/STLink-gdb-server/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/local/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/appleinternal/bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.78.2 
Visual Studio Code language en 
Visual Studio Code shell /bin/zsh 
ESP-IDF Extension version 1.6.3 
Workspace folder <HOMEPATH>/test3 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) <HOMEPATH>/esp/esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) <HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/11.2_20220823/xtensa-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/11.2_20220823/riscv32-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin:<HOMEPATH>/.espressif/tools/xtensa-esp32s2-elf/esp-2022r1-11.2.0/xtensa-esp32s2-elf/bin:<HOMEPATH>/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin:<HOMEPATH>/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:<HOMEPATH>/.espressif/tools/cmake/3.24.0/CMake.app/Contents/bin:<HOMEPATH>/.espressif/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/bin:<HOMEPATH>/.espressif/tools/ninja/1.10.2:<HOMEPATH>/.espressif/tools/esp-rom-elfs/20220823
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: <HOMEPATH>/.espressif/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/share/openocd/scripts
    ESP_ROM_ELF_DIR: <HOMEPATH>/.espressif/tools/esp-rom-elfs/20220823/
Virtual env Python Path (idf.pythonBinPath) <HOMEPATH>/.espressif/python_env/idf5.0_py3.11_env/bin/python
Serial port (idf.port) /dev/cu.usbserial-14140
OpenOCD Configs (idf.openOcdConfigs) board/esp32c3-ftdi.cfg
ESP-IDF Tools Path (idf.toolsPath) <HOMEPATH>/.espressif
Git Path (idf.gitPath) /usr/bin/git
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to <HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/11.2_20220823/xtensa-esp-elf-gdb/bin: true
Access to <HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/11.2_20220823/riscv32-esp-elf-gdb/bin: true
Access to <HOMEPATH>/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/xtensa-esp32s2-elf/esp-2022r1-11.2.0/xtensa-esp32s2-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/cmake/3.24.0/CMake.app/Contents/bin: false
Access to <HOMEPATH>/.espressif/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/bin: true
Access to <HOMEPATH>/.espressif/tools/ninja/1.10.2: false
Access to <HOMEPATH>/.espressif/tools/esp-rom-elfs/20220823: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH true
Access to Ninja in environment PATH true
Access to ESP-IDF Tools Path (idf.toolsPath) true
-------------------------------------------------------- Configurations has spaces -------------------------------------------------------------
Spaces in system environment Path false
Spaces in ESP-ADF Path (idf.espAdfPath) false
Spaces in ESP-IDF Path (idf.espIdfPath) false
Spaces in ESP-MDF Path (idf.espMdfPath) false
Spaces in ESP-Matter Path (idf.espMatterPath) false
Spaces in ESP-IDF Custom extra paths
Spaces in <HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/11.2_20220823/xtensa-esp-elf-gdb/bin: false
Spaces in <HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/11.2_20220823/riscv32-esp-elf-gdb/bin: false
Spaces in <HOMEPATH>/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin: false
Spaces in <HOMEPATH>/.espressif/tools/xtensa-esp32s2-elf/esp-2022r1-11.2.0/xtensa-esp32s2-elf/bin: false
Spaces in <HOMEPATH>/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin: false
Spaces in <HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin: false
Spaces in <HOMEPATH>/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin: false
Spaces in <HOMEPATH>/.espressif/tools/cmake/3.24.0/CMake.app/Contents/bin: false
[... cut ...]
                    "${config:idf.espAdfPath}/components/**",
                    "${config:idf.espAdfPathWin}/components/**",
                    "${workspaceFolder}"
                ],
                "limitSymbolsToIncludedHeaders": false
            },
            "configurationProvider": "ms-vscode.cmake-tools"
        }
    ],
    "version": 4
} 
--------------------------------------------------------------------------------------------------------------------------------------------

```

eos1d3, 2023-05-29:
Using diff command, you can see the differences in two logs:

```
< ESP-IDF version 5.1
---
> ESP-IDF version 5.0.2
122c122
< Check ESP-IDF Python packages Error
---
> Check ESP-IDF Python packages Python requirements are satisfied.
166,167d165
< ----------------------------------------------------------- Latest error -----------------------------------------------------------------
< Latest error at Unknown error in ESP-IDF doctor command
```

brianignacio5, 2023-05-30:
Could you check the extension log ? What was the error when installing ESP-IDF python packages ?

Notice that you are using the same virtual environment for both, which should not be the case 
`<HOMEPATH>/.espressif/python_env/idf5.0_py3.11_env/bin/python` works for 5.0.2 as it should but should be part of 5.1

brianignacio5, 2023-05-30:
It could also be that you have local settings that override User or workspace settings. Check the value of `idf.pythonBinPath` in settings.json

As described in the [troubleshooting section](https://github.com/espressif/vscode-esp-idf-extension#troubleshooting) the log and the issue template is:

Windows: `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log`
Linux & MacOSX: `$HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log`

eos1d3, 2023-05-30:
I usually do not have any extra settings. And I use the extension to create
a new blink project. Only 5.0.2 can compile. 5.1 gives the same errors.

Creating new project by the extension tells nothing is related to local
settings. It is right?

brianignacio5, 2023-05-30:
I just don't understand what happened with `idf.pythonBinPath` when you changed to 5.1. Seems you used the same ESP-IDF directory for both versions. Is that the case? How did you change the esp-idf version ? Did you use the extension setup wizard to download and install 5.1 ? 

For some reason the `"idf.pythonBinPath": "/Users/andy/.espressif/python_env/idf5.0_py3.11_env/bin/python"` is being used because is defined in some settings.json (either User, workspace or the current project .vscode/settings.json)

eos1d3, 2023-05-30:
I never keep two versions of IDF at the same time. Before installing IDF 5.1, I delete `esp-idf` folder completely.

I always use the extension setup wizard to download IDF and tools for this test.  I never change anything after installation. And I immediately create a new blink example. Only IDF 5.0.2 and below work.

I do not define anything manually in user settings or workspace. So should I delete .vscode folder to test again?
I did delete .espressif yesterday and this is not helpful.

I never add this `"/Users/andy/.espressif/python_env/idf5.0_py3.11_env/bin/python"` manually. So I think it is done by the extension ( or the previous version of the extension). I will check .vscode folder.

eos1d3, 2023-05-30:
Hi, after deleting `.vscode` folder, everything works now.

The main point is, I never change `idf.pythonBinPath` manually. It can be created by the extension setup wizard as I used it to setup IDF 5.0.1 and 5.0.2 before. But in order to make IDF 5.1 work, the `.vscode` has to be cleaned.

I do not know if the extension can correct the path for 5.1 automatically or not.

brianignacio5, 2023-05-30:
The thing is that extension setup wizard can save settings in different locations. Global, Workspace or Workspace Folder (current project). If for some reason you saved the settings in the `.vscode/settings.json` for 5.0.2, if you try to use the extension setup wizard to set up 5.1 in the Global location, the setup will work, but vscode will still use the `<current_project_dir>/.vscode/settings.json` because of precedence. This is explained in [here](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/MULTI_PROJECTS.md)

eos1d3, 2023-05-30:
@brianignacio5 

It is not right for the extension. When I delete IDF 5.1 and use the wizard to install 5.0.2, it still default everything to 5.1.

So new setup of IDF 5.0.2 never use the new 5.0.2 settings. It also uses previous settings instead.

Even I created a new blink example. idf.pythonBinPath now points to 5.1 but I am using 5.0.2.

And the worst is, even I modify idf.pythonBinPath in settings.json, the new example still cannot compile!!!

```
-- Found Git: /usr/bin/git (found version "2.39.2 (Apple Git-143)") 
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /Users/andy/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/riscv32-esp-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Users/andy/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/riscv32-esp-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Users/andy/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/riscv32-esp-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32c3
Processing 2 dependencies:
[1/2] espressif/led_strip (2.3.2)
[2/2] idf (5.0.2)
-- Project sdkconfig file /Users/andy/test5/sdkconfig
CMake Error at /Users/andy/esp/esp-idf/tools/cmake/tool_version_check.cmake:36 (message):
  

  Tool doesn't match supported version from list ['esp-2022r1-11.2.0']:
  /Users/andy/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/riscv32-esp-elf-gcc

  Please try to run 'idf.py fullclean' to solve it.

Call Stack (most recent call first):
  /Users/andy/esp/esp-idf/components/esp_common/project_include.cmake:10 (check_expected_tool_version)
  /Users/andy/esp/esp-idf/tools/cmake/build.cmake:380 (include)
  /Users/andy/esp/esp-idf/tools/cmake/build.cmake:599 (__build_process_project_includes)
  /Users/andy/esp/esp-idf/tools/cmake/project.cmake:447 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!

```

eos1d3, 2023-05-30:
After installation of IDF 5.0.2, this is the doctor comment log

```
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS darwin x64 22.5.0 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 /usr/local/bin:/System/Cryptexes/App/usr/bin:/usr/bin:/bin:/usr/sbin:/sbin:/opt/ST/STM32CubeCLT:/opt/ST/STM32CubeCLT/GNU-tools-for-STM32/bin:/opt/ST/STM32CubeCLT/STM32CubeProgrammer/bin:/opt/ST/STM32CubeCLT/STLink-gdb-server/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/local/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/appleinternal/bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.78.2 
Visual Studio Code language en 
Visual Studio Code shell /bin/zsh 
ESP-IDF Extension version 1.6.3 
Workspace folder <HOMEPATH>/test5 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) <HOMEPATH>/esp/esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) <HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/12.1_20221002/xtensa-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/12.1_20221002/riscv32-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin:<HOMEPATH>/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin:<HOMEPATH>/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin:<HOMEPATH>/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:<HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230419/openocd-esp32/bin:<HOMEPATH>/.espressif/tools/esp-rom-elfs/20230320
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: <HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230419/openocd-esp32/share/openocd/scripts
    ESP_ROM_ELF_DIR: <HOMEPATH>/.espressif/tools/esp-rom-elfs/20230320/
Virtual env Python Path (idf.pythonBinPath) <HOMEPATH>/.espressif/python_env/idf5.0_py3.11_env/bin/python
Serial port (idf.port) /dev/cu.SLAB_USBtoUART
OpenOCD Configs (idf.openOcdConfigs) board/esp32c3-ftdi.cfg
ESP-IDF Tools Path (idf.toolsPath) <HOMEPATH>/.espressif
Git Path (idf.gitPath) /usr/bin/git
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to <HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/12.1_20221002/xtensa-esp-elf-gdb/bin: true
Access to <HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/12.1_20221002/riscv32-esp-elf-gdb/bin: true
Access to <HOMEPATH>/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230419/openocd-esp32/bin: true
Access to <HOMEPATH>/.espressif/tools/esp-rom-elfs/20230320: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH true
Access to Ninja in environment PATH true
Access to ESP-IDF Tools Path (idf.toolsPath) true
-------------------------------------------------------- Configurations has spaces -------------------------------------------------------------
Spaces in system environment Path false
Spaces in ESP-ADF Path (idf.espAdfPath) false
Spaces in ESP-IDF Path (idf.espIdfPath) false
Spaces in ESP-MDF Path (idf.espMdfPath) false
Spaces in ESP-Matter Path (idf.espMatterPath) false
Spaces in ESP-IDF Custom extra paths
Spaces in <HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/12.1_20221002/xtensa-esp-elf-gdb/bin: false
Spaces in <HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/12.1_20221002/riscv32-esp-elf-gdb/bin: false
Spaces in <HOMEPATH>/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin: false
Spaces in <HOMEPATH>/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin: false
Spaces in <HOMEPATH>/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin: false
Spaces in <HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin: false
Spaces in <HOMEPATH>/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin: false
Spaces in <HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230419/openocd-esp32/bin: false
Spaces in <HOMEPATH>/.espressif/tools/esp-rom-elfs/20230320: false
Spaces in Virtual env Python Path (idf.pythonBinPath) false
Spaces in ESP-IDF Tools Path (idf.toolsPath) false
[... cut ...]
                    "${config:idf.espIdfPathWin}/components",
                    "${config:idf.espAdfPath}/components/**",
                    "${config:idf.espAdfPathWin}/components/**",
                    "${workspaceFolder}"
                ],
                "limitSymbolsToIncludedHeaders": false
            }
        }
    ],
    "version": 4
}
 
--------------------------------------------------------------------------------------------------------------------------------------------

```

eos1d3, 2023-05-30:
The bug is: Previous IDF Wizard setup will kill new IDF setup, until you delete .vscode folder.

brianignacio5, 2023-05-31:
Again, the reason is that settings can be written in 3 places where the Workspace Folder will always have precedence over workspace settings which will always have precedence over user settings. You need to make sure that when you install the new version the settings are saved in the proper location. 

When you use the setup wizard, did you selected workspace folder  in the select where to save settings drop down ?

Is not a bug. Is a VSCode setting resolution feature. It allows you to use multiple esp-idf versions in the same computer. The reason that the previous setup is still being used is because those settings were not replaced.