espressif/vscode-esp-idf-extension #980: Build, flash and monitor doesn't work without Size task

kossnikita, 2023-05-29:
### OS

Windows

### Operating System version

W11

### Visual Studio Code version

1.76.2

### ESP-IDF version

5.0.1

### Python version

3.8

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.22000 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Python311\Scripts\;C:\Python311\;C:\Program Files\Zulu\zulu-17\bin\;C:\Program Files\Eclipse Adoptium\jdk-17.0.2.8-hotspot\bin;C:\Program Files\Common Files\Oracle\Java\javapath;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\MATLAB\R2022a\runtime\win64;C:\Program Files\MATLAB\R2022a\bin;C:\Program Files\dotnet\;C:\Program Files\Git\cmd;C:\ProgramData\chocolatey\bin;C:<HOMEPATH>\AppData\Roaming\nvm;C:\Program Files\nodejs;C:\Program Files (x86)\ForceBindIP;C:\Python310;C:\Python310\Scripts\;C:\Program Files (x86)\STMicroelectronics\STM32 ST-LINK Utility\ST-LINK Utility;C:\Program Files\apache-maven-3.8.6\bin;C:\Program Files (x86)\avr8-gnu-toolchain\avr\bin;C:\Program Files (x86)\avr8-gnu-toolchain\bin;C:\Program Files\w64devkit\bin;C:\Program Files\avrdude;C:\Program Files (x86)\avr8-gnu-toolchain\simulavr;C:\Program Files\xpack-openocd\bin;C:\Program Files\CMake\bin;C:\Program Files (x86)\ninja-win;C:\Program Files\qemu;C:\Program Files\doxygen\bin;C:\Program Files\Graphviz\bin;C:\Program Files\Microchip\xc8\v2.40\bin;C:\Program Files\nodejs\;C:\Program Files\PowerShell\7\;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin;C:<HOMEPATH>\AppData\Roaming\nvm;C:\Program Files\nodejs;C:<HOMEPATH>\AppData\Roaming\.minecraft\MultiMC;C:<HOMEPATH>\.dotnet\tools;C:\Program Files\SDCC\bin;C:<HOMEPATH>\.cargo\bin;C:<HOMEPATH>\AppData\Roaming\npm 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.76.2 
Visual Studio Code language en 
Visual Studio Code shell C:\Program Files\PowerShell\7\pwsh.exe 
ESP-IDF Extension version 1.6.3 
Workspace folder d:\Nevoton\unilect-meter\firmware 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:/Espressif/frameworks/esp-idf-v5.0.1/
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) C:\Espressif\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin;C:\Espressif\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin;C:\Espressif\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin;C:\Espressif\tools\xtensa-esp32s2-elf\esp-2022r1-11.2.0\xtensa-esp32s2-elf\bin;C:\Espressif\tools\xtensa-esp32s3-elf\esp-2022r1-11.2.0\xtensa-esp32s3-elf\bin;C:\Espressif\tools\riscv32-esp-elf\esp-2022r1-11.2.0\riscv32-esp-elf\bin;C:\Espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:\Espressif\tools\cmake\3.24.0\bin;C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\bin;C:\Espressif\tools\ninja\1.10.2;C:\Espressif\tools\idf-exe\1.0.3;C:\Espressif\tools\ccache\4.6.2\ccache-4.6.2-windows-x86_64;C:\Espressif\tools\dfu-util\0.9\dfu-util-0.9-win64;C:\Espressif\tools\esp-rom-elfs\20220823
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20221026/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
    ESP_ROM_ELF_DIR: C:\Espressif\tools\esp-rom-elfs\20220823/
Virtual env Python Path (idf.pythonBinPath) C:\Espressif\python_env\idf5.0_py3.8_env\Scripts\python.exe
Serial port (idf.port) COM9
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,target/esp32.cfg
ESP-IDF Tools Path (idf.toolsPath) C:\Espressif
Git Path (idf.gitPath) C:\Espressif\tools\idf-git\2.30.1\cmd\git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to C:\Espressif\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin: true
Access to C:\Espressif\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin: true
Access to C:\Espressif\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin: true
Access to C:\Espressif\tools\xtensa-esp32s2-elf\esp-2022r1-11.2.0\xtensa-esp32s2-elf\bin: true
Access to C:\Espressif\tools\xtensa-esp32s3-elf\esp-2022r1-11.2.0\xtensa-esp32s3-elf\bin: true
Access to C:\Espressif\tools\riscv32-esp-elf\esp-2022r1-11.2.0\riscv32-esp-elf\bin: true
Access to C:\Espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin: true
Access to C:\Espressif\tools\cmake\3.24.0\bin: true
Access to C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\bin: true
Access to C:\Espressif\tools\ninja\1.10.2: true
Access to C:\Espressif\tools\idf-exe\1.0.3: true
Access to C:\Espressif\tools\ccache\4.6.2\ccache-4.6.2-windows-x86_64: true
Access to C:\Espressif\tools\dfu-util\0.9\dfu-util-0.9-win64: true
Access to C:\Espressif\tools\esp-rom-elfs\20220823: true
[... 119 lines cut ...]
                    "${config:idf.espAdfPathWin}/components/**",
                    "${workspaceFolder}"
                ],
                "limitSymbolsToIncludedHeaders": false
            }
        }
    ],
    "version": 4
}
 
--------------------------------------------------------------------------------------------------------------------------------------------

### Description

In last extension update with Size task disable option, after disable "Flash, build and monitor" stuck after build. This is not followed by flashing. It just stops forever in the place where the Size task used to be. Enabling the Size task option solves the problem.

### Debug Message

```plain
No error message
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

vsh-ektos, 2023-05-29:
@kossnikita I warned them but they ignored me
https://github.com/espressif/vscode-esp-idf-extension/issues/954#issuecomment-1552736264

brianignacio5, 2023-05-30:
Apologies I missed the comment.

I added the fix in the linked PR. You can try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/721155814.zip) to see if it is fixed.

vsh-ektos, 2023-05-30:
@brianignacio5 Worked
thanks

PR #982 (2023-06-15): fix resolve tasks manager
## Description

Fix tasks manager executing empty tasks. This happened when idf size task was disable.

Fixes #980

## Type of change

Please delete options that are not relevant.
- Bug fix (non-breaking change which fixes an issue)

## Steps to test this pull request

Provide a list of steps to test changes in this PR and required output

1. Click on "ESP-IDF: Build your project" when `idf.enableSizeTaskAfterBuildTask` is false.
2. Execute action.
3. Observe results.

- Expected behaviour: 

Build notification is not hanging anymore. Same as IDF Size task is enable.
 [... cut]
closed 2023-06-15 by brianignacio5