espressif/vscode-esp-idf-extension #981: Missing executables and non-functional Python environment

Nistp, 2023-05-30:
### OS

Windows

### Operating System version

OS win32 x64 10.0.19045 

### Visual Studio Code version

1.78.2 

### ESP-IDF version

5.02

### Python version

3.8.7

### Doctor command output

```
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19045 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\OpenSSH\;C:\Windows\SysWOW64;C:\Program Files\Python39\Scripts\;C:\Program Files\Python39\;C:\ProgramData\Oracle\Java\javapath;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files\Common Files\Autodesk Shared\;C:\Program Files\OpenCV\opencv\build\x64\vc14\bin;C:\Program Files (x86)\GtkSharp\2.12\bin;C:\Program Files\Microsoft SQL Server\120\Tools\Binn\;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\Git\cmd;C:\Program Files\usbipd-win\;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin\;C:\Program Files\IVI Foundation\VISA\Win64\Bin\;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin;C:\Program Files\Docker\Docker\resources\bin;C:\ProgramData\DockerDesktop\version-bin;C:\Program Files\Microsoft VS Code\bin;C:\mozilla-build\;C:\Program Files\TortoiseHg\;C:\Program Files\dotnet\;C:\Program Files\nodejs\;C:\Program Files\WireGuard\;C:\CMake\bin;C:<HOMEPATH>\AppData\Local\Programs\MiKTeX\miktex\bin\x64\;C:\texlive\2022\bin\win32;M:\Toolchains\Arm\11.2 2022.02\bin;C:<HOMEPATH>\.cargo\bin;C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2021.10\bin;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\AppData\Local\GitHubDesktop\bin;M:\Intel\ModelSim\modelsim_ase\win32aloem;C:\Program Files\OpenCV\opencv\build\x64\vc14\bin;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\.dotnet\tools;C:<HOMEPATH>\AppData\Roaming\npm;C:<HOMEPATH>\AppData\Local\Programs\Julia-1.8.5\bin; 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.78.2 
Visual Studio Code language en 
Visual Studio Code shell C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.6.3 
Workspace folder c:<HOMEPATH>\Documents\Projects\ESP32Projects\hello_world 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) c:\Espressif\frameworks\esp-idf-v5.0.2
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) c:\Espressif\tools\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin;c:\Espressif\tools\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin;c:\Espressif\tools\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin;c:\Espressif\tools\tools\xtensa-esp32s2-elf\esp-2022r1-11.2.0\xtensa-esp32s2-elf\bin;c:\Espressif\tools\tools\xtensa-esp32s3-elf\esp-2022r1-11.2.0\xtensa-esp32s3-elf\bin;c:\Espressif\tools\tools\riscv32-esp-elf\esp-2022r1-11.2.0\riscv32-esp-elf\bin;c:\Espressif\tools\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;c:\Espressif\tools\tools\cmake\3.24.0\bin;c:\Espressif\tools\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\bin;c:\Espressif\tools\tools\ninja\1.10.2;c:\Espressif\tools\tools\idf-exe\1.0.3;c:\Espressif\tools\tools\ccache\4.6.2\ccache-4.6.2-windows-x86_64;c:\Espressif\tools\tools\dfu-util\0.9\dfu-util-0.9-win64;c:\Espressif\tools\tools\esp-rom-elfs\20220823
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: c:\Espressif\tools\tools\openocd-esp32\v0.11.0-esp32-20221026/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
    ESP_ROM_ELF_DIR: c:\Espressif\tools\tools\esp-rom-elfs\20220823/
Virtual env Python Path (idf.pythonBinPath) c:\Espressif\tools\python_env\idf5.0_py3.8_env\Scripts\python.exe
Serial port (idf.port) COM3
OpenOCD Configs (idf.openOcdConfigs) board/esp32c3-builtin.cfg
ESP-IDF Tools Path (idf.toolsPath) c:\Espressif\tools
Git Path (idf.gitPath) c:\Espressif\tools\tools\idf-git\2.30.1\cmd\git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to c:\Espressif\tools\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin: true
Access to c:\Espressif\tools\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin: true
Access to c:\Espressif\tools\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin: true
Access to c:\Espressif\tools\tools\xtensa-esp32s2-elf\esp-2022r1-11.2.0\xtensa-esp32s2-elf\bin: true
Access to c:\Espressif\tools\tools\xtensa-esp32s3-elf\esp-2022r1-11.2.0\xtensa-esp32s3-elf\bin: true
Access to c:\Espressif\tools\tools\riscv32-esp-elf\esp-2022r1-11.2.0\riscv32-esp-elf\bin: true
Access to c:\Espressif\tools\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin: true
Access to c:\Espressif\tools\tools\cmake\3.24.0\bin: true
Access to c:\Espressif\tools\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\bin: true
Access to c:\Espressif\tools\tools\ninja\1.10.2: true
Access to c:\Espressif\tools\tools\idf-exe\1.0.3: true
Access to c:\Espressif\tools\tools\ccache\4.6.2\ccache-4.6.2-windows-x86_64: true
Access to c:\Espressif\tools\tools\dfu-util\0.9\dfu-util-0.9-win64: true
[... 184 lines cut ...]
{"message":"Command failed: ccache.exe --version\n'ccache.exe' is not recognized as an internal or external command,\r\noperable program or batch file.\r\n","stack":"Error: Command failed: ccache.exe --version\n'ccache.exe' is not recognized as an internal or external command,\r\noperable program or batch file.\r\n\n\tat ChildProcess.exithandler (node:child_process:409:12)\n\tat ChildProcess.emit (node:events:513:28)\n\tat maybeClose (node:internal/child_process:1112:16)\n\tat ChildProcess._handle.onexit (node:internal/child_process:304:5)","level":"error","timestamp":"2023-05-30T00:51:08.253Z"}
{"message":"Error checking ccache version","stack":"Error: Command failed: ccache.exe --version\n'ccache.exe' is not recognized as an internal or external command,\r\noperable program or batch file.\r\n\n\tat ChildProcess.exithandler (node:child_process:409:12)\n\tat ChildProcess.emit (node:events:513:28)\n\tat maybeClose (node:internal/child_process:1112:16)\n\tat ChildProcess._handle.onexit (node:internal/child_process:304:5)","level":"error","timestamp":"2023-05-30T00:51:08.254Z"}
{"message":"Command failed: xtensa-esp-elf-gdb-no-python --version\n'xtensa-esp-elf-gdb-no-python' is not recognized as an internal or external command,\r\noperable program or batch file.\r\n","stack":"Error: Command failed: xtensa-esp-elf-gdb-no-python --version\n'xtensa-esp-elf-gdb-no-python' is not recognized as an internal or external command,\r\noperable program or batch file.\r\n\n\tat ChildProcess.exithandler (node:child_process:409:12)\n\tat ChildProcess.emit (node:events:513:28)\n\tat maybeClose (node:internal/child_process:1112:16)\n\tat ChildProcess._handle.onexit (node:internal/child_process:304:5)","level":"error","timestamp":"2023-05-30T00:51:16.843Z"}
{"message":"Error checking xtensa-esp-elf-gdb version","stack":"Error: Command failed: xtensa-esp-elf-gdb-no-python --version\n'xtensa-esp-elf-gdb-no-python' is not recognized as an internal or external command,\r\noperable program or batch file.\r\n\n\tat ChildProcess.exithandler (node:child_process:409:12)\n\tat ChildProcess.emit (node:events:513:28)\n\tat maybeClose (node:internal/child_process:1112:16)\n\tat ChildProcess._handle.onexit (node:internal/child_process:304:5)","level":"error","timestamp":"2023-05-30T00:51:16.844Z"}
```

### Description

Hi,

I came back to the VSC and tried to install the Espressif IDF 5.02 thru the extension and I'm getting the same errors as before when I opened issue #695. Can we try this again? I went ahead and installed it thru the official Windows installer after trying to install and re-install thru t [... cut ...] and again and nothing changes. I also have Python installed already, aside from what gets installed by ESP's installer (which, by the way, is a bit confusing - there is 3.11 in the python environment and 3.8.7 in tools - is that correct?).

Any advice?

### Debug Message

```plain
Will link in another comment.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Nistp, 2023-05-30:
Debug:
```
Using existing c:\Espressif\tools\dist\idf-git-2.30.1-win64.zip
Installing c:\Espressif\tools\tools\idf-git\2.30.1 ...
Warning File c:\Espressif\tools\tools\idf-git\2.30.1\cmd\git.exe
                                        already exists and was not updated.
Warning File c:\Espressif\tools\tools\idf-git\2.30.1\dev\_._
                                        already exists and was not updated.
Warning File c:\Espressif\tools\tools\idf-git\2.30.1\etc\bash.bash_logout
                                        already exists and was not updated.
Warning File c:\Espressif\tools\tools\idf-git\2.30.1\etc\bash.bashrc
                                        already exists and was not updated.
Warning File c:\Espressif\tools\tools\idf-git\2.30.1\etc\fstab
                                        already exists and was not updated.
Warning File c:\Espressif\tools\tools\idf-git\2.30.1\etc\gitattributes
                                        already exists and was not updated.
Warning File c:\Espressif\tools\tools\idf-git\2.30.1\etc\msystem
                                        already exists and was not updated.
Warning File c:\Espressif\tools\tools\idf-git\2.30.1\etc\nsswitch.conf
                                        already exists and was not updated.
Warning File c:\Espressif\tools\tools\idf-git\2.30.1\etc\profile
                                        already exists and was not updated.
Extracted c:\Espressif\tools\tools\idf-git\2.30.1 ...
Using existing c:\Espressif\tools\tools\idf-python\3.8.7
'riscv32-esp-elf-gdb-no-python' is not recognized as an internal or external command,
operable program or batch file.
Command failed: riscv32-esp-elf-gdb-no-python --version
'riscv32-esp-elf-gdb-no-python' is not recognized as an internal or external command,
operable program or batch file.
Error checking riscv32-esp-elf-gdb version
Error: Command failed: riscv32-esp-elf-gdb-no-python --version
'riscv32-esp-elf-gdb-no-python' is not recognized as an internal or external command,
operable program or batch file.

'xtensa-esp-elf-gdb-no-python' is not recognized as an internal or external command,
operable program or batch file.
Command failed: xtensa-esp-elf-gdb-no-python --version
'xtensa-esp-elf-gdb-no-python' is not recognized as an internal or external command,
operable program or batch file.
Error checking xtensa-esp-elf-gdb version
Error: Command failed: xtensa-esp-elf-gdb-no-python --version
'xtensa-esp-elf-gdb-no-python' is not recognized as an internal or external command,
operable program or batch file.

'ccache.exe' is not recognized as an internal or external command,
operable program or batch file.
Command failed: ccache.exe --version
'ccache.exe' is not recognized as an internal or external command,
operable program or batch file.
Error checking ccache version
Error: Command failed: ccache.exe --version
'ccache.exe' is not recognized as an internal or external command,
operable program or batch file.

'openocd' is not recognized as an internal or external command,
operable program or batch file.
Command failed: openocd --version
'openocd' is not recognized as an internal or external command,
operable program or batch file.
Error checking openocd-esp32 version
Error: Command failed: openocd --version
'openocd' is not recognized as an internal or external command,
operable program or batch file.

'esp32ulp-elf-as' is not recognized as an internal or external command,
operable program or batch file.
Command failed: esp32ulp-elf-as --version
'esp32ulp-elf-as' is not recognized as an internal or external command,
operable program or batch file.
Error checking esp32ulp-elf version
[... 447 lines cut ...]
Error checking xtensa-esp32-elf version
Error: Command failed: xtensa-esp32-elf-gcc --version
'xtensa-esp32-elf-gcc' is not recognized as an internal or external command,
operable program or batch file.

'dfu-util' is not recognized as an internal or external command,
operable program or batch file.
Command failed: dfu-util --version
'dfu-util' is not recognized as an internal or external command,
operable program or batch file.
Error checking dfu-util version
Error: Command failed: dfu-util --version
'dfu-util' is not recognized as an internal or external command,
operable program or batch file.

'xtensa-esp32s3-elf-gcc' is not recognized as an internal or external command,
operable program or batch file.
Command failed: xtensa-esp32s3-elf-gcc --version
'xtensa-esp32s3-elf-gcc' is not recognized as an internal or external command,
operable program or batch file.
Error checking xtensa-esp32s3-elf version
Error: Command failed: xtensa-esp32s3-elf-gcc --version
'xtensa-esp32s3-elf-gcc' is not recognized as an internal or external command,
operable program or batch file.
```

brianignacio5, 2023-05-30:
Based on your doctor command seems the extension is configured. Have you tried building and other extension commands ?

Nistp, 2023-05-30:
@brianignacio5 yeah, for example, when I copy the hello_world example to a different folder and try to build it there, I get the following output:

```
C:\CMake\bin\cmake.exe

INFO: Could not find files for the given pattern(s).
```

If I go to the folder thru the CLI, I can run idf-py commands and it at least starts building the project.

brianignacio5, 2023-05-30:
Seems to be resolving to the CMake you already have in PATH instead of the extension one. This happens because vscode doesn't pre-append to environment PATH, only post-append. So it will use the CMake you have in PATH first. If you use export.bat it will pre-append to PATH. 

We tried to pre-append to PATH in vscode but it didn't work. Is this doctor command from the new hello_world example directory. Also what terminal are you using ? What is the command you are trying to use ?

Nistp, 2023-05-30:
@brianignacio5 yeah, the export.bat only works with the CLI. 

Yeah, it's the doctor command from the new hello_world example directory. I'm using ESP-IDF within VSC and I click on the build icon on the bottom left

brianignacio5, 2023-05-30:
Which build button ? I ask because many users use the CMake extension too which also has a build button.

What is the full build output ? Could you share the extension log file also ?

Nistp, 2023-05-30:
@brianignacio5 the ESP-IDF Build project button. For the CMake one, it asks me to set a target, so I know for sure I'm always pressing the right one.

Linking my extension log. 
[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/11600294/esp_idf_vsc_ext.log)

The debug message that I posted is pretty much the build output. I'll run it again in the evening (my time) and post it as a file.

Nistp, 2023-05-31:
@brianignacio5 this is the full build output:

```
C:\CMake\bin\cmake.exe

INFO: Could not find files for the given pattern(s).
```

I get a lot of errors when I first start up VSC and the ESP-IDF terminal loads up in the workspace:
```
'riscv32-esp-elf-gcc' is not recognized as an internal or external command,
operable program or batch file.
Command failed: riscv32-esp-elf-gcc --version
'riscv32-esp-elf-gcc' is not recognized as an internal or external command,
operable program or batch file.
Error checking riscv32-esp-elf version
Error: Command failed: riscv32-esp-elf-gcc --version
'riscv32-esp-elf-gcc' is not recognized as an internal or external command,
operable program or batch file.

'xtensa-esp32-elf-gcc' is not recognized as an internal or external command,
operable program or batch file.
Command failed: xtensa-esp32-elf-gcc --version
'xtensa-esp32-elf-gcc' is not recognized as an internal or external command,
operable program or batch file.
Error checking xtensa-esp32-elf version
Error: Command failed: xtensa-esp32-elf-gcc --version
'xtensa-esp32-elf-gcc' is not recognized as an internal or external command,
operable program or batch file.

'xtensa-esp32s2-elf-gcc' is not recognized as an internal or external command,
operable program or batch file.
Command failed: xtensa-esp32s2-elf-gcc --version
'xtensa-esp32s2-elf-gcc' is not recognized as an internal or external command,
operable program or batch file.
Error checking xtensa-esp32s2-elf version
Error: Command failed: xtensa-esp32s2-elf-gcc --version
'xtensa-esp32s2-elf-gcc' is not recognized as an internal or external command,
operable program or batch file.

'riscv32-esp-elf-gdb-no-python' is not recognized as an internal or external command,
operable program or batch file.
Command failed: riscv32-esp-elf-gdb-no-python --version
'riscv32-esp-elf-gdb-no-python' is not recognized as an internal or external command,
operable program or batch file.
Error checking riscv32-esp-elf-gdb version
Error: Command failed: riscv32-esp-elf-gdb-no-python --version
'riscv32-esp-elf-gdb-no-python' is not recognized as an internal or external command,
operable program or batch file.

'xtensa-esp-elf-gdb-no-python' is not recognized as an internal or external command,
operable program or batch file.
Command failed: xtensa-esp-elf-gdb-no-python --version
'xtensa-esp-elf-gdb-no-python' is not recognized as an internal or external command,
operable program or batch file.
Error checking xtensa-esp-elf-gdb version
Error: Command failed: xtensa-esp-elf-gdb-no-python --version
'xtensa-esp-elf-gdb-no-python' is not recognized as an internal or external command,
operable program or batch file.

'xtensa-esp32s3-elf-gcc' is not recognized as an internal or external command,
operable program or batch file.
Command failed: xtensa-esp32s3-elf-gcc --version
'xtensa-esp32s3-elf-gcc' is not recognized as an internal or external command,
operable program or batch file.
Error checking xtensa-esp32s3-elf version
Error: Command failed: xtensa-esp32s3-elf-gcc --version
'xtensa-esp32s3-elf-gcc' is not recognized as an internal or external command,
operable program or batch file.

'esp32ulp-elf-as' is not recognized as an internal or external command,
operable program or batch file.
Command failed: esp32ulp-elf-as --version
'esp32ulp-elf-as' is not recognized as an internal or external command,
operable program or batch file.
Error checking esp32ulp-elf version
Error: Command failed: esp32ulp-elf-as --version
'esp32ulp-elf-as' is not recognized as an internal or external command,
operable program or batch file.

'openocd' is not recognized as an internal or external command,
operable program or batch file.
Command failed: openocd --version
'openocd' is not recognized as an internal or external command,
operable program or batch file.
Error checking openocd-esp32 version
Error: Command failed: openocd --version
'openocd' is not recognized as an internal or external command,
operable program or batch file.

'idf.py.exe' is not recognized as an internal or external command,
operable program or batch file.
Command failed: idf.py.exe -v
'idf.py.exe' is not recognized as an internal or external command,
operable program or batch file.
Error checking idf-exe version
Error: Command failed: idf.py.exe -v
'idf.py.exe' is not recognized as an internal or external command,
operable program or batch file.

'ccache.exe' is not recognized as an internal or external command,
operable program or batch file.
Command failed: ccache.exe --version
'ccache.exe' is not recognized as an internal or external command,
operable program or batch file.
Error checking ccache version
Error: Command failed: ccache.exe --version
'ccache.exe' is not recognized as an internal or external command,
operable program or batch file.

'dfu-util' is not recognized as an internal or external command,
operable program or batch file.
Command failed: dfu-util --version
'dfu-util' is not recognized as an internal or external command,
operable program or batch file.
Error checking dfu-util version
Error: Command failed: dfu-util --version
'dfu-util' is not recognized as an internal or external command,
operable program or batch file.

'ninja' is not recognized as an internal or external command,
operable program or batch file.
Command failed: ninja --version
'ninja' is not recognized as an internal or external command,
operable program or batch file.
Error checking ninja version
Error: Command failed: ninja --version
'ninja' is not recognized as an internal or external command,
operable program or batch file.
```

brianignacio5, 2023-05-31:
I think the issue is your environment variable PATH. For some reason the executables are not found.  Based on your doctor command it does look like the extension is already configured. What is the value of `idf.cmakeCompilerArgs` of your settings.json ?

I can't reproduce the issue. If you try to use the `Open ESP-IDF Terminal` command can you build from this extension configured terminal ?

Nistp, 2023-06-01:
@brianignacio5 it's possible. When I install ESP-IDF thru the extension, the export is never ran so the relevant paths are never added to my env. PATH. Am I supposed to put anything in? In a previous version I think you were supposed to add some paths manually. I currently have the paths set for IDF_PYTHON_ENV_PATH and IDF_TOOLS_PATH.

>What is the value of idf.cmakeCompilerArgs of your settings.json ?

...I don't have one. Check out my settings.json:

```
{
  "C_Cpp.intelliSenseEngine": "Tag Parser",
  "idf.portWin": "COM3",
  "idf.adapterTargetName": "esp32c3",
  "idf.openOcdConfigs": [
    "board/esp32c3-builtin.cfg"
  ],
  "idf.flashType": "UART",
  "cmake.configureOnOpen": false
}
```

>If you try to use the Open ESP-IDF Terminal command can you build from this extension configured terminal ? 

Within VSC? Nope. Just out of curiosity, I installed Espressif-IDE with ESP-IDF v5.0.1 and that is working fine out of the box. I would prefer using VSC just so I can keep all of my different environments within one (or at least, try to minimize the number) IDE.

closed 2023-06-22 by github-actions[bot]