espressif/vscode-esp-idf-extension #990: Command palette not working for projects in workspace

Samuel-Zhang-Rheem, 2023-06-11:
### OS

Windows

### Operating System version

Windows 10

### Visual Studio Code version

1.78.2

### ESP-IDF version

4.4.4

### Python version

3.8

### Doctor command output

Nothing

### Extension

[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/11714516/esp_idf_vsc_ext.log)

### Description

By using "ESP-IDF: show examples projects" to generate several example projects, each project works fine, can be  built and download to ESP32 kit.

When I create a workspace, add those project to this workspace, then "view-->command palette" only work for the workspace, not work for each project. If I "ESP-IDF: build your projects", it fails: **does not appear to contain CMakeLists.txt.**

ESP project works fine in Eclipse workspace, how to make workspace function in VS code ? 

### Debug Message

```plain
NA
```

### Other Steps to Reproduce

NA

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

radurentea, 2023-06-12:
Hi @Samuel-Zhang-Rheem,

You can have a look over the [Working with multiple projects documentation](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/MULTI_PROJECTS.md).

If you continue to have issues, run the "Doctor command" in the command palette and attach the output here. (After running the doctor command, you will have to use CTRL + V to paste the output to a txt file)

Samuel-Zhang-Rheem, 2023-06-18:
Hello Radu,

I follow you attached link, this is not exactly my question. Here is what I did:
**Step1.** Use "view"->"command palette" ->"Show Examples Projects" to generate three examples: softAP, scan, station.
**Step2.**Use "command palette" to build, flash to device and OpenOCD debug, each examples works fine.
**Step3.** create "workspace", "add folders to workspace" to add softAP, scan, station to this workspace.
**Step4.** within the workspace,  I do "build, flash to device and OpenOCD debug", but every time, it works for the first project "softAP" only, even I point to other project.

Is there any **"you tube"** video explain how to work on multiple ESP32 VScode projects in one workspace?
[image]

brianignacio5, 2023-06-19:
Did you select the workspace folder you want to use ? In the steps you didn't mention selecting the workspace folder.

> You can select the current project by clicking the **ESP-IDF Current Project** Item in the Visual Studio Code Status bar or by pressing F1 and typing **ESP-IDF: Pick a workspace folder**

Samuel-Zhang-Rheem, 2023-06-25:
It works for each folder in a workspace.

The display "Task" is wrong, that is bug? It always display "ESP-IDF" Build (softAP) even I build other project.
The Terminal display the right action.
[image]

brianignacio5, 2023-06-26:
So the task works it just the display. I'll take a look

PR #1001 (2023-06-30): Bugfix/idf versions
## Description

- Add download with retries to idf_versions.txt file and update local fallback file.
- Add timestamps IDF version validation (IDF >= 4.4)
- Add changelog for v1.6.4 release
- Fix workspace folders scope in TaskManager.addTask
- Update readme

Fixes #998
Fixes #990 

## Type of change

Please delete options that are not relevant.
- Bug fix (non-breaking change which fixes an issue)

## Steps to test this pull request

Provide a list of steps to test changes in this PR and required output

1. Click on "Configure ESP-IDF extension" command
2. Execute action.
3. [... cut]
closed 2023-06-30 by brianignacio5