espressif/vscode-esp-idf-extension #993: [Feature Request]: How to uninstall ESP-IDF and ESP-IDF tools installed via ESP-IDF extensions?

hwqchi, 2023-06-15:
### Is your feature request related to a problem?
The installation tutorial does not provide any steps or methods for uninstallation. 
I cannot find any uninstall programs in the directories of ESP-IDF and ESP-IDF Tools, and I am concerned that deleting these two directories directly may leave some remnants in the registry or environment variables.

**Environment:** Windows11, ESP-IDF 5.0.1

### Describe the solution you'd like
Please provide a method for uninstallation in the installation tutorial.

brianignacio5, 2023-06-16:
If you install ESP-IDF with the extension the environment variables are not modified. The only thing is that the extension saves settings in settings.json. As long as you make sure your projects settings.json, the User and Workspace ESP-IDF (`idf.*`) settings are deleted.

Since the extension can re use ESP-IDF from other install like Windows Installer or Eclipse setup and settings.json can be defined on any project's settings.json is a bit complicated to design such uninstall function.

hwqchi, 2023-06-26:
Thanks a lot for your response. Based on your information, I have deleted the following directories and all the files within them:
`C:\Users\%Username%\esp`
`C:\Users\%Username%\.espressif`
After deleting these directories, I proceeded to reinstall the latest version of ESP-IDF framework and successfully upgraded it without encountering any issues during the installation process.

closed 2023-06-26 by hwqchi