espressif/vscode-esp-idf-extension #996: 如何把已经安装在电脑中的idf环境和vscode里的idf插件配对？

woshiwstt50123, 2023-06-16:
### OS

Windows

### Operating System version

Windows 11

### Visual Studio Code version

1.73.1

### ESP-IDF version

5.0.2

### Python version

3.11.2

### Doctor command output

Null

### Description

已经使用“esp-idf-tools-setup-offline-5.0.2.exe”安装了esp-idf 5.0.2，在配置“ESP-IDF Extension”插件时却仍然需要安装相关文件，安装的文件版本甚至低于用“esp-idf-tools-setup-offline-5.0.2.exe”安装的文件。

### Debug Message

```plain
null
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2023-06-16:
You can use the existing ESP-IDF for the Visual Studio Code extension by specifying the path with the `ESP-IDF: Configure ESP-IDF extension` command which shows a setup wizard.

woshiwstt50123, 2023-06-16:
是的，我已经使用了这个命令，进入了一个包含“EXPRESS”“ADVANCED”“USE EXISTING SETUP”三个按钮的界面。我选择第三个按钮，却依然需要下载多余的文件。我想知道原因和避免下载这些文件的方法。毕竟我已经在powershell里成功把Example文件夹中的工程烧录到芯片里了。

brianignacio5, 2023-06-16:
Express -> Select ESP-IDF version: Find ESP-IDF in your system -> Enter ESP-IDF Tools directory (IDF_TOOLS_PATH) to existing IDF_TOOLS_PATH.

Some additional python packages are installed because the extension needs them.

woshiwstt50123, 2023-06-16:
我在esp-idf的文件夹中看到了python 3.11.2，插件又尝试下载python3.8，这是为什么？插件不能使用python3.11吗

woshiwstt50123, 2023-06-28:
新的插件在下载python3.11.2时不会下载pip，具体表现是“.\idf-python\3.11.2\Lib\site-packages”目录下只有一个README.txt文件夹。

woshiwstt50123, 2023-06-28:
C/C++插件报错，无法找到: c:\Users\TangTao\MyWork\ESP_C6\hello_world\${env:ADF_PATH}\components\目录

brianignacio5, 2023-06-28:
I've made a PR to update the extension to download Python 3.11 instead of 3.8

With regards to C/C++, you need to update the `<project_directory_path>/.vscode/c_cpp_properties.json` to include ADF_PATH. How is that file configured ? You can also just use the compile_commands.json and it will be loaded from build system.

woshiwstt50123, 2023-06-28:
是的，我现在正在使用这个新的插件。

我这边的c_cpp_properties.json文件中的内容是这样的：
{
    "configurations": [
        {
            "name": "ESP-IDF",
            "compilerPath": "c:\\Users\\TangTao\\esp\\tools\\xtensa-esp32-elf\\esp-2022r1-11.2.0\\xtensa-esp32-elf\\bin\\xtensa-esp32-elf-gcc.exe",
            "includePath": [
                "${config:idf.espIdfPath}/components/**",
                "${config:idf.espIdfPathWin}/components/**",
                "${config:idf.espAdfPath}/components/**",
                "${config:idf.espAdfPathWin}/components/**",
                "${workspaceFolder}/**"
            ],
            "browse": {
                "path": [
                    "${config:idf.espIdfPath}/components",
                    "${config:idf.espIdfPathWin}/components",
                    "${config:idf.espAdfPath}/components/**",
                    "${config:idf.espAdfPathWin}/components/**",
                    "${workspaceFolder}"
                ],
                "limitSymbolsToIncludedHeaders": false
            }
        }
    ],
    "version": 4
}

compile_commands.json文件我这里好像没有生成。

woshiwstt50123, 2023-06-28:
另外，我通过复制从python官方下载的相关文件到指定的site-package目录后，解决了安装时报错的问题。

brianignacio5, 2023-06-28:
Add `"compileCommands": "${workspaceFolder}/build/compile_commands.json"` or Check this documentation:

https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/C_CPP_CONFIGURATION.md#configuration-with-compile_commandsjson

woshiwstt50123, 2023-06-28:
程序报错的问题解决了。现在有新的问题。
当我烧录程序时，在打开OPENOCD这一步会报错。报错内容如下：

c:\Users\TangTao\esp\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\bin\openocd.exe

[OpenOCD]
Open On-Chip Debugger v0.11.0-esp32-20221026 (2022-10-26-14:48)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html

[OpenOCD]
debug_level: 2

[OpenOCD]
adapter speed: 20000 kHz

[OpenOCD]
Info : Listening on port 6666 for tc
[OpenOCD]
l connections
Info : Listening on port 4444 for telnet connections

[OpenOCD]
❌ Error: no device found
Error: unable to open ftdi device with vid 0403, pid 6010, descr
[OpenOCD]
iption '*', serial '*' at bus location '*'

[OpenOCD]
❌ Error: no device found
Error: unable to open ftdi device with vid 0403, pid 6014, description '*', s
[OpenOCD]
erial '*' at bus location '*'

[OpenOCD]
[Stopped] : OpenOCD Server

这个问题偶尔会有发生。为此已经重装好几回了。

brianignacio5, 2023-06-28:
Did you configure the IDF Target and OpenOCD config files ?

Check the debug tutorial [here](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/debugging.md)

woshiwstt50123, 2023-06-29:
我现在把IDF Target设置成esp32 chip(via ESP USB bridge)，OpenOCD config file选择的是["interface/ftdi/esp32_devkitj_v1.cfg", "target/esp32.cfg"]。目标开发板是ESP32-DevKitC，它使用USB线和电脑连接。PC中显示的相关驱动信息是“Silicon Labs CP210x USB to UART Bridge”。

现在OPENOCD的报错提示如下：
[OpenOCD]
Open On-Chip Debugger v0.11.0-esp32-20221026 (2022-10-26-14:48)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html

[OpenOCD]
debug_level: 2

[OpenOCD]
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1002
Info 
[OpenOCD]
: esp_usb_jtag: capabilities descriptor set to 0x30a
Info : esp_usb_jtag: target chip id set to 1
Warn : Transport "jtag" was already selected

[OpenOCD]
Info : Listening on port 6666 for tcl connections

[OpenOCD]
Info : Listening on port 4444 for telnet connections

[OpenOCD]
❌ Error: esp_usb_jtag: could not find or
[OpenOCD]
 open device!

[OpenOCD]
[Stopped] : OpenOCD Server

woshiwstt50123, 2023-06-29:
此外，C/C++插件又报出一个错误：
Error while processingC/C++(clang-diagnostic-error)
reent.h(9, 14): 'sys/reent.h' file not found

PR #999 (2023-06-29): update embed git python
## Description

Update embed python and embed git to fix checking for existing paths.

Fixes #996
Fixes #1000 

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## Steps to test this pull request

Provide a list of steps to test changes in this PR and required output

1. Click on "Configure ESP-IDF extension" command
2. Run the setup wizard
3. Observe results. Check that the downloaded python is 3.11.2 and git is 2.39.2

- Expected behaviour:

- Expected output:

## How has this been tested?

Manual testing

**Test Configuration**:
* ESP-IDF Vers [... cut]
closed 2023-06-29 by brianignacio5