espressif/vscode-esp-idf-extension #998: Cannot restart install/configuration procedure for Espressif's VS Code extension

benniflash, 2023-06-20:
### OS

Windows

### Operating System version

Windows 10

### Visual Studio Code version

1.79.2

### ESP-IDF version

5.0

### Python version

3.10.4

### Doctor command output

I cannot launch ESP-IDF Doctor command anymore

### Description

Starting from this link https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/get-started/index.html and following the steps to install VSCode and the Espressif IDF extension (https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/get-started/index.html), I cannot configure ESPRESSIF extension: I didn't have the Python line during the Express configuration 

[image]

Then, I tried to uninstall everything, VS Code and Espressif IDF extension, but ASAP I restart the configuration, it doesn't restart from the beginning and it does not advance.
[image]

How can I restart everything from the beginning?

### Debug Message

```plain
Cannot get Debug Message
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2023-06-20:
Can you share the log file ?

It was part of the issue template Extension part:

Please share related content of extension log file located in:

Windows (%USERPROFILE%.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log)
Linux-MacOS ($HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log)

brianignacio5, 2023-06-20:
Seems there is an error getting this file:

https://dl.espressif.com/dl/esp-idf/idf_versions.txt

Can you access to it directly? Also When you try to open the `ESP-IDF: Configure ESP-IDF extension`, what does happen?

benniflash, 2023-06-20:
This is my idf_version.txt file
[idf_versions.txt](https://github.com/espressif/vscode-esp-idf-extension/files/11798373/idf_versions.txt)

I cannot open ESP-IDF: Configure ESP-IDF extension from command palette, maybe because it is still loading the initial configuration:
[image]

benniflash, 2023-06-20:
I can open your file

benniflash, 2023-06-20:
[image]

brianignacio5, 2023-06-20:
Then it seems there is a connection issue from vscode itself.  Have you try closing all windows of vscode an re opening them ? 

Are you using a proxy in vscode ?

benniflash, 2023-06-20:
Yes, I'm using my company's Proxy

radurentea, 2023-06-23:
Hi @benniflash,

You can follow the steps for uninstalling [here](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/INSTALL.md#uninstalling-the-plugin).

You can also delete the esp-idf. Default location is C:\Espressif

closed 2023-06-30 by brianignacio5

benniflash, 2023-07-04:
Hello again,
I'm now trying to follow the step in the following link https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/windows-setup.html
I've updated the driver in order to establish a USB connection with the ESP32-C3-DevKitM-1.
However when I try to configure my project (hello_world) I get this error .log. How can I solve this problem?
[CMakeError.log](https://github.com/espressif/vscode-esp-idf-extension/files/11946847/CMakeError.log)
[CMakeOutput.log](https://github.com/espressif/vscode-esp-idf-extension/files/11946849/CMakeOutput.log)

brianignacio5, 2023-07-04:
How are your running the build? Did you follow the [install tutorial](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/install.md) and [basic use tutorial](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/basic_use.md) ?

brianignacio5, 2023-07-04:
I'm very confused. So if you using Windows you didn't need to install Python yourself (the extension and Windows installer will install Python for your on Windows). The `/usr/bin/python` tells me you are trying to use Linux. Are you running from Linux in Windows (WSL) ?

The CMake error points me your environment is not properly configured. So we need to get that configured properly first.

benniflash, 2023-07-04:
Actually I'm not running WSL. I followed step by step the guide for the windows installation, so I don't know where I switch from Win10 to Linux process.

How can I configure my environment properly?
However If I follow the link you attached I'm not able to CONFIGURE espressif, even if I download both Visual studio and Espressif extension properly.

benniflash, 2023-07-04:
I have an update. I tried to use the espressif IDE instead of VS, I tried to build the hello_world project, but I get this error : 
[image]
So i followed this youtube video to install the missing Tools, and everything seemed to be ok but than i get the following error log file. Can you help me?
[Setup Log 2023-07-04 #003.txt](https://github.com/espressif/vscode-esp-idf-extension/files/11949455/Setup.Log.2023-07-04.003.txt)

brianignacio5, 2023-07-05:
Have you tried the new version of the extension in visual studio code ? 

About Espressif IDE you need to go to the [Eclipse Plugin repository](https://github.com/espressif/idf-eclipse-plugin) for assistance

PR #1001 (2023-06-30): Bugfix/idf versions
## Description

- Add download with retries to idf_versions.txt file and update local fallback file.
- Add timestamps IDF version validation (IDF >= 4.4)
- Add changelog for v1.6.4 release
- Fix workspace folders scope in TaskManager.addTask
- Update readme

Fixes #998
Fixes #990 

## Type of change

Please delete options that are not relevant.
- Bug fix (non-breaking change which fixes an issue)

## Steps to test this pull request

Provide a list of steps to test changes in this PR and required output

1. Click on "Configure ESP-IDF extension" command
2. Execute action.
3. [... cut]