espressif/vscode-esp-idf-extension #1002: ESP-IDF不能烧录程序

woshiwstt50123, 2023-07-03:
### OS

Windows

### Operating System version

Windows11 22H2

### Visual Studio Code version

1.73.1

### ESP-IDF version

5.0.2

### Python version

3.11.2

### Doctor command output

---------------------------------------------------- Visual Studio Code launch.json --------------------------------------------------------
{
  "version": "0.2.0",
  "configurations": [
    {
      "type": "espidf",
      "name": "Launch",
      "request": "launch"
    }
  ]
} 
---------------------------------------------------- Visual Studio Code c_cpp_properties.json ----------------------------------------------
{
    "configurations": [
        {
            "name": "ESP-IDF",
            "compilerPath": "c:\\Users\\TangTao\\esp\\tools\\xtensa-esp32-elf\\esp-2022r1-11.2.0\\xtensa-esp32-elf\\bin\\xtensa-esp32-elf-gcc.exe",
            "includePath": [
                "${config:idf.espIdfPath}/components/**",
                "${config:idf.espIdfPathWin}/components/**",
                "${config:idf.espAdfPath}/components/**",
                "${config:idf.espAdfPathWin}/components/**",
                "${workspaceFolder}/**"
            ],
            "browse": {
                "path": [
                    "${config:idf.espIdfPath}/components",
                    "${config:idf.espIdfPathWin}/components",
                    "${config:idf.espAdfPath}/components/**",
                    "${config:idf.espAdfPathWin}/components/**",
                    "${workspaceFolder}"
                ],
                "limitSymbolsToIncludedHeaders": false
            },
            "compileCommands": "${workspaceFolder}/build/compile_commands.json"
        }
    ],
    "version": 4
}

### Extension

{\"espIdfPath\":true,\"espAdfPath\":false,\"espMdfPath\":false,\"espMatterPath\":false,\"espIdfToolsPaths\":{\"c:<HOMEPATH>\\\\esp\\\\tools\\\\xtensa-esp-elf-gdb\\\\11.2_20220823\\\\xtensa-esp-elf-gdb\\\\bin\":true,\"c:<HOMEPATH>\\\\esp\\\\tools\\\\riscv32-esp-elf-gdb\\\\11.2_20220823\\\\riscv32-esp-elf-gdb\\\\bin\":true,\"c:<HOMEPATH>\\\\esp\\\\tools\\\\xtensa-esp32-elf\\\\esp-2022r1-11.2.0\\\\xtensa-esp32-elf\\\\bin\":true,\"c:<HOMEPATH>\\\\esp\\\\tools\\\\xtensa-esp32s2-elf\\\\esp-2022r1-11.2.0\\\\xtensa-esp32s2-elf\\\\bin\":true,\"c:<HOMEPATH>\\\\esp\\\\tools\\\\xtensa-esp32s3-elf\\\\esp-2022r1-11.2.0\\\\xtensa-esp32s3-elf\\\\bin\":true,\"c:<HOMEPATH>\\\\esp\\\\tools\\\\riscv32-esp-elf\\\\esp-2022r1-11.2.0\\\\riscv32-esp-elf\\\\bin\":true,\"c:<HOMEPATH>\\\\esp\\\\tools\\\\esp32ulp-elf\\\\2.35_20220830\\\\esp32ulp-elf\\\\bin\":true,\"c:<HOMEPATH>\\\\esp\\\\tools\\\\cmake\\\\3.24.0\\\\bin\":true,\"c:<HOMEPATH>\\\\esp\\\\tools\\\\openocd-esp32\\\\v0.11.0-esp32-20221026\\\\openocd-esp32\\\\bin\":true,\"c:<HOMEPATH>\\\\esp\\\\tools\\\\ninja\\\\1.10.2\":true,\"c:<HOMEPATH>\\\\esp\\\\tools\\\\idf-exe\\\\1.0.3\":true,\"c:<HOMEPATH>\\\\esp\\\\tools\\\\ccache\\\\4.6.2\\\\ccache-4.6.2-windows-x86_64\":true,\"c:<HOMEPATH>\\\\esp\\\\tools\\\\dfu-util\\\\0.9\\\\dfu-util-0.9-win64\":true,\"c:<HOMEPATH>\\\\esp\\\\tools\\\\esp-rom-elfs\\\\20220823\":true},\"pythonBinPath\":true,\"toolsPath\":true},\"configurationSpacesValidation\":{\"customExtraPaths\":{\"c:<HOMEPATH>\\\\esp\\\\tools\\\\xtensa-esp-elf-gdb\\\\11.2_20220823\\\\xtensa-esp-elf-gdb\\\\bin\":false,\"c:<HOMEPATH>\\\\esp\\\\tools\\\\riscv32-esp-elf-gdb\\\\11.2_20220823\\\\riscv32-esp-elf-gdb\\\\bin\":false,\"c:<HOMEPATH>\\\\esp\\\\tools\\\\xtensa-esp32-elf\\\\esp-2022r1-11.2.0\\\\xtensa-esp32-elf\\\\bin\":false,\"c:<HOMEPATH>\\\\esp\\\\tools\\\\xtensa-esp32s2-elf\\\\esp-2022r1-11.2.0\\\\xtensa-esp32s2-elf\\\\bin\":false,\"c:<HOMEPATH>\\\\esp\\\\tools\\\\xtensa-esp32s3-elf\\\\esp-2022r1-11.2.0\\\\xtensa-esp32s3-elf\\\\bin\":false,\"c:<HOMEPATH>\\\\esp\\\\tools\\\\riscv32-esp-elf\\\\esp-2022r1-11.2.0\\\\riscv32-esp-elf\\\\bin\":false,\"c:<HOMEPATH>\\\\esp\\\\tools\\\\esp32ulp-elf\\\\2.35_20220830\\\\esp32ulp-elf\\\\bin\":false,\"c:<HOMEPATH>\\\\esp\\\\tools\\\\cmake\\\\3.24.0\\\\bin\":false,\"c:<HOMEPATH>\\\\esp\\\\tools\\\\openocd-esp32\\\\v0.11.0-esp32-20221026\\\\openocd-esp32\\\\bin\":false,\"c:<HOMEPATH>\\\\esp\\\\tools\\\\ninja\\\\1.10.2\":false,\"c:<HOMEPATH>\\\\esp\\\\tools\\\\idf-exe\\\\1.0.3\":false,\"c:<HOMEPATH>\\\\esp\\\\tools\\\\ccache\\\\4.6.2\\\\ccache-4.6.2-windows-x86_64\":false,\"c:<HOMEPATH>\\\\esp\\\\tools\\\\dfu-util\\\\0.9\\\\dfu-util-0.9-win64\":false,\"c:<HOMEPATH>\\\\esp\\\\tools\\\\esp-rom-elfs\\\\20220823\":false},\"espAdfPath\":false,\"espIdfPath\":false,\"espMatterPath\":false,\"espMdfPath\":false,\"gitPath\":false,\"pythonBinPath\":false,\"toolsPath\":false,\"systemEnvPath\":true},\"debugAdapterRequirements\":{\"output\":\"C:<HOMEPATH>\\\\esp\\\\esp-idf\\\\tools\\\\check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html\\r\\n  import pkg_resources\\r\\nPython requirements are satisfied.\",\"result\":\"C:<HOMEPATH>\\\\esp\\\\esp-idf\\\\tools\\\\check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html\\r\\n  import pkg_resources\\r\\nPython requirements are satisfied.\"},\"espIdfToolsVersions\":[{\"actual\":\"11.2_20220823\",\"description\":\"GDB for Xtensa\",\"doesToolExist\":true,\"env\":{},\"expected\":\"11.2_20220823\",\"hashResult\":true,\"hasFailed\":false,\"id\":\"xtensa-esp-elf-gdb\",\"name\":\"xtensa-esp-elf-gdb\",\"path\":\"c:<HOMEPATH>\\\\esp\\\\xtensa-esp-elf-gdb\\\\11.2_20220823\\\\xtensa-esp-elf-gdb\\\\bin\",\"progress\":\"0.00%\",\"progressDetail\":\"\"},{\"actual\":\"11.2_20220823\",\"description\":\"GDB for RISC-V\",\"doesToolExist\":true,\"env\":{},\"expected\":\"11.2_20220823\",\"hashResult\":true,\"hasFailed\":false,\"id\":\"riscv32-esp-elf-gdb\",\"name\":\"riscv32-esp-elf-gdb\",\"path\":\"c:<HOMEPATH>\\\\esp\\\\riscv32-esp-elf-gdb\\\\11.2_20220823\\\\riscv32-esp-elf-gdb\\\\bin\",\"progress\":\"0.00%\",\"progressDetail\":\"\"},{\"actual\":\"esp-2022r1-11.2.0\",\"description\":\"Toolchain for Xtensa (ESP32) based on GCC\",\"doesToolExist\":true,\"env\":{},\"expected\":\"esp-2022r1-11.2.0\",\"hashResult\":true,\"hasFailed\":false,\"id\":\"xtensa-esp32-elf\",\"name\":\"xtensa-esp32-elf\",\"path\":\"c:<HOMEPATH>\\\\esp\\\\xtensa-esp32-elf\\\\esp-2022r1-11.2.0\\\\xtensa-esp32-elf\\\\bin\",\"progress\":\"0.00%\",\"progressDetail\":\"\"},{\"actual\":\"esp-2022r1-11.2.0\",\"description\":\"Toolchain for Xtensa (ESP32-S2) based on GCC\",\"doesToolExist\":true,\"env\":{},\"expected\":\"esp-2022r1-11.2.0\",\"hashResult\":true,\"hasFailed\":false,\"id\":\"xtensa-esp32s2-elf\",\"name\":\"xtensa-esp32s2-elf\",\"path\":\"c:<HOMEPATH>\\\\esp\\\\xtensa-esp32s2-elf\\\\esp-2022r1-11.2.0\\\\xtensa-esp32s2-elf\\\\bin\",\"progress\":\"0.00%\",\"progressDetail\":\"\"},{\"actual\":\"esp-2022r1-11.2.0\",\"description\":\"Toolchain for Xtensa (ESP32-S3) based on GCC\",\"doesToolExist\":true,\"env\":{},\"expected\":\"esp-2022r1-11.2.0\",\"hashResult\":true,\"hasFailed\":false,\"id\":\"xtensa-esp32s3-elf\",\"name\":\"xtensa-esp32s3-elf\",\"path\":\"c:<HOMEPATH>\\\\esp\\\\xtensa-esp32s3-elf\\\\esp-2022r1-11.2.0\\\\xtensa-esp32s3-elf\\\\bin\",\"progress\":\"0.00%\",\"progressDetail\":\"\"},{\"actual\":\"esp-2022r1-11.2.0\",\"description\":\"Toolchain for 32-bit RISC-V based on GCC\",\"doesToolExist\":true,\"env\":{},\"expected\":\"esp-2022r1-11.2.0\",\"hashResult\":true,\"hasFailed\":false,\"id\":\"riscv32-esp-elf\",\"name\":\"riscv32-esp-elf\",\"path\":\"c:<HOMEPATH>\\\\esp\\\\riscv32-esp-elf\\\\esp-2022r1-11.2.0\\\\riscv32-esp-elf\\\\bin\",\"progress\":\"0.00%\",\"progressDetail\":\"\"},{\"actual\":\"2.35_20220830\",\"description\":\"Toolchain for ESP32 ULP coprocessor\",\"doesToolExist\":true,\"env\":{},\"expected\":\"2.35_20220830\",\"hashResult\":true,\"hasFailed\":false,\"id\":\"esp32ulp-elf\",\"name\":\"esp32ulp-elf\",\"path\":\"c:<HOMEPATH>\\\\esp\\\\esp32ulp-elf\\\\2.35_20220830\\\\esp32ulp-elf\\\\bin\",\"progress\":\"0.00%\",\"progressDetail\":\"\"},{\"actual\":\"3.24.0\",\"description\":\"CMake build system\",\"doesToolExist\":true,\"env\":{},\"expected\":\"3.24.0,3.16.3\",\"hashResult\":true,\"hasFailed\":false,\"id\":\"cmake\",\"name\":\"cmake\",\"path\":\"c:<HOMEPATH>\\\\esp\\\\cmake\\\\3.24.0\\\\bin\",\"progress\":\"0.00%\",\"progressDetail\":\"\"},{\"actual\":\"v0.11.0-esp32-20221026\",\"description\":\"OpenOCD for ESP32\",\"doesToolExist\":true,\"env\":{\"OPENOCD_SCRIPTS\":\"c:<HOMEPATH>\\\\esp\\\\ope [... cut ...]
Error: unable to open f
[OpenOCD]
tdi device with vid 0403, pid 6010, description '*', serial '*' at bus location '*'

[OpenOCD]
❌ Error: no device found
Error: unable to
[OpenOCD]
 open ftdi device with vid 0403, pid 6014, description '*', serial '*' at bus location '*'

[OpenOCD]

[OpenOCD]
[Stopped] : OpenOCD Server
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2023-07-03:
Have you configured the openOCD config files ? Your doctor command output is incomplete.

Check the `ESP-IDF: Select OpenOCD Board Configuration` command and choose your board. Make sure the connection to your device is correct.

woshiwstt50123, 2023-07-03:
[image]
这个是我能选择的所有选项，目前我选择的是“esp32 chip(via ESP USB bridge)”。从电脑的设备管理器上可以看到esp32确实已经连接到电脑上了，对应的端口是Silicon Labs CP210x USB to UART Bridge。

brianignacio5, 2023-07-03:
The devkitC doesn't have  USB-to-JTAG bridge. You need and external JTAG as described in [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/configure-other-jtag.html)

You can also check the [openOCD troubleshooting page](https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ)

woshiwstt50123, 2023-07-03:
CMSIS-DAP可以代替external JTAG吗

woshiwstt50123, 2023-07-06:
这个问题已经解决了。一般的jtag仿真器好像确实有问题，需要新的仿真器。

closed 2023-07-06 by woshiwstt50123