espressif/vscode-esp-idf-extension #1005: Unable to launch debugging: "no module esp_debug_backend"

crNewton, 2023-07-06:
### OS

Windows

### Operating System version

Windows 10

### Visual Studio Code version

1.79.2

### ESP-IDF version

5.0.2

### Python version

3.8

### Doctor command output

output AFTER I fixed it:

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19045 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Program Files\Microsoft\jdk-11.0.16.101-hotspot\bin;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\ProgramData\chocolatey\bin;C:\Program Files\Git\cmd;C:<HOMEPATH>\AppData\Local\Programs\Python\Python311\;C:<HOMEPATH>\AppData\Local\Programs\Python\Python311\Scripts\;C:\Program Files\dotnet\;C:\Program Files\Docker\Docker\resources\bin;C:\Program Files\Cppcheck;C:\Program Files\Graphviz\bin;C;C:\Program Files\LLVM\bin;C:<HOMEPATH>\AppData\Local\Programs\Python\Python311\Scripts\;C:<HOMEPATH>\AppData\Local\Programs\Python\Python311\;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin;C:<HOMEPATH>\AppData\Local\Microsoft\WinGet\Links;C:<HOMEPATH>\.dotnet\tools 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.79.2 
Visual Studio Code language en 
Visual Studio Code shell C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.6.4 
Workspace folder c:<HOMEPATH>\Documents\_MyDocuments_\repos\RP-HV_PLAYGROUND\RP-HV\hello_world 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:/Espressif/frameworks/esp-idf-v5.0.2/
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) c:\Espressif\tools\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin;c:\Espressif\tools\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin;c:\Espressif\tools\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin;c:\Espressif\tools\tools\xtensa-esp32s2-elf\esp-2022r1-11.2.0\xtensa-esp32s2-elf\bin;c:\Espressif\tools\tools\xtensa-esp32s3-elf\esp-2022r1-11.2.0\xtensa-esp32s3-elf\bin;c:\Espressif\tools\tools\riscv32-esp-elf\esp-2022r1-11.2.0\riscv32-esp-elf\bin;c:\Espressif\tools\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;c:\Espressif\tools\tools\cmake\3.24.0\bin;c:\Espressif\tools\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\bin;c:\Espressif\tools\tools\ninja\1.10.2;c:\Espressif\tools\tools\idf-exe\1.0.3;c:\Espressif\tools\tools\ccache\4.6.2\ccache-4.6.2-windows-x86_64;c:\Espressif\tools\tools\dfu-util\0.9\dfu-util-0.9-win64;c:\Espressif\tools\tools\esp-rom-elfs\20220823
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: c:\Espressif\tools\tools\openocd-esp32\v0.11.0-esp32-20221026/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
    ESP_ROM_ELF_DIR: c:\Espressif\tools\tools\esp-rom-elfs\20220823/
Virtual env Python Path (idf.pythonBinPath) c:\Espressif\tools\python_env\idf5.0_py3.8_env\Scripts\python.exe
Serial port (idf.port) COM20
OpenOCD Configs (idf.openOcdConfigs) board/esp32s3-builtin.cfg
ESP-IDF Tools Path (idf.toolsPath) c:\Espressif\tools
Git Path (idf.gitPath) c:\Espressif\tools\tools\idf-git\2.30.1\cmd\git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to c:\Espressif\tools\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin: true
Access to c:\Espressif\tools\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin: true
Access to c:\Espressif\tools\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin: true
Access to c:\Espressif\tools\tools\xtensa-esp32s2-elf\esp-2022r1-11.2.0\xtensa-esp32s2-elf\bin: true
Access to c:\Espressif\tools\tools\xtensa-esp32s3-elf\esp-2022r1-11.2.0\xtensa-esp32s3-elf\bin: true
Access to c:\Espressif\tools\tools\riscv32-esp-elf\esp-2022r1-11.2.0\riscv32-esp-elf\bin: true
Access to c:\Espressif\tools\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin: true
Access to c:\Espressif\tools\tools\cmake\3.24.0\bin: true
Access to c:\Espressif\tools\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\bin: true
Access to c:\Espressif\tools\tools\ninja\1.10.2: true
Access to c:\Espressif\tools\tools\idf-exe\1.0.3: true
Access to c:\Espressif\tools\tools\ccache\4.6.2\ccache-4.6.2-windows-x86_64: true
[... 124 lines cut ...]
                "limitSymbolsToIncludedHeaders": false
            }
        }
    ],
    "version": 4
}
 
--------------------------------------------------------------------------------------------------------------------------------------------

### Description

Unable to launch the debugger.
OpenOCD can be started, but launching the debugger results in the error:

"no module esp_debug_backend"

### Debug Message

```plain
"no module esp_debug_backend"
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

crNewton, 2023-07-06:
Solution:

Manually install requirements.txt from the venv, as described [here](https://www.esp32.com/viewtopic.php?t=15642):

`python -m pip install -r c:/......./vscode-esp-idf-extension/esp_debug_adapter/requirements.txt`

closed 2023-07-06 by crNewton

brianignacio5, 2023-07-10:
I'm not sure what was the error because you can observe in the doctor command that `esp-debug-backend version: 1.0.3` is installed and debug adapter requirements are validated. Maybe the project had a different `idf.pythonBinPath` ?