espressif/vscode-esp-idf-extension #1006: debugger failed with ModuleNotFoundError: No module named 'esp_debug_backend' on extension verison 1.6.4

emaayan, 2023-07-07:
### OS

Windows

### Operating System version

windows 10

### Visual Studio Code version

1.80.0

### ESP-IDF version

1.6.4

### Python version

embedded python

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19045 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\NVIDIA Corporation\NVIDIA NGX;C:\Program Files\Intel\WiFi\bin\;C:\Program Files\Common Files\Intel\WirelessCommon\;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\dev\nodejs\;C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;C:\Program Files\Microsoft SQL Server\Client SDK\ODBC\170\Tools\Binn\;c:<HOMEPATH>\.platformio\;c:<HOMEPATH>\.platformio\penv\Scripts\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files (x86)\PuTTY\;C:\Program Files\Git\cmd;C:\Program Files\dotnet\;C:\Program Files\Docker\Docker\resources\bin;C:\msys64\mingw64\bin\;C:\Program Files\Cppcheck\;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:\Program Files\Intel\WiFi\bin\;C:\Program Files\Common Files\Intel\WirelessCommon\;C:<HOMEPATH>\AppData\Local\GitHubDesktop\bin;C:<HOMEPATH>\AppData\Roaming\npm;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:\msys64\mingw64\bin\;C:\Program Files (x86)\Nmap;C:<HOMEPATH>\AppData\Roaming\Python\Python37\Scripts 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.80.0 
Visual Studio Code language en 
Visual Studio Code shell C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.6.4 
Workspace folder c:<HOMEPATH>\Documents\PlatformIO\Projects\WifiSniffer\Esp32Sniffer 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:<HOMEPATH>\esp\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32s2-elf\esp-2022r1-11.2.0\xtensa-esp32s2-elf\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32s3-elf\esp-2022r1-11.2.0\xtensa-esp32s3-elf\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-2022r1-11.2.0\riscv32-esp-elf\bin;C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:<HOMEPATH>\.espressif\tools\cmake\3.24.0\bin;C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\bin;C:<HOMEPATH>\.espressif\tools\ninja\1.10.2;C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3;C:<HOMEPATH>\.espressif\tools\ccache\4.6.2\ccache-4.6.2-windows-x86_64;C:<HOMEPATH>\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64;C:<HOMEPATH>\.espressif\tools\esp-rom-elfs\20220823
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.11.0-esp32-20221026/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
    ESP_ROM_ELF_DIR: C:<HOMEPATH>\.espressif\tools\esp-rom-elfs\20220823/
Virtual env Python Path (idf.pythonBinPath) C:<HOMEPATH>\.espressif\python_env\idf5.0_py3.8_env\Scripts\python.exe
Serial port (idf.port) COM7
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,target/esp32.cfg
ESP-IDF Tools Path (idf.toolsPath) C:<HOMEPATH>\.espressif
Git Path (idf.gitPath) C:<HOMEPATH>\.espressif\tools\idf-git\2.30.1\cmd\git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32s2-elf\esp-2022r1-11.2.0\xtensa-esp32s2-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32s3-elf\esp-2022r1-11.2.0\xtensa-esp32s3-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-2022r1-11.2.0\riscv32-esp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\cmake\3.24.0\bin: true
Access to C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\bin: true
Access to C:<HOMEPATH>\.espressif\tools\ninja\1.10.2: true
Access to C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3: true
Access to C:<HOMEPATH>\.espressif\tools\ccache\4.6.2\ccache-4.6.2-windows-x86_64: true
Access to C:<HOMEPATH>\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64: true
Access to C:<HOMEPATH>\.espressif\tools\esp-rom-elfs\20220823: true
[... 107 lines cut ...]
Open On-Chip Debugger v0.11.0-esp32-20221026 (2022-10-26-14:48)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
User : 3 4 options.c:52 configuration_output_handler(): debug_level: 3
User : 4 4 options.c:52 configuration_output_handler(): 
Debug: 5 4 options.c:233 add_default_dirs(): bindir=/builds/idf/openocd-esp32/_build/../openocd-esp32/bin
Debug: 6 4 options.c:234 add_default_dirs(): pkgdatadir=/builds/idf/openocd-esp32/_build/../openocd-esp32/share/openocd
Debug: 7 4 options.c:235 add_default_dirs(): exepath=C:/Users/User/.espressif/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/bin
Debug: 8 4 options.c:236 add_default_dirs(): bin2data=../share/openocd
Debug: 9 4 configuration.c:33 add_script_search_dir(): adding C:\Users\User\.espressif\tools\openocd-esp32\v0.11.0-esp32-20221026/openocd-esp32/share/openocd/scripts
Debug: 10 5 configuration.c:33 add_script_search_dir(): adding C:/Users/User/AppData/Roaming/OpenOCD
Debug: 11 5 configuration.c:33 add_script_search_dir(): adding C:/Users/User/.espressif/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/bin/../share/openocd/site
Debug: 12 5 configuration.c:33 add_script_search_dir(): adding C:/Users/User/.espressif/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/bin/../share/openocd/scripts
Debug: 13 5 command.c:155 script_debug(): command - ocd_find interface/ftdi/esp32_devkitj_v1.cfg
Debug: 14 6 configuration.c:88 find_file(): found C:\Users\User\.espressif\tools\openocd-esp32\v0.11.0-esp32-20221026/openocd-esp32/share/openocd/scripts/interface/ftdi/esp32_devkitj_v1.cfg
```

### Other Steps to Reproduce

i've tried downgrading to 1.6.3 version, and it doesn't happen there. 

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2023-07-10:
Try installing the debug adapter requirements again with the `ESP-IDF: Install ESP-IDF Python Packages` command. The debug adapter was refactored and some part is now added as debug adapter python package (esp_debug_backend)

brianignacio5, 2023-07-21:
Closing this issue since the solution is proposed. Comment here if still have problems about this.

closed 2023-07-21 by brianignacio5

ashelman-ec, 2024-03-04:
Note: I experienced this `No module named 'esp_debug_backend'` error and the solution worked for me. Thanks!