espressif/vscode-esp-idf-extension #1008: identifier "CONFIG_IDF_TARGET" is undefined和identifier "CONFIG_FREERTOS_HZ" is undefined

lomon-emerson, 2023-07-10:
### OS

MacOS

### Operating System version

macOS 13.2.1

### Visual Studio Code version

1.79.2

### ESP-IDF version

5.1

### Python version

3.10.9

### Doctor command output

你好，我是mac电脑安装了vscode和esp-idf。为什么显示identifier "CONFIG_IDF_TARGET" is undefined和identifier "CONFIG_FREERTOS_HZ" is undefined

### Extension

null

### Description

你好，我是mac电脑安装了vscode和esp-idf。为什么显示identifier "CONFIG_IDF_TARGET" is undefined和identifier "CONFIG_FREERTOS_HZ" is undefined

### Debug Message

```plain
你好，我是mac电脑安装了vscode和esp-idf。为什么显示identifier "CONFIG_IDF_TARGET" is undefined和identifier "CONFIG_FREERTOS_HZ" is undefined
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2023-07-10:
We need more information to reproduce your issue. How are you getting this message?

Can you show some steps to reproduce and share the required troubleshooting information ?

lomon-emerson, 2023-07-10:
mac,s3-box，install vscode and esp-idf，use example project"hello world”，at hello_world_main.c：

problems:
identifier "CONFIG_IDF_TARGET" is undefined
identifier "CONFIG_FREERTOS_HZ" is undefined

[image]

ertekinsener, 2023-07-10:
i got the same error in blink example
[image]

brianignacio5, 2023-07-11:
Have you configure the c_cpp_properties.json in your project ? Check [C/C++ Configuration](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/C_CPP_CONFIGURATION.md)

zenz, 2023-07-20:
I have the same issue with ESP-IDF 5.0

and my c_cpp_properties.json is as below:

```
{
    "configurations": [
        {
            "name": "ESP-IDF",
            "compilerPath": "/Users/simon/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc",
            "includePath": [
                "${config:idf.espIdfPath}/components/**",
                "${config:idf.espIdfPathWin}/components/**",
                "${config:idf.espAdfPath}/components/**",
                "${config:idf.espAdfPathWin}/components/**",
                "${workspaceFolder}/**"
            ],
            "browse": {
                "path": [
                    "${config:idf.espIdfPath}/components",
                    "${config:idf.espIdfPathWin}/components",
                    "${config:idf.espAdfPath}/components/**",
                    "${config:idf.espAdfPathWin}/components/**",
                    "${workspaceFolder}"
                ],
                "limitSymbolsToIncludedHeaders": false
            }
        }
    ],
    "version": 4
}

```

brianignacio5, 2023-07-21:
Try to use the `"compileCommands": "${workspaceFolder}/build/compile_commands.json"` which gets the values from the current build files.

zenz, 2023-07-21:
> Try to use the `"compileCommands": "${workspaceFolder}/build/compile_commands.json"` which gets the values from the current build files.

This worked. but it's so weird. need to compile first.

ugurakas, 2023-07-24:
When ı paste  "compileCommands": "${workspaceFolder}/build/compile_commands.json" its gives me a different types of error
[image]

brianignacio5, 2023-07-28:
Check your source code @ugurakas Seems the some functions references are not defined or you are referring to older esp-idf versions.

Closing this issue for lack of response from original user and solution seems to apply. Comment here if any additional issues.

closed 2023-07-28 by brianignacio5

lnsriravichandran, 2023-09-09:
> Try to use the `"compileCommands": "${workspaceFolder}/build/compile_commands.json"` which gets the values from the current build files.

For this the project has to be built says the documentation. This has to be fixed for successful build isn't that right?

minnyqiu, 2023-09-16:
> Try to use the `"compileCommands": "${workspaceFolder}/build/compile_commands.json"` which gets the values from the current build files.

I met the same problem on macOS, is that because I both installed ESP-IDF and VS Code ESP-IDF extension?

brianignacio5, 2023-09-18:
You can build the project even with the error. @lnsriravichandran The value is generated by cmake but the C/C++ Configuration doesn't know where to find the symbol.

Is not related to this @minnyqiu . The thing is that the C/C++ extension can't find the symbol. Need to update the c_cpp_properties.json and rebuild the database or use the compile_commands.json approach.

Aditi0305, 2023-09-22:
Facing the same issue with ESP-IDF version 5.1.1 on windows.

FranLanderreche, 2023-11-02:
I had the same issue;

Fixed it by modifying this:
```
"path": [
                    "${config:idf.espIdfPath}/components",
                    "${config:idf.espIdfPathWin}/components",
                    "${config:idf.espAdfPath}/components/**",
                    "${config:idf.espAdfPathWin}/components/**",
                    "${workspaceFolder}"
                ],
```

for this (deleted the two first lines):
```
"path": [
                    "${config:idf.espAdfPath}/components/**",
                    "${config:idf.espAdfPathWin}/components/**",
                    "${workspaceFolder}"
                ],
```

I believe is something to do with the path configuration. Not sure how to proper fix that, so have to modified this in every project.

k-kipruto, 2023-11-02:
Add this line to the c_cpp_properties.json 

`"compileCommands": "${workspaceFolder}/build/compile_commands.json",`

[image]

wh-2691, 2023-12-21:
Same problem, and I added `"compileCommands": "${workspaceFolder}/build/compile_commands.json",` into c_cpp_properties.json, but it turns out that I do not have this file.

Cannot find "c:\Users\USER\Documents\ESP32-example\Internet-of-Things-Projects-with-ESP32-master\Internet-of-Things-Projects-with-ESP32-master\Chapter02\dhtdemo\build\compile_commands.json".

How could I get this file?

brianignacio5, 2023-12-22:
Build the project with `ESP-IDF: Build your project`

krupis, 2024-01-02:
I had the same issue. I have fixed it by adding 
``` "compileCommands": "${workspaceFolder}/build/compile_commands.json", ```

To the c_cpp_properties.json as @k-kipruto have suggested

Seefer, 2024-01-16:
Is this an example of a C/C++ IntelliSense configuration problem and should the ESP-IDF example code downloaded during ESP-IDF extension installation and tooling setup not already have `"compileCommands": "${workspaceFolder}/build/compile_commands.json",` set in the source code?