espressif/vscode-esp-idf-extension #1011: [4.4.5] Ninja not found on incremental compilation

KonssnoK, 2023-07-17:
### OS

Windows

### Operating System version

Win11

### Visual Studio Code version

1.80.1

### ESP-IDF version

4.4.5

### Python version

3.11.2

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.22621 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Programmi86\VMware\VMware Player\bin\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Programmi\PuTTY\;C:\Program Files\WireGuard\;C:\cygwin64\bin;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\dotnet\;C:\Programmi\TortoiseSVN\bin;C:\Program Files\Docker\Docker\resources\bin;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\AppData\Local\gitkraken\bin;C:\Programmi86\Microsoft Visual Studio\2022\Community\Common7\IDE\CommonExtensions\Microsoft\TeamFoundation\Team Explorer\Git\mingw64\bin;C:\Programmi\Microsoft VS Code\bin;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:\src\tiko\srecord-1.64_cyg\bin\;C:<HOMEPATH>\.dotnet\tools;C:<HOMEPATH>\AppData\Roaming\Python\Python311\Scripts; 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.80.1 
Visual Studio Code language en 
Visual Studio Code shell C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.6.4 
Workspace folder c:\src\v3\firmware 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:/src/esp-idf/
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) C:\src\espressif\.espressif\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin;C:\src\espressif\.espressif\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin;C:\src\espressif\.espressif\tools\xtensa-esp32-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\bin;C:\src\espressif\.espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s2-elf\bin;C:\src\espressif\.espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s3-elf\bin;C:\src\espressif\.espressif\tools\riscv32-esp-elf\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\bin;C:\src\espressif\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:\src\espressif\.espressif\tools\cmake\3.23.1\bin;C:\src\espressif\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230419\openocd-esp32\bin;C:\src\espressif\.espressif\tools\ninja\1.10.2;C:\src\espressif\.espressif\tools\idf-exe\1.0.3;C:\src\espressif\.espressif\tools\ccache\4.3\ccache-4.3-windows-64;C:\src\espressif\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: C:\src\espressif\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230419/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
Virtual env Python Path (idf.pythonBinPath) C:\src\espressif\.espressif\python_env\idf4.4_py3.11_env\Scripts\python.exe
Serial port (idf.port) COM10
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,target/esp32s3.cfg
ESP-IDF Tools Path (idf.toolsPath) C:/src/espressif/.espressif
Git Path (idf.gitPath) C:\src\espressif\.espressif\tools\idf-git\2.39.2\cmd\git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to C:\src\espressif\.espressif\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin: true
Access to C:\src\espressif\.espressif\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin: true
Access to C:\src\espressif\.espressif\tools\xtensa-esp32-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\bin: true
Access to C:\src\espressif\.espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s2-elf\bin: true
Access to C:\src\espressif\.espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s3-elf\bin: true
Access to C:\src\espressif\.espressif\tools\riscv32-esp-elf\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\bin: true
Access to C:\src\espressif\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin: true
Access to C:\src\espressif\.espressif\tools\cmake\3.23.1\bin: true
Access to C:\src\espressif\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230419\openocd-esp32\bin: true
Access to C:\src\espressif\.espressif\tools\ninja\1.10.2: true
Access to C:\src\espressif\.espressif\tools\idf-exe\1.0.3: true
Access to C:\src\espressif\.espressif\tools\ccache\4.3\ccache-4.3-windows-64: true
Access to C:\src\espressif\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH undefined
[... 140 lines cut ...]

### Description

if i do:
- Full clean project
- Build your project -> ALL ok
- Close VSCode
- Open VSCode
- Build your project

The second time i get:

### Debug Message

```plain
-- Build files have been written to: C:/src/v3/firmware/build
ninja: error: build.ninja:30: loading 'CMakeFiles/rules.ninja': The system cannot find the file specified.

include CMakeFiles/rules.ninja
                              ^ near here
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

radurentea, 2023-07-19:
Hi @KonssnoK,

It seems the issue comes from the compiler configuration which you can find in .vscode/c_cpp_properties.json . First thing that pops is that you don't have a compilerPath property in your json, which should link to the executable. Here is an example:
`"compilerPath": "d:\\ESPRESSIF\\tools\\xtensa-esp32s2-elf\\esp-12.2.0_20230208\\xtensa-esp32s2-elf\\bin\\xtensa-esp32s2-elf-gcc.exe",`

I suggest to have a look over this part of the [documentation](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/C_CPP_CONFIGURATION.md) which shows a default configuration for a project

KonssnoK, 2023-07-20:
> Hi @KonssnoK,
>
> It seems the issue comes from the compiler configuration which you can find in .vscode/c_cpp_properties.json . First thing that pops is that you don't have a compilerPath property in [...]

Hi @radurentea ,
how can i keep in sync with esp-idf installations in i have to put manually the compiler in compilerPath?
Depending on the commit of esp-idf i have used 
- esp-2021r2-8.4.0
- esp-2021r2-patch3-8.4.0
- esp-2021r2-patch3-8.5.0
- esp-2022r1

radurentea, 2023-07-20:
If you install another esp-idf version and you configure the extension. You could use the **`ESP-IDF: Add vscode configuration folder`** which will update the settings and the compilerPath.

KonssnoK, 2023-07-20:
oh ok, thanks

KonssnoK, 2023-07-20:
```
-- Configuring done
-- Generating done
-- Build files have been written to: C:/src/v3/firmware/build
ninja: error: build.ninja:30: loading 'CMakeFiles/rules.ninja': The system cannot find the file specified.

include CMakeFiles/rules.ninja
                              ^ near here

 *  The termina
```

The issue is still there, my file now is 

```
{
    "configurations": [
        {
            "name": "esp32s3",
            "includePath": [
                "${workspaceFolder}/**",
                "${config:idf.espIdfPath}/components/**",
                "${config:idf.espIdfPathWin}/components/**",
                "C:/src/v3/firmware/**",
                "C:/src/v3/esp-protocols/components/esp_modem/**"
            ],
            "browse": {
                "path": [
                    "${workspaceFolder}",
                    "${config:idf.espIdfPath}/components",
                    "${config:idf.espIdfPathWin}/components",
                    "C:/src/v3/esp-protocols/components/esp_modem"
                ],
                "limitSymbolsToIncludedHeaders": false
            },
            "defines": [
                "_DEBUG",
                "UNICODE",
                "_UNICODE"
            ],
            "windowsSdkVersion": "10.0.19041.0",
            "cStandard": "c17",
            "cppStandard": "c++17",
            "intelliSenseMode": "windows-gcc-x86",
            "configurationProvider": "ms-vscode.makefile-tools",
            "compilerPath": "C:\\src\\espressif\\.espressif\\tools\\xtensa-esp32s3-elf\\esp-2022r1\\xtensa-esp32s3-elf\\bin\\xtensa-esp32s3-elf-gcc.exe"
        }
    ],
    "version": 4
}
```

KonssnoK, 2023-07-21:
I changed the cmake path,
did a clean + build
closed VSCode,
reopened VSCode
build

It seems that ninja now is found, but the other points are missing..
I would expect those settings to use a relative path to the esp-idf tools folder 🤔 

```
[0/1] Re-running CMake...
CONFIG_FW_STAGING_NVS is enabled
-- Component directory C:/src/esp-idf/components/esp_coex does not contain a CMakeLists.txt file. No component will be added
-- The CXX compiler identification is unknown
-- The ASM compiler identification is unknown
-- Found assembler: C:/cygwin64/bin/cc
CMake Error at C:/src/esp-idf/tools/cmake/project.cmake:381 (__project):
  No CMAKE_CXX_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:30 (project)

-- Warning: Did not find file Compiler/-ASM
-- Configuring incomplete, errors occurred!
See also "C:/src/v3/firmware/build/CMakeFiles/CMakeOutput.log".
See also "C:/src/v3/firmware/build/CMakeFiles/CMakeError.log".
FAILED: build.ninja 
C:\src\espressif\.espressif\tools\cmake\3.23.1\bin\cmake.exe --regenerate-during-build -SC:\src\v3\firmware -BC:\src\v3\firmware\build
ninja: error: rebuilding 'build.ninja': subcommand failed

 *  The terminal process "C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -Command ninja " terminated with exit code: 1. 

```

radurentea, 2023-07-21:
At this point I think it will be easier to make a clean reinstall of the extension, esp-idf and esp-idf tools. Sometimes, errors like this can be caused by a corrupted or incomplete installation.

You can use this link to install the [esp-idf and tools](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/windows-setup.html)

To uninstall the extension, you should follow [these steps](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/INSTALL.md#uninstalling-the-plugin)

For the installation and configuration of the extension you can have a look [here](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/install.md)

KonssnoK, 2023-07-24:
the installation of idf and tools is complete and performed via the esp-idf extension.
Of course it was performed multiple times since we started using the v4.4 branch, moving between multiple compilers.

The compilation works after a full clean, meaning that the issue is only when starting from a cached state.

Having to reinstall everything doesn't seem the best approach 🤔 .
I'll continue looking it out in my free time

radurentea, 2023-07-24:
I see. It's hard for me to reproduce and I haven't run into such issues. But I will keep them in mind and think if there is a way to prevent similar configuration issues.

closed 2023-07-24 by radurentea

brianignacio5, 2023-07-25:
> ```
> -- Configuring done
> -- Generating done
> -- Build files have been written to: C:/src/v3/firmware/build
> ninja: error: build.ninja:30: loading 'CMakeFiles/rules.ninja': The system cannot find the file specified.
> 
> include CMakeFiles/rules.ninja
>                               ^ near here
> 
>  *  The termina
> ```
> 
> 
>      
>   
> 
> The issue is still there, my file now is
> 
> 
> ```
> {
>     "configurations": [
>         {
>             "name": "esp32s3",
>             "includePath": [
>                 "${workspaceFolder}/**",
>                 "${config:idf.espIdfPath}/components/**",
>                 "${config:idf.espIdfPathWin}/components/**",
>                 "C:/src/v3/firmware/**",
>                 "C:/src/v3/esp-protocols/components/esp_modem/**"
>             ],
>             "browse": {
>                 "path": [
>                     "${workspaceFolder}",
>                     "${config:idf.espIdfPath}/components",
>                     "${config:idf.espIdfPathWin}/components",
>                     "C:/src/v3/esp-protocols/components/esp_modem"
>                 ],
>                 "limitSymbolsToIncludedHeaders": false
>             },
>             "defines": [
>                 "_DEBUG",
>                 "UNICODE",
>                 "_UNICODE"
>             ],
>             "windowsSdkVersion": "10.0.19041.0",
>             "cStandard": "c17",
>             "cppStandard": "c++17",
>             "intelliSenseMode": "windows-gcc-x86",
>             "configurationProvider": "ms-vscode.makefile-tools",
>             "compilerPath": "C:\\src\\espressif\\.espressif\\tools\\xtensa-esp32s3-elf\\esp-2022r1\\xtensa-esp32s3-elf\\bin\\xtensa-esp32s3-elf-gcc.exe"
>         }
>     ],
>     "version": 4
> }
> ```
> 
> 
>     
>  
>   
> 
> The output at VSCode open is the following,
> 
> 
> ```
> variant] Loaded new set of variants
> [kit] Successfully loaded 7 kits from C:\Users\NR6085\AppData\Local\CMakeTools\cmake-tools-kits.json
> [proc] Executing command: "C:\Programmi86\Microsoft Visual Studio\2022\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin\cmake.exe" --version
> [proc] The command: ninja --version failed with error: Error: spawn ninja ENOENT
> [proc] The command: ninja-build --version failed with error: Error: spawn ninja-build ENOENT
> [cmakefileapi-driver] Removing c:/src/v3/firmware/build/CMakeCache.txt
> [main] Configuring project: firmware 
> [proc] Executing command: "C:\Programmi86\Microsoft Visual Studio\2022\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin\cmake.exe" --no-warn-unused-cli -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -DCMAKE_BUILD_TYPE:STRING=Debug -SC:/src/v3/firmware -Bc:/src/v3/firmware/build -G "Unix Makefiles"
> [cmake] Not searching for unused variables given on the command line.
> [cmake] CONFIG_FW_STAGING_NVS is enabled
> [cmake] CMake Error at CMakeLists.txt:29 (include):
> [cmake]   include could not find requested file:
> [cmake] 
> [cmake]     /tools/cmake/project.cmake
> [cmake] 
> [cmake] 
> [cmake] -- The C compiler identification is unknown
> [cmake] -- The CXX compiler identification is unknown
> [cmake] -- Detecting C compiler ABI info
> [cmake] -- Detecting C compiler ABI info - done
> [cmake] -- Check for working C compiler: C:/cygwin64/bin/cc - skipped
> [cmake] CMake Error at CMakeLists.txt:30 (project):
> [cmake]   No CMAKE_CXX_COMPILER could be found.
> [cmake] 
> [cmake]   Tell CMake where to find the compiler by setting either the environment
> [cmake]   variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
> [cmake]   to the compiler, or to the compiler name if it is in the PATH.
> [cmake] 
> [cmake] 
> [cmake] -- Configuring incomplete, errors occurred!
> [cmake] See also "C:/src/v3/firmware/build/CMakeFiles/CMakeOutput.log".
> [cmake] See also "C:/src/v3/firmware/build/CMakeFiles/CMakeError.log".
> [proc] The command: "C:\Programmi86\Microsoft Visual Studio\2022\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin\cmake.exe" --no-warn-unused-cli -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -DCMAKE_BUILD_TYPE:STRING=Debug -SC:/src/v3/firmware -Bc:/src/v3/firmware/build -G "Unix Makefiles" exited with code: 1
> ```

Also I've noticed this output is from a different extension build, not this ESP-IDF extension. This is from the CMake extension. When you build, are you using our commands as described in the tutorials ?

KonssnoK, 2023-07-25:
@brianignacio5 , if you do a build without doing a clean before the issue persists.
if you do a clean and then a build it works.