espressif/vscode-esp-idf-extension #1015: ESP-IDF instalation failed on Creating a new Python environment

NightSkySK, 2023-07-26:
### OS

Windows

### Operating System version

Windows 10

### Visual Studio Code version

1.80.1

### ESP-IDF version

4.4.5

### Python version

3.11.4

### Doctor command output

```
pip 22.3.1 from D:\Espressif\tools\tools\idf-python\3.11.2\Lib\site-packages\pip (python 3.11)

Collecting virtualenv
  Using cached virtualenv-20.24.2-py3-none-any.whl (3.0 MB)
Collecting distlib<1,>=0.3.7
  Using cached distlib-0.3.7-py2.py3-none-any.whl (468 kB)
Collecting filelock<4,>=3.12.2
  Using cached filelock-3.12.2-py3-none-any.whl (10 kB)
Collecting platformdirs<4,>=3.9.1
  Using cached platformdirs-3.9.1-py3-none-any.whl (16 kB)
Installing collected packages: distlib, platformdirs, filelock, virtualenv
Successfully installed distlib-0.3.7 filelock-3.12.2 platformdirs-3.9.1 virtualenv-20.24.2
Installing virtualenv
Creating a new Python environment in D:\Espressif\tools\python_env\idf4.4_py3.11_env

[notice] A new release of pip available: 22.3.1 -> 23.2.1
[notice] To update, run: D:\Espressif\tools\tools\idf-python\3.11.2\python.exe -m pip install --upgrade pip
D:\Espressif\tools\tools\idf-python\3.11.2\python.exe: No module named virtualenv
Traceback (most recent call last):
  File "D:\Espressif\esp-idf\tools\idf_tools.py", line 1976, in <module>
    main(sys.argv[1:])
  File "D:\Espressif\esp-idf\tools\idf_tools.py", line 1972, in main
    action_func(args)
  File "D:\Espressif\esp-idf\tools\idf_tools.py", line 1598, in action_install_python_env
    subprocess.check_call([sys.executable, '-m', 'virtualenv',
  File "subprocess.py", line 413, in check_call
subprocess.CalledProcessError: Command '['D:\\Espressif\\tools\\tools\\idf-python\\3.11.2\\python.exe', '-m', 'virtualenv', '--python', 'D:\\Espressif\\tools\\tools\\idf-python\\3.11.2\\python.exe', '--seeder', 'pip', 'D:\\Espressif\\tools\\python_env\\idf4.4_py3.11_env']' returned non-zero exit status 1.
Command failed: "D:\Espressif\tools\tools\idf-python\3.11.2\python.exe" "D:\Espressif\esp-idf\tools\idf_tools.py" install-python-env

[notice] A new release of pip available: 22.3.1 -> 23.2.1
[notice] To update, run: D:\Espressif\tools\tools\idf-python\3.11.2\python.exe -m pip install --upgrade pip
D:\Espressif\tools\tools\idf-python\3.11.2\python.exe: No module named virtualenv
Traceback (most recent call last):
  File "D:\Espressif\esp-idf\tools\idf_tools.py", line 1976, in <module>
    main(sys.argv[1:])
  File "D:\Espressif\esp-idf\tools\idf_tools.py", line 1972, in main
    action_func(args)
  File "D:\Espressif\esp-idf\tools\idf_tools.py", line 1598, in action_install_python_env
    subprocess.check_call([sys.executable, '-m', 'virtualenv',
  File "subprocess.py", line 413, in check_call
subprocess.CalledProcessError: Command '['D:\\Espressif\\tools\\tools\\idf-python\\3.11.2\\python.exe', '-m', 'virtualenv', '--python', 'D:\\Espressif\\tools\\tools\\idf-python\\3.11.2\\python.exe', '--seeder', 'pip', 'D:\\Espressif\\tools\\python_env\\idf4.4_py3.11_env']' returned non-zero exit status 1.
Command failed: "D:\Espressif\tools\tools\idf-python\3.11.2\python.exe" "D:\Espressif\esp-idf\tools\idf_tools.py" install-python-env

[notice] A new release of pip available: 22.3.1 -> 23.2.1
[notice] To update, run: D:\Espressif\tools\tools\idf-python\3.11.2\python.exe -m pip install --upgrade pip
D:\Espressif\tools\tools\idf-python\3.11.2\python.exe: No module named virtualenv
Traceback (most recent call last):
  File "D:\Espressif\esp-idf\tools\idf_tools.py", line 1976, in <module>
    main(sys.argv[1:])
  File "D:\Espressif\esp-idf\tools\idf_tools.py", line 1972, in main
    action_func(args)
  File "D:\Espressif\esp-idf\tools\idf_tools.py", line 1598, in action_install_python_env
    subprocess.check_call([sys.executable, '-m', 'virtualenv',
  File "subprocess.py", line 413, in check_call
subprocess.CalledProcessError: Command '['D:\\Espressif\\tools\\tools\\idf-python\\3.11.2\\python.exe', '-m', 'virtualenv', '--python', 'D:\\Espressif\\tools\\tools\\idf-python\\3.11.2\\python.exe', '--seeder', 'pip', 'D:\\Espressif\\tools\\python_env\\idf4.4_py3.11_env']' returned non-zero exit status 1.

```
[consol.log](https://github.com/espressif/vscode-esp-idf-extension/files/12172370/consol.log)

### Extension

[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/12172393/esp_idf_vsc_ext.log)

### Description

I've tried to install ESP-IDF using esp-idf-extension-1.6.4 in VSCode with version v4.3.5 and v4.4.5 and always get the same issue with creation of virtual environment. However once I copy and paste it to terminal window I can create this virtualenv using command `D:\\Espressif\\tools\\tools\\idf-python\\3.11.2\\python.exe -m virtualenv --python D:\\Espressif\\tools\\tools\\idf-python\\3.11.2\\python.exe --seeder pip D:\\Espressif\\tools\\python_env\\idf4.4_py3.11_env`
I've even changed the default integrated terminal from bash to cmd (as I thought that might be a problem) but it still the same

### Debug Message

```plain
N/A
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

NightSkySK, 2023-07-27:
To make full view, I've been able to create the virtualenv manually from terminal, and after reboot of computer I've tried to recover installation using Global --> Use existing setup --> Search ESP-IDF in system. And here is the log captured from terminal in this process.
[consol_2.log](https://github.com/espressif/vscode-esp-idf-extension/files/12179652/consol_2.log)

policeman0077, 2023-07-28:
can confirm I have same problem. seem they changed how to they setup the virtual environment. So the latest esp-idf tools(idf-python) is not compatible with v4.4.x idf

brianignacio5, 2023-07-28:
The problem is the use of the new python version works well for v5.0 or higher but not the other versions and we still need to add virtualenv because it was used before.

I'll add a PR for this.

policeman0077, 2023-07-28:
> D:\\Espressif\\tools\\tools\\idf-python\\3.11.2\\python.exe -m virtualenv --python D:\\Espressif\\tools\\tools\\idf-python\\3.11.2\\python.exe --seeder pip D:\\Espressif\\tools\\python_env\\idf4.4_py3.11_env

After manually type this `D:\\Espressif\\tools\\tools\\idf-python\\3.11.2\\python.exe -m virtualenv --python D:\\Espressif\\tools\\tools\\idf-python\\3.11.2\\python.exe --seeder pip D:\\Espressif\\tools\\python_env\\idf4.4_py3.11_env` in terminal. Config the extension using existing ESP-IDF, it worked for me. Thank you.

mitchjs, 2023-07-29:
same here, and same doing the command in terminal worked
@brianignacio5 are you saying if the extension used python 3.8 (like my older virtual environments) it would have worked?
but does NOTt explain that we can get the virtual environment created with the command line
so to me i dont think its python 3.11

BUT you know the folder
:\Users\mitchjs\.espressif\tools\idf-python\3.11.2\Scripts" DOES NOT CONTAIN "virtualenv.exe"
so i guess its not included in python 3.11?
but then how did the manual command (ubove post) work?
did it search my installed python scripts folder, i bet it did
since i have python 3.9 installed for other stuff

oh because it fails during offical config, i dont get my settings.json updated
but rerunning the configuration tool, does save it 
since all tools and idf and virenvironment is set up, its quick

thanks
Mitch

HendrikSchmi, 2023-08-04:
Hello all, 
i also had this problem and solved it by working with an older esp-idf extension in the vs code.
I downloaded version 1.6.1 manually at:
https://github.com/espressif/vscode-esp-idf-extension/tags

And then manually installed the .vsix file in the extensions.

Then old esp-idf versions also run.

Many greetings
Hendrik

noqnio, 2023-08-04:
> Hello all, i also had this problem and solved it by working with an older esp-idf extension in the vs code. I downloaded version 1.6.1 manually at: [...]

Thank you, this worked for me too. You can also directly downgrade through VSCode by right clicking on the extension in the extensions tab

brianignacio5, 2023-08-04:
Added the fix in #1021 You can try the vsix installer artifacts from this pull request.

closed 2023-08-24 by brianignacio5

gonzabrusco, 2023-08-29:
Using that vsix installer I got the following when trying to install 4.4.4 on this folder: `c:\Espressif\4.4.4\`

`c:\Espressif\4.4.4\tools\idf-python\3.8.7\python.exe: No module named ensurepip`

gonzabrusco, 2023-08-29:
I'm still struggling with this bug. It seems that Python is installing every package (that includes virtualenv) on a user folder (in my case `C:\Users\Gonzalo\AppData\Roaming\Python\Python311`). It should be installing everything on the idf python folder (in my case `C:\Espressif\4.4.4\tools\idf-python`). Then when it wants to install the virtualenv, it does not find it.

I'm on Windows 10.

gonzabrusco, 2023-08-29:
I can confirm that installing the older version (1.6.3) worked. It installed fine but when openning VSCode the install wizard was opening again. The solution was to update the settings.json in .vscode folder.

kumarsai588, 2023-08-31:
please anyone help me I am also facing the same issue.

brianignacio5, 2023-09-01:
Hmm is interesting your case @gonzabrusco why is it using the Python311 reference if you are installing the 3.8.7 version ? Or are you using the IDF v5.0  for this? Can you provide any extension log or doctor command output ? See the readme troubleshooting section. Also you @kumarsai588

gonzabrusco, 2023-09-01:
_Murphy's law strikes again!_

I tried again with the vsix installer you attached and it simply worked. I could not make it fail. So I guess I had bad luck, I don't know. If It happens again I will attach the log.

Thanks
Gonzalo

bill-richards, 2023-09-15:
1. Entering the following in a terminal window 
> D:\Espressif\tools\tools\idf-python\3.11.2\python.exe -m virtualenv --python D:\Espressif\tools\tools\idf-python\3.11.2\python.exe --seeder pip D:\Espressif\tools\python_env\idf4.4_py3.11_env

2. Restart Visual Studio Code
3. Continue with ESP-IDF: Configure ESP-IDF extension 

That was the solution for me

> D:\Espressif\tools\tools\idf-python\3.11.2\python.exe -m virtualenv --python D:\Espressif\tools\tools\idf-python\3.11.2\python.exe --seeder pip D:\Espressif\tools\python_env\idf4.4_py3.11_env

Cyfarw9dd, 2023-09-28:
i have the same problem. but i have saw the terminal said that 

> Successfully installed distlib-0.3.7 filelock-3.12.4 platformdirs-3.10.0 virtualenv-20.24.5

. i didnt sure about whether i truely install the virtualenv. the log still said 

> subprocess.CalledProcessError: Command '['d:\\.espressif\\tools\\idf-python\\3.11.2\\python.exe', '-m', 'virtualenv', '--python', 'd:\\.espressif\\tools\\idf-python\\3.11.2\\python.exe', '--seeder', 'pip', 'd:\\.espressif\\python_env\\idf4.3_py3.11_env']' returned non-zero exit status 1.

PR #1021 (2023-08-24): use older embed python for older that idf 5
## Description

Use Embed Python 3.8.7 for ESP-IDF < v5.0 

Fixes #1015

## Type of change

Please delete options that are not relevant.
- Bug fix (non-breaking change which fixes an issue)

## Steps to test this pull request

1. Click on "Configure ESP-IDF extension" command
2. Install ESP-IDF v4.4.5 or older
3. Observe results. Check installed embed Python is 3.8.7 and virtual environment is created.
4. Click on "Configure ESP-IDF extension" command
5. Install ESP-IDF v5.0 or higher
6. Observe results. Check installed embed Python is 3.11.2 and vir [... cut]