espressif/vscode-esp-idf-extension #1017: Problems using espressif libraries in Vscode

FilipeSDias, 2023-07-27:
### OS

Windows

### Operating System version

Windows 10

### Visual Studio Code version

1.80.1

### ESP-IDF version

5.0

### Python version

3.11.4

### Doctor command output

ESP32-WROVER-E

### Description

I configured VSCode with the Espressif IDF extension and I need to compile a code, however, it gives an error in the standard espress libraries and I can't find a way to solve it.

[image]

It's the first time I'm using VSCode to program with ESP32 (I used Arduino IDE before).

Does anyone know how I can fix this library issue? Where can I look for them?

Thanks

### Debug Message

```plain
"#include errors detected. Please update your includePath. Squiggles are disabled for
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2023-07-28:
You need to configure the Microsoft C/C++ extension  as described in the README.md file:

> Configure the .vscode/c_cpp_properties.json as explained in [C/C++ Configuration](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/C_CPP_CONFIGURATION.md). There is a default configuration that should work when you create a new project with the extension commands but you might want to modify it to your needs.

brianignacio5, 2023-08-03:
Closing this issue for lack of response. Comment here for further question.

closed 2023-08-03 by brianignacio5