espressif/vscode-esp-idf-extension #1019: Just stopped working, all commands result in error

mitchjs, 2023-08-02:
### OS

Windows

### Operating System version

11

### Visual Studio Code version

1.80.2

### ESP-IDF version

n/a

### Python version

3.9

### Doctor command output

cannot do dr, command results in errror

### Extension

no log in present

### Description

i have been using this for long long time, even today i was writing code..
and building / flashing you name it
then suddenly NOTHING works
all buttons on task bar result in error, every esp-idf:xxx command results in error
i deleted the extension folder, reinstalled no change all show error

example error
command 'espidf.builddevice' not found
same for doctor command, all commands not found

i thought this was some old issue now fixed
something is wrong
and now this is killing my work

i have NO idea why there is no installation log where the extension is installed
but its not there... 

### Debug Message

```plain
n/a
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mitchjs, 2023-08-02:
more info looking at the vscode extensions host log

2023-08-02 02:04:03.013 [error] Activating extension espressif.esp-idf-extension failed due to an error:
2023-08-02 02:04:03.013 [error] Error: Unable to write into folder settings. Please open the 'BT1026-BT-Sink' folder settings to correct errors/warnings in it and try again.
    at e.$hYb.y (vscode-file://vscode-app/c:/Users/mitchjs/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/workbench.desktop.main.js:2271:45283)
    at e.$hYb.o (vscode-file://vscode-app/c:/Users/mitchjs/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/workbench.desktop.main.js:2271:42678)
    at e.$hYb.n (vscode-file://vscode-app/c:/Users/mitchjs/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/workbench.desktop.main.js:2271:42569)
    at async Object.factory (vscode-file://vscode-app/c:/Users/mitchjs/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/out/vs/workbench/workbench.desktop.main.js:2271:42319)

mitchjs, 2023-08-02:
ok, that folder is open, i closed it and commands now work...
so what is wrong in that folder?

mitchjs, 2023-08-02:
ok, if the error was better... i would have got it 1st time
settings.json, got corrupted entry... no idea how... i dont edit it ever manually
i manually fixed it

would be better if it said:
Please open the 'BT1026-BT-Sink' .**vscode** folder **and check settings.json** to correct errors/warnings in it and try again.

brianignacio5, 2023-08-03:
Hi @mitchjs so this error happens outside the extension, you can see the extension is not even activated.

 Hard to know what to do here. If the settings.json is corrupted, our extension is not even activated so we can't fix anything really.

mitchjs, 2023-08-03:
its kinda activated, no? i see icons on the status bar... the esp-idf: commands show up
did you create the error message in the extensions host log? 

if nothing can be done... thats not good, not your fault... but i can see this happening to others

mitch

radurentea, 2023-08-10:
Hi @mitchjs,

Make sure you uninstall the extension the right way, you can follow the steps [here](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/INSTALL.md#uninstalling-the-plugin) and try to run the configuration from command palette. 

If commands are still not working, maybe reinstall IDF manually as well with [ESP-IDF Windows Installer](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/windows-setup.html#esp-idf-tools-installer)

If that doesn't help and you save your configuration intro User Settings, you can open the User Settings within Command Pallette `Open User Settings(JSON)` or find the file at:

Windows: %APPDATA%\Code\User\settings.json
macOS: ~/Library/Application Support/Code/User/settings.json
Linux: ~/.config/Code/User/settings.json

And there delete all the commands starting with `idf.` and reinstall the extension.

mitchjs, 2023-08-10:
@radurentea, it was a corrupted settings.json
uninstalling didnt make any difference of course since it had nothing to do with the install
this could be closed, i was just hoping something could be done to better inform the user
but it could all be a vs code issue on that

thanks
mitch

closed 2023-08-11 by radurentea