espressif/vscode-esp-idf-extension #1020: CMake Error at run_serial_tool.cmake:66

WelkinDev, 2023-08-03:
### OS

MacOS

### Operating System version

12.6.3 (21G419)

### Visual Studio Code version

Version: 1.80.2 (Universal) Commit: 2ccd690cbff1569e4a83d7c43d45101f817401dc Date: 2023-07-27T21:05:41.366Z Electron: 22.3.14 ElectronBuildId: 22695494 Chromium: 108.0.5359.215 Node.js: 16.17.1 V8: 10.8.168.25-electron.0 OS: Darwin x64 21.6.0

### ESP-IDF version

V5.1

### Python version

Python 3.11.4

### Doctor command output

```bash
/Users/user/esp/esp-idf/tools/check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Executing action: flash
Running ninja in directory /Users/user/Documents/Code/Embedded/Demo_EPS/hello_world/build
Executing "ninja flash"...
[1/5] cd /Users/user/Documents/Code/Embedded/Demo_EPS/hello_world/build/esp-idf/esptool_py && /Users/user/.espressif/python_env/idf5.1_py3.11_env/bin/python /Users/user/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 partition --type app /Users/user/Documents/Code/Embedded/Demo_EPS/hello_world/build/partition_table/partition-table.bin /Users/user/Documents/Code/Embedded/Demo_EPS/hello_world/build/hello_world.bin
hello_world.bin binary size 0x31360 bytes. Smallest app partition is 0x100000 bytes. 0xceca0 bytes (81%) free.
[2/5] Performing build step for 'bootloader'
[1/1] cd /Users/user/Documents/Code/Embedded/Demo_EPS/hello_world/build/bootloader/esp-idf/esptool_py && /Users/user/.espressif/python_env/idf5.1_py3.11_env/bin/python /Users/user/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 /Users/user/Documents/Code/Embedded/Demo_EPS/hello_world/build/bootloader/bootloader.bin
Bootloader binary size 0x5200 bytes. 0x2e00 bytes (36%) free.
[3/5] No install step for 'bootloader'
[4/5] Completed 'bootloader'
[4/5] cd /Users/user/esp/esp-idf/components/esptool_py && /usr/local/Cellar/cmake/3.27.1/bin/cmake -D IDF_PATH=/Users/user/esp/esp-idf -D "SERIAL_TOOL=/Users/user/.espressif/python_env/idf5.1_py3.11_env/bin/python;;/Users/user/esp/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32s3" -D "SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;write_flash;@flash_args" -D WORKING_DIRECTORY=/Users/user/Documents/Code/Embedded/Demo_EPS/hello_world/build -P /Users/user/esp/esp-idf/components/esptool_py/run_serial_tool.cmake
esptool esp32s3 -p /dev/cu.usbmodem55340017611 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x0 bootloader/bootloader.bin 0x10000 hello_world.bin 0x8000 partition_table/partition-table.bin
esptool.py v4.6.2
Serial port /dev/cu.usbmodem55340017611
Connecting....
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: f4:12:fa:e7:0f:a0
Uploading stub...

A fatal error occurred: Failed to write to target RAM (result was 01070000: Operation timed out)
CMake Error at run_serial_tool.cmake:66 (message):

  /Users/user/.espressif/python_env/idf5.1_py3.11_env/bin/python;;/Users/user/esp/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32s3
  failed.

FAILED: CMakeFiles/flash /Users/user/Documents/Code/Embedded/Demo_EPS/hello_world/build/CMakeFiles/flash
cd /Users/user/esp/esp-idf/components/esptool_py && /usr/local/Cellar/cmake/3.27.1/bin/cmake -D IDF_PATH=/Users/user/esp/esp-idf -D "SERIAL_TOOL=/Users/user/.espressif/python_env/idf5.1_py3.11_env/bin/python;;/Users/user/esp/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32s3" -D "SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;write_flash;@flash_args" -D WORKING_DIRECTORY=/Users/user/Documents/Code/Embedded/Demo_EPS/hello_world/build -P /Users/user/esp/esp-idf/components/esptool_py/run_serial_tool.cmake
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /Users/user/Documents/Code/Embedded/Demo_EPS/hello_world/build/log/idf_py_stderr_output_35466 and /Users/user/Documents/Code/Embedded/Demo_EPS/hello_world/build/log/idf_py_stdout_output_35466
```

### Extension

log file content is 
```bash
[1/5] cd /Users/user/Documents/Code/Embedded/Demo_EPS/hello_world/build/esp-idf/esptool_py && /Users/user/.espressif/python_env/idf5.1_py3.11_env/bin/python /Users/user/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 partition --type app /Users/user/Documents/Code/Embedded/Demo_EPS/hello_world/build/partition_table/partition-table.bin /Users/user/Documents/Code/Embedded/Demo_EPS/hello_world/build/hello_world.bin
hello_world.bin binary size 0x31360 bytes. Smallest app partition is 0x100000 bytes. 0xceca0 bytes (81%) free.
[2/5] Performing build step for 'bootloader'
[1/1] cd /Users/user/Documents/Code/Embedded/Demo_EPS/hello_world/build/bootloader/esp-idf/esptool_py && /Users/user/.espressif/python_env/idf5.1_py3.11_env/bin/python /Users/user/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 /Users/user/Documents/Code/Embedded/Demo_EPS/hello_world/build/bootloader/bootloader.bin
Bootloader binary size 0x5200 bytes. 0x2e00 bytes (36%) free.
[3/5] No install step for 'bootloader'
[4/5] Completed 'bootloader'
[4/5] cd /Users/user/esp/esp-idf/components/esptool_py && /usr/local/Cellar/cmake/3.27.1/bin/cmake -D IDF_PATH=/Users/user/esp/esp-idf -D "SERIAL_TOOL=/Users/user/.espressif/python_env/idf5.1_py3.11_env/bin/python;;/Users/user/esp/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32s3" -D "SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;write_flash;@flash_args" -D WORKING_DIRECTORY=/Users/user/Documents/Code/Embedded/Demo_EPS/hello_world/build -P /Users/user/esp/esp-idf/components/esptool_py/run_serial_tool.cmake
esptool esp32s3 -p /dev/cu.usbmodem55340017611 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x0 bootloader/bootloader.bin 0x10000 hello_world.bin 0x8000 partition_table/partition-table.bin
esptool.py v4.6.2
Serial port /dev/cu.usbmodem55340017611
Connecting....
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: f4:12:fa:e7:0f:a0
Uploading stub...

A fatal error occurred: Failed to write to target RAM (result was 01070000: Operation timed out)
FAILED: CMakeFiles/flash /Users/user/Documents/Code/Embedded/Demo_EPS/hello_world/build/CMakeFiles/flash 
cd /Users/user/esp/esp-idf/components/esptool_py && /usr/local/Cellar/cmake/3.27.1/bin/cmake -D IDF_PATH=/Users/user/esp/esp-idf -D "SERIAL_TOOL=/Users/user/.espressif/python_env/idf5.1_py3.11_env/bin/python;;/Users/user/esp/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32s3" -D "SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;write_flash;@flash_args" -D WORKING_DIRECTORY=/Users/user/Documents/Code/Embedded/Demo_EPS/hello_world/build -P /Users/user/esp/esp-idf/components/esptool_py/run_serial_tool.cmake
ninja: build stopped: subcommand failed.

```

### Description

I refer to [this answer](https://github.com/espressif/esp-idf/issues/10371), but it doesn't solve the problem.

### Debug Message

```plain
Null
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

radurentea, 2023-08-08:
Hi @WelkinDev,

Please open Command Palette (CMD+ SHIFT + P) and run "Doctor Command", this will copy the output to your clipboard and then paste the output here in a reply, or in a txt file so that I can have a look at your configuration settings.

I already see that here:

"SERIAL_TOOL=/Users/user/.espressif/python_env/idf5.1_py3.11_env/bin/python`;;`/Users/user/esp..."

You have an extra ;, or an argument is missing.

closed 2023-08-29 by github-actions[bot]

dongyangdsp, 2023-10-17:
How this problem should be solved