espressif/vscode-esp-idf-extension #1025: ESP-IDF VSCode Extension 1.6.4 - Install Succeeds but Cannot Open Examples

J1mbo, 2023-08-16:
### OS

Windows

### Operating System version

Windows 11

### Visual Studio Code version

1.81.1

### ESP-IDF version

5.1

### Python version

3.11

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.22621 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Program Files (x86)\VMware\VMware Player\bin\;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\Git\cmd;C:\Program Files\CMake\bin;C:\Program Files\PuTTY\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Wincupl\WINCUPL\EXE;C:\Wincupl\WINCUPL\FITTERS;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.81.1 
Visual Studio Code language en 
Visual Studio Code shell C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.6.4 
Workspace folder c:<HOMEPATH>\iCloudDrive\Development\ESP32 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:<HOMEPATH>\esp\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin;C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:<HOMEPATH>\.espressif\tools\cmake\3.24.0\bin;C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230419\openocd-esp32\bin;C:<HOMEPATH>\.espressif\tools\ninja\1.10.2;C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3;C:<HOMEPATH>\.espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64;C:<HOMEPATH>\.espressif\tools\dfu-util\0.11\dfu-util-0.11-win64;C:<HOMEPATH>\.espressif\tools\esp-rom-elfs\20230320
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230419/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
    ESP_ROM_ELF_DIR: C:<HOMEPATH>\.espressif\tools\esp-rom-elfs\20230320/
Virtual env Python Path (idf.pythonBinPath) C:<HOMEPATH>\.espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe
Serial port (idf.port) COM1
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) C:<HOMEPATH>\.espressif
Git Path (idf.gitPath) C:<HOMEPATH>\.espressif\tools\idf-git\2.39.2\cmd\git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\cmake\3.24.0\bin: true
Access to C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230419\openocd-esp32\bin: true
Access to C:<HOMEPATH>\.espressif\tools\ninja\1.10.2: true
Access to C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3: true
Access to C:<HOMEPATH>\.espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64: true
Access to C:<HOMEPATH>\.espressif\tools\dfu-util\0.11\dfu-util-0.11-win64: true
Access to C:<HOMEPATH>\.espressif\tools\esp-rom-elfs\20230320: true
[... 93 lines cut ...]
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at Unknown error in ESP-IDF doctor command
--------------------------------------------------------------------------------------------------------------------------------------------

### Extension

{"message":"Cannot access filePath: c:\\Users\\[username]\\iCloudDrive\\Development\\ESP32\\CMakeLists.txt","stack":"Error: ENOENT: no such file or directory, access 'c:\\Users\\[username]\\iCloudDrive\\Development\\ESP32\\CMakeLists.txt'\n\tat Object.accessSync (node:fs:260:3)\n\tat t.accessSync (node:electron/js2c/asar_bundle:2:8282)\n\tat De (\\dist\\extension.js:2:1192563)\n\tat \\dist\\extension.js:2:1194020\n\tat je (\\dist\\extension.js:2:1194206)\n\tat \\dist\\extension.js:2:1189617\n\tat Generator.next (<anonymous>)\n\tat \\dist\\extension.js:2:1188704\n\tat new Promise (<anonymous>)\n\tat me (\\dist\\extension.js:2:1188449)\n\tat be (\\dist\\extension.js:2:1189576)\n\tat \\dist\\extension.js:2:1464410\n\tat Generator.next (<anonymous>)\n\tat \\dist\\extension.js:2:1462241\n\tat new Promise (<anonymous>)\n\tat Vs (\\dist\\extension.js:2:1461986)\n\tat va (\\dist\\extension.js:2:1462885)\n\tat Dn.eb (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:116:13102)\n\tat Dn.db (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:116:12816)\n\tat \\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:116:10913\n\tat async p.n (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:102:6206)\n\tat async p.m (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:102:6169)\n\tat async p.l (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:102:5626)","level":"error","timestamp":"2023-08-16T06:58:13.441Z"}
{"message":"Open On-Chip Debugger v0.12.0-esp32-20230419 (2023-04-18-22:02)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n","stack":"Error: Open On-Chip Debugger v0.12.0-esp32-20230419 (2023-04-18-22:02)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n\n\tat \\dist\\extension.js:2:1195478\n\tat ChildProcess.exithandler (node:child_process:401:7)\n\tat ChildProcess.emit (node:events:513:28)\n\tat maybeClose (node:internal/child_process:1121:16)\n\tat Socket.<anonymous> (node:internal/child_process:479:11)\n\tat Socket.emit (node:events:513:28)\n\tat Pipe.<anonymous> (node:net:757:14)","level":"error","timestamp":"2023-08-16T06:58:13.999Z"}
{"message":"C:\\Users\\[username]\\esp\\esp-idf\\tools\\check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html\r\n  import pkg_resources\r\n","stack":"Error: C:\\Users\\[username]\\esp\\esp-idf\\tools\\check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html\r\n  import pkg_resources\r\n\n\tat \\dist\\extension.js:2:1195478\n\tat ChildProcess.exithandler (node:child_process:401:7)\n\tat ChildProcess.emit (node:events:513:28)\n\tat maybeClose (node:internal/child_process:1121:16)\n\tat ChildProcess._handle.onexit (node:internal/child_process:304:5)","level":"error","timestamp":"2023-08-16T06:58:14.414Z"}

### Description

Cannot open example - "ESP-IDF: Loading examples" is displayed for a period of time, which then disapears and nothing else happens. Nothing is generated in the log. Doctor reports "Latest error at Unknown error in ESP-IDF doctor command".

VSCode and Python installed from Windows store, neither of which I have installed previously.

### Debug Message

```plain
None shown.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

radurentea, 2023-08-16:
Hi @J1mbo,

When you are pressing the "Show Examples button" there is an input up top which ask you to select and IDF version. The notification with loading examples runs until an idf version is selected. Can you confirm just a input has appeared and you've selected the idf version you wanted to use? (default is Current version of ESP-IDF)

J1mbo, 2023-08-16:
I did not even see that appear directly under the command box. Thank you.

closed 2023-08-16 by J1mbo