espressif/vscode-esp-idf-extension #1026: couldn't find a debug adapter descriptor for debug type espidf

5ami, 2023-08-16:
### OS

Windows

### Operating System version

windows 11 

### Visual Studio Code version

1.81.1

### ESP-IDF version

5.0

### Python version

3.8.7

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.22621 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files (x86)\OpenText\Viewer\bin;C:\Program Files\PuTTY\;C:\MathWorks\MATLAB\R2022a\runtime\win64;C:\MathWorks\MATLAB\R2022a\bin;C:\Program Files\TortoiseSVN\bin;C:\Program Files (x86)\Cypress\EZ-USB FX3 SDK\1.3\util\elf2img\;C:\Program Files\MATLAB\MATLAB Runtime\v912\runtime\win64;C;C:\Program Files\dotnet\;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin;C:<HOMEPATH>\AppData\Local\Programs\jaguar;C:<HOMEPATH>\AppData\Local\GitHubDesktop\bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.81.1 
Visual Studio Code language en 
Visual Studio Code shell C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.6.4 
Workspace folder c:<HOMEPATH>\eplanet_test 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:<HOMEPATH>\esp\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32s2-elf\esp-2022r1-11.2.0\xtensa-esp32s2-elf\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32s3-elf\esp-2022r1-11.2.0\xtensa-esp32s3-elf\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-2022r1-11.2.0\riscv32-esp-elf\bin;C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:<HOMEPATH>\.espressif\tools\cmake\3.24.0\bin;C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.11.0-esp32-20220706\openocd-esp32\bin;C:<HOMEPATH>\.espressif\tools\ninja\1.10.2;C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3;C:<HOMEPATH>\.espressif\tools\ccache\4.6.2\ccache-4.6.2-windows-x86_64;C:<HOMEPATH>\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars)
    0: {
    1: "
    2: O
    3: P
    4: E
    5: N
    6: O
    7: C
    8: D
    9: _
    10: S
    11: C
    12: R
    13: I
    14: P
    15: T
    16: S
    17: "
    18: :
    19: "
    20: C
    21: :
    22: \
    23: \
    24: U
    25: s
    26: e
    27: r
[... 292 lines cut ...]
          }
     ],
     "version": 4
}
 
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at Unknown error in ESP-IDF doctor command
--------------------------------------------------------------------------------------------------------------------------------------------

### Description

I used to run the debugger in a normal why, then the debugger suddenly stop giving this message error: 
couldn't find a debug adapter descriptor for debug type 'espidf' (Extension might have failed to activate )

[image]

### Debug Message

```plain
I have No debug message
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

radurentea, 2023-08-17:
Hi @5ami,

It seems that the python package esp-debug-backend it's not present. Could you try to install the python packages again? You can do so running the `ESP-IDF: Install ESP-IDF extension Python Packages` from the Command Palette (CTRL + SHIFT + P). If the issue persists, run another "Doctor Command" and paste it in here.

5ami, 2023-08-17:
Hi @radurentea, 
Thank you for your reply, I used the command **ESP-IDF: Install ESP-IDF extension Python Packages** and esp-debug-backend version: 1.0.3  is present.  But this didn't solve my problem. I still get the same messag(couldn't find a debug adapter descriptor for debug type 'espidf' )

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.22621 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files (x86)\OpenText\Viewer\bin;C:\Program Files\PuTTY\;C:\MathWorks\MATLAB\R2022a\runtime\win64;C:\MathWorks\MATLAB\R2022a\bin;C:\Program Files\TortoiseSVN\bin;C:\Program Files (x86)\Cypress\EZ-USB FX3 SDK\1.3\util\elf2img\;C:\Program Files\MATLAB\MATLAB Runtime\v912\runtime\win64;C;C:\Program Files\dotnet\;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin;C:<HOMEPATH>\AppData\Local\Programs\jaguar;C:<HOMEPATH>\AppData\Local\GitHubDesktop\bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.81.1 
Visual Studio Code language en 
Visual Studio Code shell C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.6.4 
Workspace folder c:<HOMEPATH>\eplanet_test 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:<HOMEPATH>\esp\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32s2-elf\esp-2022r1-11.2.0\xtensa-esp32s2-elf\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32s3-elf\esp-2022r1-11.2.0\xtensa-esp32s3-elf\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-2022r1-11.2.0\riscv32-esp-elf\bin;C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:<HOMEPATH>\.espressif\tools\cmake\3.24.0\bin;C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.11.0-esp32-20220706\openocd-esp32\bin;C:<HOMEPATH>\.espressif\tools\ninja\1.10.2;C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3;C:<HOMEPATH>\.espressif\tools\ccache\4.6.2\ccache-4.6.2-windows-x86_64;C:<HOMEPATH>\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars)
    0: {
    1: "
    2: O
    3: P
    4: E
    5: N
    6: O
    7: C
    8: D
    9: _
    10: S
    11: C
    12: R
    13: I
    14: P
    15: T
    16: S
    17: "
    18: :
    19: "
    20: C
    21: :
    22: \
    23: \
    24: U
    25: s
    26: e
    27: r
    28: s
    29: \
    30: \
    31: U
    32: 4
    33: 0
    34: 5
    35: 1
    36: 7
    37: 2
    38: \
    39: \
    40: .
    41: e
    42: s
    43: p
    44: r
    45: e
    46: s
[... 264 lines cut ...]
               "cStandard": "c11",
               "cppStandard": "c++17",
               "includePath": [
                    "${config:idf.espIdfPath}/components/**",
                    "${config:idf.espIdfPathWin}/components/**",
                    "${config:idf.espAdfPath}/components/**",
                    "${config:idf.espAdfPathWin}/components/**",
                    "${workspaceFolder}/**"
               ],
               "browse": {
                    "path": [
                         "${config:idf.espIdfPath}/components",
                         "${config:idf.espIdfPathWin}/components",
                         "${config:idf.espAdfPath}/components/**",
                         "${config:idf.espAdfPathWin}/components/**",
                         "${workspaceFolder}"
                    ],
                    "limitSymbolsToIncludedHeaders": false
               }
          }
     ],
     "version": 4
}
 
--------------------------------------------------------------------------------------------------------------------------------------------

radurentea, 2023-08-17:
A big red flag is the way idf.customExtraVars property's value looks like. Try running the `ESP-IDF: Configure ESP-IDF extension` and configure the extension once again, because the format of the idf.customExtraVars value is wrong.

It should be something similar to:

[image]

5ami, 2023-08-17:
Hi @radurentea , reconfiguring the extention or deleting it and installing it again, results in the same weird idf.CustumExtraVars, is there a way to change that manually (this only happens in the versions of the extension 1.6.x , in the 1.5.x it works fine)

radurentea, 2023-08-17:
You can have at the documentation [here](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/SETUP.md#json-manual-configuration). It shows some examples as well.

Let me know if you run into issues

5ami, 2023-08-17:
The idf.customExtraVars seem to be correct in the settings  as you can see in the screenshot 
[image]
However when I run the ESP-IDF: Doctor command, it displays it in that weird way

5ami, 2023-08-17:
[image]

radurentea, 2023-08-17:
Hi @5ami,

This is strange, I'm not sure what command you've used for the settings (vscode has several different scopes for settings), but lets try the `Preferences: Open User Settings (JSON)` and search for "idf.customExtraVars".

It should be in this format:

[image]

If this won't work, maybe a clean reinstall of the extension, but make sure to follow the steps from [here](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/INSTALL.md#uninstalling-the-plugin)

5ami, 2023-08-17:
This is what I see from  Preferences: Open User Settings (JSON)
[image]

5ami, 2023-08-17:
Hi @radurentea, Thank you for your support. 

I have solved the problem. It was because the **customExtravars** expects an Object, but my settings.json in the **.vscode** folder had a string passed in, where it expected an Object (probably it stored that somehow from the previous version of the extension) . 

[image]

radurentea, 2023-08-17:
Glad you've notice it!