espressif/vscode-esp-idf-extension #1027: The system cannot find the path specified.

NemesisXB, 2023-08-17:
### OS

Windows

### Operating System version

Windows 11

### Visual Studio Code version

1.81.1

### ESP-IDF version

4.4.5

### Python version

3.8.7

### Doctor command output

```
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.22621 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Program Files\Microsoft\jdk-11.0.16.101-hotspot\bin;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\Git\cmd;C:\Program Files\TortoiseGit\bin;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\PuTTY\;C:\Program Files (x86)\Bitvise SSH Client;C:\Program Files\dotnet\;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\.espressif\tools\idf-python\3.11.2\Scripts;C:\Program Files (x86)\GnuWin32\bin;C:<HOMEPATH>\.espressif\tools\ninja\1.10.2;C:<HOMEPATH>\.dotnet\tools;C:<HOMEPATH>\.espressif\tools\cmake\3.23.1\bin;;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.81.1 
Visual Studio Code language en 
Visual Studio Code shell C:\Windows\System32\cmd.exe 
ESP-IDF Extension version 1.6.4 
Workspace folder d:\Projects\Test\Test_MainBoard 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:<HOMEPATH>\esp\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s2-elf\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s3-elf\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\bin;C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:<HOMEPATH>\.espressif\tools\cmake\3.23.1\bin;C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\bin;C:<HOMEPATH>\.espressif\tools\ninja\1.10.2;C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3;C:<HOMEPATH>\.espressif\tools\ccache\4.3\ccache-4.3-windows-64;C:<HOMEPATH>\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.11.0-esp32-20221026/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
Virtual env Python Path (idf.pythonBinPath) C:<HOMEPATH>\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe
Serial port (idf.port) COM8
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,target/esp32.cfg
ESP-IDF Tools Path (idf.toolsPath) C:<HOMEPATH>\.espressif
Git Path (idf.gitPath) C:<HOMEPATH>\.espressif\tools\idf-git\2.30.1\cmd\git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s2-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s3-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\cmake\3.23.1\bin: true
Access to C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\bin: true
Access to C:<HOMEPATH>\.espressif\tools\ninja\1.10.2: true
Access to C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3: true
Access to C:<HOMEPATH>\.espressif\tools\ccache\4.3\ccache-4.3-windows-64: true
Access to C:<HOMEPATH>\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64: true
Access to Virtual env Python Path (idf.pythonBinPath) true
[... 186 lines cut ...]
  import pkg_resources
Executing action: all (aliases: build)
Running ninja in directory D:\Projects\Test\Test_MainBoard\build
Executing "ninja all"...
```

### Debug Message

```plain
Executing task: ninja  

The system cannot find the path specified.
```

### Other Steps to Reproduce

It used to work previously. I replaced my HDD and did a fresh install of Windows, VSCODE, esp-idf extension. And now it does not work.

I have tried uninstalling everything and reinstalling, no change. I tried older versions of the extension, other versions of ESP-IDF, but no change.

I have no idea where to look for the issue.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

radurentea, 2023-08-17:
Hi @NemesisXB,

The only thing that looks suspicious is maybe in your system environment variable **PATH**, I can see that after the cmake variable, there is an empty variable hence the double ;;
[image]

I've verified my path variable and  that's where my ninja path is. This is far stretched cause it shouldn't affect the behavior, but try to change your env **PATH** variable by cutting the ninja path and putting it after the cmake one and delete that extra `;`

Another thing you should consider is to check the settings.json found in .vscode folder inside your project just to make sure the settings are not overwritten there.

If these 2 solutions don't work, try to do a clean re-install by following this [guide](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/INSTALL.md#uninstalling-the-plugin) and if that doesn't help either, run the doctor command and paste it in your next comment.

NemesisXB, 2023-08-20:
Thanks @radurentea, 

I tried your recommendations. But still the same. I also forgot to mention this happens even when creating a new project based on one of the examples.

But here is the output of the doctor command:
```
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.22621 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Program Files\Microsoft\jdk-11.0.16.101-hotspot\bin;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\Git\cmd;C:\Program Files\TortoiseGit\bin;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\PuTTY\;C:\Program Files (x86)\Bitvise SSH Client;C:\Program Files\dotnet\;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\.espressif\tools\idf-python\3.11.2\Scripts;C:\Program Files (x86)\GnuWin32\bin;C:<HOMEPATH>\.espressif\tools\ninja\1.10.2;C:<HOMEPATH>\.dotnet\tools;C:<HOMEPATH>\.espressif\tools\cmake\3.23.1\bin;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin;C:<HOMEPATH>\esp\esp-idf\tools; 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.81.1 
Visual Studio Code language en 
Visual Studio Code shell C:\Windows\System32\cmd.exe 
ESP-IDF Extension version 1.6.4 
Workspace folder d:\Projects\Test\Test_MainBoard 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:<HOMEPATH>\esp\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s2-elf\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s3-elf\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\bin;C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:<HOMEPATH>\.espressif\tools\cmake\3.23.1\bin;C:<HOMEPATH>\.espressif\tools\ninja\1.10.2;C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\bin;C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3;C:<HOMEPATH>\.espressif\tools\ccache\4.3\ccache-4.3-windows-64;C:<HOMEPATH>\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.11.0-esp32-20221026/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
Virtual env Python Path (idf.pythonBinPath) C:<HOMEPATH>\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe
Serial port (idf.port) COM8
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,target/esp32.cfg
ESP-IDF Tools Path (idf.toolsPath) C:<HOMEPATH>\.espressif
Git Path (idf.gitPath) C:<HOMEPATH>\.espressif\tools\idf-git\2.30.1\cmd\git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s2-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s3-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\cmake\3.23.1\bin: true
Access to C:<HOMEPATH>\.espressif\tools\ninja\1.10.2: true
Access to C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\bin: true
Access to C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3: true
Access to C:<HOMEPATH>\.espressif\tools\ccache\4.3\ccache-4.3-windows-64: true
Access to C:<HOMEPATH>\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) true
-------------------------------------------------------- Configurations has spaces -------------------------------------------------------------
Spaces in system environment Path true
Spaces in ESP-ADF Path (idf.espAdfPath) false
Spaces in ESP-IDF Path (idf.espIdfPath) false
Spaces in ESP-MDF Path (idf.espMdfPath) false
[... cut ...]
    "idf.espIdfPathWin": "C:\\Users\\NemesisXB\\esp\\esp-idf",
    "idf.openOcdConfigs": [
        "interface/ftdi/esp32_devkitj_v1.cfg",
        "target/esp32.cfg"
    ],
    "idf.portWin": "COM8",
    "idf.pythonBinPathWin": "C:\\Users\\NemesisXB\\.espressif\\python_env\\idf4.4_py3.8_env\\Scripts\\python.exe",
    "idf.toolsPathWin": "C:\\Users\\NemesisXB\\.espressif",
    "idf.flashType": "UART",
    "cmake.configureOnOpen": false,
    "idf.gitPathWin": "C:\\Users\\NemesisXB\\.espressif\\tools\\idf-git\\2.30.1\\cmd\\git.exe"
```

radurentea, 2023-08-21:
Settings look ok.

You've already clean uninstall using the guide? (removing the %USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\ folder)

If that's the case, please go to that folder again after you've ran the build command again and copy the last 10 rows of the `esp_idf_vsc_ext.log` file and paste them in here.

NemesisXB, 2023-08-21:
Yes, I followed the instruction and deleted the file. The file however does not update when I run the build command. The last update seems to be the doctor command I ran last night. Here is the file anyway.

```
{"message":"Open On-Chip Debugger v0.11.0-esp32-20221026 (2022-10-26-14:48)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n","stack":"Error: Open On-Chip Debugger v0.11.0-esp32-20221026 (2022-10-26-14:48)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n\n\tat \\dist\\extension.js:2:1195478\n\tat ChildProcess.exithandler (node:child_process:401:7)\n\tat ChildProcess.emit (node:events:513:28)\n\tat maybeClose (node:internal/child_process:1121:16)\n\tat ChildProcess._handle.onexit (node:internal/child_process:304:5)","level":"error","timestamp":"2023-08-20T20:14:28.039Z"}
{"message":"C:\\Users\\NemesisXB\\esp\\esp-idf\\tools\\check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html\r\n  import pkg_resources\r\n","stack":"Error: C:\\Users\\NemesisXB\\esp\\esp-idf\\tools\\check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html\r\n  import pkg_resources\r\n\n\tat \\dist\\extension.js:2:1195478\n\tat ChildProcess.exithandler (node:child_process:401:7)\n\tat ChildProcess.emit (node:events:513:28)\n\tat maybeClose (node:internal/child_process:1121:16)\n\tat ChildProcess._handle.onexit (node:internal/child_process:304:5)","level":"error","timestamp":"2023-08-20T20:14:28.323Z"}
{"message":"Open On-Chip Debugger v0.11.0-esp32-20221026 (2022-10-26-14:48)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n","stack":"Error: Open On-Chip Debugger v0.11.0-esp32-20221026 (2022-10-26-14:48)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n\n\tat \\dist\\extension.js:2:1195478\n\tat ChildProcess.exithandler (node:child_process:401:7)\n\tat ChildProcess.emit (node:events:513:28)\n\tat maybeClose (node:internal/child_process:1121:16)\n\tat ChildProcess._handle.onexit (node:internal/child_process:304:5)","level":"error","timestamp":"2023-08-20T20:14:28.570Z"}
{"message":"C:\\Users\\NemesisXB\\esp\\esp-idf\\tools\\check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html\r\n  import pkg_resources\r\n","stack":"Error: C:\\Users\\NemesisXB\\esp\\esp-idf\\tools\\check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html\r\n  import pkg_resources\r\n\n\tat \\dist\\extension.js:2:1195478\n\tat ChildProcess.exithandler (node:child_process:401:7)\n\tat ChildProcess.emit (node:events:513:28)\n\tat maybeClose (node:internal/child_process:1121:16)\n\tat ChildProcess._handle.onexit (node:internal/child_process:304:5)","level":"error","timestamp":"2023-08-20T20:14:28.836Z"}
{"message":"Open On-Chip Debugger v0.11.0-esp32-20221026 (2022-10-26-14:48)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n","stack":"Error: Open On-Chip Debugger v0.11.0-esp32-20221026 (2022-10-26-14:48)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n\n\tat \\dist\\extension.js:2:1195478\n\tat ChildProcess.exithandler (node:child_process:401:7)\n\tat ChildProcess.emit (node:events:513:28)\n\tat maybeClose (node:internal/child_process:1121:16)\n\tat ChildProcess._handle.onexit (node:internal/child_process:304:5)","level":"error","timestamp":"2023-08-20T20:14:30.601Z"}
{"message":"C:\\Users\\NemesisXB\\esp\\esp-idf\\tools\\check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html\r\n  import pkg_resources\r\n","stack":"Error: C:\\Users\\NemesisXB\\esp\\esp-idf\\tools\\check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html\r\n  import pkg_resources\r\n\n\tat \\dist\\extension.js:2:1195478\n\tat ChildProcess.exithandler (node:child_process:401:7)\n\tat ChildProcess.emit (node:events:513:28)\n\tat maybeClose (node:internal/child_process:1121:16)\n\tat ChildProcess._handle.onexit (node:internal/child_process:304:5)","level":"error","timestamp":"2023-08-20T20:14:30.878Z"}
{"message":"Open On-Chip Debugger v0.11.0-esp32-20221026 (2022-10-26-14:48)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n","stack":"Error: Open On-Chip Debugger v0.11.0-esp32-20221026 (2022-10-26-14:48)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n\n\tat \\dist\\extension.js:2:1195478\n\tat ChildProcess.exithandler (node:child_process:401:7)\n\tat ChildProcess.emit (node:events:513:28)\n\tat maybeClose (node:internal/child_process:1121:16)\n\tat ChildProcess._handle.onexit (node:internal/child_process:304:5)","level":"error","timestamp":"2023-08-20T20:14:31.111Z"}
[... cut ...] HOMEPATH>\\\\.espressif\\\\tools\\\\cmake\\\\3.23.1\\\\bin;C:<HOMEPATH>\\\\AppData\\\\Local\\\\Programs\\\\Microsoft VS Code\\\\bin;C:<HOMEPATH>\\\\esp\\\\esp-idf\\\\tools;\",\"extensionVersion\":\"1.6.4\",\"language\":\"en\",\"shell\":\"C:\\\\Windows\\\\System32\\\\cmd.exe\",\"platform\":\"win32\",\"systemName\":\"10.0.22621\",\"vscodeVersion\":\"1.81.1\"},\"workspaceFolder\":\"d:\\\\Projects\\\\Test\\\\Test_MainBoard\",\"projectConfigurations\":{}}","timestamp":"2023-08-20T20:17:07.319Z"}

```

brianignacio5, 2023-08-21:
Hmm what are the values of these settings?

`idf.buildPathWin` ? `idf.cmakeCompilerArgs`? `idf.ninjaArgs`? Can you delete the build subdirectory and try again?

NemesisXB, 2023-08-21:
> Hmm what are the values of these settings?
> 
> `idf.buildPathWin` ? `idf.cmakeCompilerArgs`? `idf.ninjaArgs`? Can you delete the build subdirectory and try again?

"idf.buildPathWin": "${workspaceFolder}\\build"

"idf.cmakeCompilerArgs": [
	"-G",
	"Ninja",
	"-DPYTHON_DEPS_CHECKED=1",
	"-DESP_PLATFORM=1"
],

"idf.ninjaArgs": [],

Deleted the build directory. The error then looks like this:
```
Executing task: cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -B d:\Projects\Test\Test_MainBoard\build -S d:\Projects\Test\Test_MainBoard 

The system cannot find the path specified.

D:\Projects\Test\Test_MainBoard\build>
```

NemesisXB, 2023-08-22:
> Hmm, try to modify buildPathWin to `${workspaceFolder}\\build` , using double `\\`

It then results in this:
```
Executing task: cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -B c:\Users\NemesisXB\Test\\build -S c:\Users\NemesisXB\Test 

The system cannot find the path specified.
```
when I leave it as `${workspaceFolder}\build`, I get:
```
Executing task: cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -B c:\Users\NemesisXB\Test\build -S c:\Users\NemesisXB\Test 

The system cannot find the path specified.
```

radurentea, 2023-08-22:
are you modifying the value from the json file or from the Settings UI ?

I'm asking cause these how the settings are on my working machine in the settings json file:
[image]

Another thing that comes to mind is to make sure you are using one of our example projects like "blink" while we are troubleshooting the error so that we minimise mistakes on project configuration.

NemesisXB, 2023-08-22:
@radurentea, yes I did switch to the hello world example already. You will see the project location and name changed in the previous comment. It is now called test and is in the "default" location (c:\Users\{username}\{project_name} in my case c:\Users\NemesisXB\Test). This also places the project on the c drive where my actual project is on the d drive (not that I thought that it would make a difference). But the error persists.

With regards to the settings, I did edit in the JSON. Here is mine currently:
[image]

At this point I am willing to give you teamviewer or anydesk access to my machine so you can directly try and figure it out.

brianignacio5, 2023-08-23:
There is something weird in the build task that we can't seem to reproduce.

Can you open an terminal with the `ESP-IDF: Open ESP-IDF Terminal` and run the `cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -B c:\Users\NemesisXB\Test\build -S c:\Users\NemesisXB\Test` . The environment should be more or less the same as the build task.

Also have you check this solution in [here](https://superuser.com/questions/727316/error-in-command-line-the-system-cannot-find-the-path-specified) related to make itself.

NemesisXB, 2023-08-23:
Ok, I have it narrowed down.
The default command promt is this:
```
"Command Prompt": {
            "path": "C:\\Windows\\System32\\cmd.exe",
            "args": [
                "/k",
                "C:/Users/NemesisXB/.espressif/tools/Espressif/idf_cmd_init.bat esp-idf-a0f2bdeffee2b243ea113d63044fd506"
            ]
        }
```

However, `idf_cmd_init.bat` is in `C:/Users/NemesisXB/.espressif/tools/`. There is no Espressif subfolder under tools. But removing the Espressif in the path results in another error:
```
Executing task: ninja  

'C:/Users/NemesisXB/.espressif/tools/idf_cmd_init.bat' is not recognized as an internal or external command,
operable program or batch file.
```

It does work if I change it to use just the plain powershell as the default:
```
"PowerShell": {
            "source": "PowerShell",
            "icon": "terminal-powershell"
        },
```

brianignacio5, 2023-08-23:
Interesting. It seems that the IDF Installer does a post install to your Command prompt in here:

https://github.com/espressif/idf-installer/blob/main/src/InnoSetup/PostInstall.iss#L78

So this actually assumes you are using the IDF Installer setup from there forward.

NemesisXB, 2023-08-23:
> C:/Users/NemesisXB/.espressif/tools/idf_cmd_init.bat

I had the path incorrect. I had: `C:/Users/NemesisXB/.espressif/tools/idf_cmd_init.bat` but it is actually `C:/Users/NemesisXB/.espressif/idf_cmd_init.bat`

but now it just runs the batch file and does not actually build the project:
```
Executing task: cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -B c:\Users\NemesisXB\Test\build -S c:\Users\NemesisXB\Test 

Setting PYTHONNOUSERSITE, was not set
Using Python in C:\Users\NemesisXB\.espressif\python_env\idf4.4_py3.11_env\Scripts\
Python 3.11.2
Using Git in C:\Users\NemesisXB\.espressif\tools\idf-git\2.39.2\cmd\
git version 2.39.2.windows.1
Setting IDF_PATH: C:\Users\NemesisXB\esp\esp-idf

Adding ESP-IDF tools to PATH...
No directories added to PATH:

C:\Users\NemesisXB\.espressif\python_env\idf4.4_py3.11_env\Scripts\;C:\Users\NemesisXB\.espressif\tools\idf-git\2.39.2\cmd\;C:\Users\NemesisXB\.espressif;C:\Users\NemesisXB\esp\esp-idf\components\esptool_py\esptool;C:\Users\NemesisXB\esp\esp-idf\components\espcoredump;C:\Users\NemesisXB\esp\esp-idf\components\partition_table;C:\Users\NemesisXB\.espressif\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin;C:\Users\NemesisXB\.espressif\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin;C:\Users\NemesisXB\.espressif\tools\xtensa-esp32-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\bin;C:\Users\NemesisXB\.espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s2-elf\bin;C:\Users\NemesisXB\.espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s3-elf\bin;C:\Users\NemesisXB\.espressif\tools\riscv32-esp-elf\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\bin;C:\Users\NemesisXB\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:\Users\NemesisXB\.espressif\tools\cmake\3.23.1\bin;C:\Users\NemesisXB\.espressif\tools\ninja\1.10.2;C:\Users\NemesisXB\.espressif\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\bin;C:\Users\NemesisXB\.espressif\tools\idf-exe\1.0.3;C:\Users\NemesisXB\.espressif\tools\ccache\4.3\ccache-4.3-windows-64;C:\Users\NemesisXB\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64;C:\Users\NemesisXB\.espressif\python_env\idf4.4_py3.11_env\Scripts;C:\Users\NemesisXB\esp\esp-idf\tools;C:\Users\NemesisXB\.espressif\tools\idf-git\2.39.2\cmd;C:\Program Files\Microsoft\jdk-11.0.16.101-hotspot\bin;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\Git\cmd;C:\Program Files\TortoiseGit\bin;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\PuTTY\;C:\Program Files (x86)\Bitvise SSH Client;C:\Program Files\dotnet\;C:\Users\NemesisXB\AppData\Local\Microsoft\WindowsApps;C:\Users\NemesisXB\.espressif\python_env\idf4.4_py3.11_env\Scripts;C:\Program Files (x86)\GnuWin32\bin;C:\Users\NemesisXB\.dotnet\tools;C:\Users\NemesisXB\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\NemesisXB\esp\esp-idf\tools;C:\Users\NemesisXB\.espressif\tools\ninja\1.10.2\;C:\Users\NemesisXB\.espressif\tools\cmake\3.23.1\bin;

Checking if Python packages are up to date...
C:\Users\NemesisXB\esp\esp-idf\tools\check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Python requirements from C:\Users\NemesisXB\esp\esp-idf\requirements.txt are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

C:\Users\NemesisXB\Test\build>
```

brianignacio5, 2023-08-23:
yeah because that is what the Command Prompt is configured to do when you run Command Prompt anytime. You could remove these arguments from Command Prompt or just use Powershell for the vscode extension.

NemesisXB, 2023-08-23:
> yeah because that is what the Command Prompt is configured to do when you run Command Prompt anytime. You could remove these arguments from default Command Prompt args or just use Powershell for the vscode extension.

Thanks. This is now working.

radurentea, 2023-08-23:
Glad we figured it out. Thanks @brianignacio5.

Closing issue.