espressif/vscode-esp-idf-extension #1029: Multiple concurrent IDF builds -- error in build process?

SolidStateLEDLighting, 2023-08-29:
### OS

Windows

### Operating System version

Win10

### Visual Studio Code version

1.8.11

### ESP-IDF version

4.4.5

### Python version

3.11.2

### Doctor command output

Hardware is custom and working fine with a normal 4.4.5 build.

### Extension

[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/12461717/esp_idf_vsc_ext.log)

### Description

I'm struggling to achieve a consistent build process that will support a migration of a commercial product from 4.4.5 to 5.0 and beyond.  We must be able to migrate while we maintain 4.4.5. 

### Debug Message

```plain
Here is one of that problems that I may have discovered:

Command failed: "c:\Users\Keith\esp\tools\tools\idf-python\3.11.2\python.exe" "c:\Users\Keith\esp\v4.4.5\idf\esp-idf\tools\idf_tools.py" install-python-env

Notice that the backslashes should be forward slashes.   I corrected this a ran it again for my process.

This is an alert so you can fix this issue.    OR, if I have it wrong, please explain it to me.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SolidStateLEDLighting, 2023-08-30:
I just manually reversed those slashes and pushed through this problem.  I would assume that you know where this would have occurred in the process.

I'm fully passed this place now and have achieved all my goals.

radurentea, 2023-08-31:
Glad to hear, I needed to check your configurations to have a full understanding of the issue. These type of issues are quite often happening because of a misconfiguration.

Since I can't reproduce the issue and you've managed to fix it yourself, I'll be closing the issue.

closed 2023-08-31 by radurentea