espressif/vscode-esp-idf-extension #1030: ESP-IDF: Configure ESP-IDF extension stuck

edieruby, 2023-08-31:
### OS

Windows

### Operating System version

windows 10

### Visual Studio Code version

1.81.1

### ESP-IDF version

4.4.4

### Python version

3.8.7

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19045 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Perl64\site\bin;C:\Perl64\bin;C:\Program Files (x86)\Razer Chroma SDK\bin;C:\Program Files\Razer Chroma SDK\bin;C:\Program Files (x86)\Razer\ChromaBroadcast\bin;C:\Program Files\Razer\ChromaBroadcast\bin;C:\windows\system32;C:\windows;C:\windows\System32\Wbem;C:\windows\System32\WindowsPowerShell\v1.0\;C:\windows\System32\OpenSSH\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\Intel\WiFi\bin\;C:\Program Files\Common Files\Intel\WirelessCommon\;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\windows\system32\config\systemprofile\AppData\Local\Microsoft\WindowsApps;C:\openjdk-8u372-b07\bin;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\dotnet\;C:\Program Files (x86)\dotnet\;C:\WINDOWS\system32\config\systemprofile\AppData\Local\Microsoft\WindowsApps;C:\Android;C:\Windows\System32;C:\Program Files\Graphviz 2.44.1\bin;C:\Program Files\TortoiseSVN\bin;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\Amazon\AWSCLIV2\;C:\Program Files (x86)\NetSarang\Xshell 7\;C:\Program Files\Git\cmd;C:\Program Files\TortoiseGit\bin;C:\Program Files (x86)\HP\Common\HPDestPlgIn\;C:\Program Files\Docker\Docker\resources\bin;C:\Go\bin;C:\apache-maven-3.9.3\bin;C:<HOMEPATH>\scoop\shims;C:<HOMEPATH>\AppData\Local\Programs\Python\Python38-32\Scripts\;C:<HOMEPATH>\AppData\Local\Programs\Python\Python38-32\;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:\Program Files\Git\bin;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin;C:\Program Files\JetBrains\GoLand 2019.3.1\bin;C:\Program Files\JetBrains\PyCharm 2019.3.3\bin;C:<HOMEPATH>\go\src\github.com\brendangregg\FlameGraph\FlameGraph;C:\Program Files\Go\bin;C:<HOMEPATH>\go\bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.81.1 
Visual Studio Code language en 
Visual Studio Code shell C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.6.4 
Workspace folder No workspace folder is open 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:<HOMEPATH>\esp4.4.4\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) C:<HOMEPATH>\.espressif4.4.4\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif4.4.4\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif4.4.4\tools\xtensa-esp32-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\bin;C:<HOMEPATH>\.espressif4.4.4\tools\xtensa-esp32s2-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s2-elf\bin;C:<HOMEPATH>\.espressif4.4.4\tools\xtensa-esp32s3-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s3-elf\bin;C:<HOMEPATH>\.espressif4.4.4\tools\riscv32-esp-elf\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\bin;C:<HOMEPATH>\.espressif4.4.4\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:<HOMEPATH>\.espressif4.4.4\tools\cmake\3.23.1\bin;C:<HOMEPATH>\.espressif4.4.4\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\bin;C:<HOMEPATH>\.espressif4.4.4\tools\ninja\1.10.2;C:<HOMEPATH>\.espressif4.4.4\tools\idf-exe\1.0.3;C:<HOMEPATH>\.espressif4.4.4\tools\ccache\4.3\ccache-4.3-windows-64;C:<HOMEPATH>\.espressif4.4.4\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: C:<HOMEPATH>\.espressif4.4.4\tools\openocd-esp32\v0.11.0-esp32-20221026/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
Virtual env Python Path (idf.pythonBinPath) C:<HOMEPATH>\.espressif4.4.4\python_env\idf4.4_py3.8_env\Scripts\python.exe
Serial port (idf.port) COM6
OpenOCD Configs (idf.openOcdConfigs) board/esp32c3-ftdi.cfg
ESP-IDF Tools Path (idf.toolsPath) C:<HOMEPATH>\.espressif4.4.4
Git Path (idf.gitPath) C:<HOMEPATH>\.espressif4.4.4\tools\idf-git\2.30.1\cmd\git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to C:<HOMEPATH>\.espressif4.4.4\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\.espressif4.4.4\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\.espressif4.4.4\tools\xtensa-esp32-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\bin: true
Access to C:<HOMEPATH>\.espressif4.4.4\tools\xtensa-esp32s2-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s2-elf\bin: true
Access to C:<HOMEPATH>\.espressif4.4.4\tools\xtensa-esp32s3-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s3-elf\bin: true
Access to C:<HOMEPATH>\.espressif4.4.4\tools\riscv32-esp-elf\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\bin: true
Access to C:<HOMEPATH>\.espressif4.4.4\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin: true
Access to C:<HOMEPATH>\.espressif4.4.4\tools\cmake\3.23.1\bin: true
Access to C:<HOMEPATH>\.espressif4.4.4\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\bin: true
Access to C:<HOMEPATH>\.espressif4.4.4\tools\ninja\1.10.2: true
Access to C:<HOMEPATH>\.espressif4.4.4\tools\idf-exe\1.0.3: true
Access to C:<HOMEPATH>\.espressif4.4.4\tools\ccache\4.3\ccache-4.3-windows-64: true
Access to C:<HOMEPATH>\.espressif4.4.4\tools\dfu-util\0.9\dfu-util-0.9-win64: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH undefined
[... 97 lines cut ...]
---------------------------------------------------- Check ESP-IDF debug adapter requirements.txt ------------------------------------------
Check Debug AdapterPython packages C:<HOMEPATH>\esp4.4.4\esp-idf\tools\check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Python requirements from c:<HOMEPATH>\.vscode\extensions\espressif.esp-idf-extension-1.6.4\esp_debug_adapter\requirements.txt are satisfied.
---------------------------------------------------- Visual Studio Code launch.json --------------------------------------------------------
No workspace folder is opened 
---------------------------------------------------- Visual Studio Code c_cpp_properties.json ----------------------------------------------
No workspace folder is opened 
--------------------------------------------------------------------------------------------------------------------------------------------

### Description

stuck ,  don't know what's wrong

[image]

### Debug Message

```plain
no
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

radurentea, 2023-08-31:
Hi @edieruby,

Can you go to `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\` and upload here the esp_idf_vsc_ext.log file ?

radurentea, 2023-09-01:
It seems from the log that you have some issues with the extension and you've tried to run again the ESP-IDF: Configure ESP-IDF but it gets stuck.

Can you explain when exactly it gets stuck? Maybe take a screenshot with the "ESP-IDF Setup" web view open (the one that opens up when running the ESP-IDF: Configure ESP-IDF command)?

edieruby, 2023-09-01:
stuck each time and there is not any output.
Just F1 then select ESP-IDF: Configure ESP-IDF extension

[image]

other command is okay.
build project.
[image]

radurentea, 2023-09-01:
Hmm, this behaviour is very strange so it does not even open the setup webview.

This is very strange, cause there is not much happening while getting the esp-idf versions, maybe some extension files got corrupted.

I would suggest to make a clean uninstall of the extension, following this [guide](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/INSTALL.md#uninstalling-the-plugin)

edieruby, 2023-09-04:
reinstalled.  
not fixed.
[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/12508705/esp_idf_vsc_ext.log)

edieruby, 2023-09-12:
> Go to a browser and try to get this file [idf_versions](https://dl.espressif.com/dl/esp-idf/idf_versions.txt)
> 
> Does it work?
> 
> Do you have any proxy configuration in your system or in Visual Studio Code? It could probably be failing because of it.

Thank you. 
I use global proxy. 
I use proxy as before. And it's okay before.
upgrade is okay, too. (v4.4.2 to v.4.4.4)
I want to upgrade to version 5 this time, and it occur this.
I will try it when proxy is close tomorrow.

[image]

closed 2023-10-03 by github-actions[bot]