espressif/vscode-esp-idf-extension #1032: ESP-IDF Installation Error(With Espressif IDE and VS Code Extension) (IDFGH-10986)

HasanYektaDemirtas, 2023-09-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.3

### Operating System used.

Windows

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I am trying  to install ESP IDF with Espressif IDE and VS Code extension.

### What is the actual behavior?

I get error.
Espressif IDE -> Installation has failed with exit code 1
Vs Code Extension -> 

### Steps to reproduce.

VS CODE
1. Step: Install Git
2. Step: Install VS Code
3. Step: Get ESP-IDF Explorer Extension
4. Step: Click View -> Command Palette
5. Step: Write "ESP-IDF: Configure ESP-IDF extension" and click
6. Step: Click "EXPRESS"

[image]

7. Step: Select ESP-IDF version as v5.0.3 and click install

[image]

ESP IDE
1. Step: Download offline install tool(espressif-ide-setup-2.10.0-with-esp-idf-5.0.2.exe)
2. Step: Click All Selections

[image]

3. Step: Click next and install

### Build or installation Logs.

```plain
Logs added below
```

### More Information.

I saw some related issues but none of them have working solutions for me.

VS Code Extension Log:
[VSCode-EspIdf-Installation-Log .txt](https://github.com/espressif/esp-idf/files/12495927/VSCode-EspIdf-Installation-Log.txt)

Espressif IDE Log:
[Setup Log 2023-09-01-002.txt](https://github.com/espressif/esp-idf/files/12495928/Setup.Log.2023-09-01-002.txt)

HasanYektaDemirtas, 2023-09-01:
> Edit: sorry, I misread the procedure where you installed IDF, so maybe my solution won't work for you, but it might work for others.
> 
> Hi, according to my tests a week ago, it is not possible to install ESP-IDF from IDF-Tools and then choose to have the VSCode Espressif IDF extension find an existing installation. Although it is the procedure that several people, courses, etc. recommend, I had problems even in a windows fresh install. The solution is to reverse the procedure: VSCode -> Espressif IDF extension download and install IDF -> IDF-Tools download and install ESP-IDF + Espressif IDE.
> 
> Steps: Uninstall Espressif IDF extension, uninstall IDF-Tools, delete espressif folders (in C:\Espressif and in user folder if there is one), then install Espressif IDF extension, choose download and install IDF option , install and when it completes, check that everything works fine. Then you can install IDF-Tools and Espressif IDE with the standard procedure.
> 
> I think the only drawback is that you'll be left with 2 IDF installations, but I've already spent too many hours or days on this solution and it's fine for me. Maybe you can save the duplicate installation if when opening IDF-Tools installer you choose the folder where Espressif IDF extension already installed ESP-IDF, but I haven't checked and you risk facing some possible errors.
> 
> It is also very likely that you will face other errors, such as Espressif IDE not allowing you to create projects as it will not find the IDF-Tools installation folder, or in VSCode you cannot run idf.py menuconfig, or compilation errors in Espressif IDE. All this I mentioned in 3 comments on a reddit post. [Here](https://www.reddit.com/r/esp32/comments/161tdf2/comment/jxx47qe/?utm_source=share&utm_medium=web2x&context=3) you can read it.

Hi @crispmfc
Thanks for response but i know the problems if i install esp-idf manually and try to use in vs code. And as you said, i deleted everything about esp idf before try installing with another way. I hope your response will help others. It looks like i get all kind of weird errors somehow :)

HasanYektaDemirtas, 2023-09-01:
Hi @tom-borcin
My problem is not related with only vs code extension. It includes ESPRESSIF IDE too.

crispmfc, 2023-09-02:
Usually part of some problems could be because the necessary folders were not added to the PATH in windows environment variables. I don't know if this is the case, but when I go to my pc I will tell you the folders that I set. This also has to do with fixing some installation errors.

tom-borcin, 2023-09-04:
Hi @HasanYektaDemirtas, I had to transfer your issue to VSCode extension repository, because ESP-IDF repository wasn't correct. Don't worry, our engineers will take care of it even if it spans across multiple repositories.

radurentea, 2023-09-04:
Hi @HasanYektaDemirtas,

For VSCode, I've installed the same version on my windows machine and I could not reproduce the issue.

I suggest to do a clean reinstall of the extension as explained in this [guide](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/INSTALL.md#uninstalling-the-plugin).

If the issue persists, please give me the new logs.

HasanYektaDemirtas, 2023-09-05:
Hi @radurentea 
I tried to install again after delete everything about esp-idf and extension but result is same with the log i shared before.

radurentea, 2023-09-11:
SSL/TLS connections can fail if the system's date and time are too far off, so lets check that first:

- Right-click on the time and date displayed on the taskbar at the bottom right corner of the screen.

- Select Adjust date/time.

- In the Date & time tab, turn on the option Set time automatically. This will synchronize your computer's clock with Microsoft's time server.

If you're time is ok, ensure that no firewall or security software is blocking the connection or causing issues with SSL/TLS.

Let me know if it helps

closed 2023-10-02 by github-actions[bot]