espressif/vscode-esp-idf-extension #1036: ERROR: ./bin/python doesn't exist! Please run the install script or "idf_tools.py install-python-env" in order to create it

daniel-frenkel, 2023-09-06:
### OS

Linux

### Operating System version

Ubuntu 22.04.3 LTS

### Visual Studio Code version

1.81.1

### ESP-IDF version

5.1.1

### Python version

3.10

### Doctor command output

N/A. Doctor does not work.

### Description

When using the VS Code plugin to run ESP-IDF SDK Configuration Editor (menuconfig), the following error is output:

ERROR: ./bin/python doesn't exist! Please run the install script or "idf_tools.py install-python-env" in order to create it

This is only an issue in VS Code and not when using the terminal. 

For some reason, VS Code is looking for python in the wrong location, even though I have specified it to look in the python_env folder during installation.

### Debug Message

```plain
ERROR: ./bin/python doesn't exist! Please run the install script or "idf_tools.py install-python-env" in order to create it
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

closed 2023-09-07 by daniel-frenkel

Erhannis, 2024-09-17:
@daniel-frenkel What did the problem turn out to be?  I'm getting this, too.

daniel-frenkel, 2024-09-17:
@Erhannis it's been a while, but I believe that I went into the ESP-IDF extension settings and manually set the path to the right directory