espressif/vscode-esp-idf-extension #1037: Installation loop on macOS without Rosetta

mickeyl, 2023-09-08:
### OS

Windows

### Operating System version

macOS 13.5.2 (22G91)

### Visual Studio Code version

1.82.0

### ESP-IDF version

5.1.1

### Python version

3.9.6

### Doctor command output

not available, since the installation does not finish.

### Description

The installation never finishes. It starts downloading the tools and then reverts back to the first screen with "select download server", etc.

### Debug Message

```plain
idf_versions.txt has (1) KB
idf_versions.txt progress: 100.00% (0.08 / 0.08) KB
/usr/bin/git

/opt/homebrew/bin/cmake

/opt/homebrew/bin/ninja

Downloading ESP-IDF v5.1.1 (release version)...
esp-idf-v5.1.1.zip has (1104766) KB
{...}
esp-idf-v5.1.1.zip progress: 0.00% (1.35 / 1104765.32) KB
esp-idf-v5.1.1.zip progress: 0.00% (2.69 / 1104765.32) KB
esp-idf-v5.1.1.zip progress: 0.00% (4.04 / 1104765.32) KB
esp-idf-v5.1.1.zip progress: 0.00% (5.38 / 1104765.32) KB
esp-idf-v5.1.1.zip progress: 0.00% (6.73 / 1104765.32) KB
esp-idf-v5.1.1.zip progress: 0.00% (8.07 / 1104765.32) KB
esp-idf-v5.1.1.zip progress: 0.00% (9.42 / 1104765.32) KB
esp-idf-v5.1.1.zip progress: 0.00% (10.77 / 1104765.32) KB
esp-idf-v5.1.1.zip progress: 0.00% (12.11 / 1104765.32) KB
esp-idf-v5.1.1.zip progress: 0.00% (13.46 / 1104765.32) KB
esp-idf-v5.1.1.zip progress: 0.00% (14.80 / 1104765.32) KB
esp-idf-v5.1.1.zip progress: 0.00% (16.15 / 1104765.32) KB
esp-idf-v5.1.1.zip progress: 0.00% (17.49 / 1104765.32) KB
esp-idf-v5.1.1.zip progress: 0.00% (18.84 / 1104765.32) KB
esp-idf-v5.1.1.zip progress: 0.00% (20.19 / 1104765.32) KB
esp-idf-v5.1.1.zip progress: 0.00% (21.53 / 1104765.32) KB
esp-idf-v5.1.1.zip progress: 0.00% (22.88 / 1104765.32) KB
esp-idf-v5.1.1.zip progress: 0.00% (24.22 / 1104765.32) KB
esp-idf-v5.1.1.zip progress: 0.00% (25.57 / 1104765.32) KB
esp-idf-v5.1.1.zip progress: 0.00% (26.91 / 1104765.32) KB
esp-idf-v5.1.1.zip progress: 0.00% (28.26 / 1104765.32) KB
esp-idf-v5.1.1.zip progress: 0.00% (29.61 / 1104765.32) KB
esp-idf-v5.1.1.zip progress: 0.00% (30.95 / 1104765.32) KB
esp-idf-v5.1.1.zip progress: 0.00% (32.00 / 1104765.32) KB
esp-idf-v5.1.1.zip progress: 0.00% (33.35 / 1104765.32) KB
esp-idf-v5.1.1.zip progress: 0.00% (34.69 / 1104765.32) KB
esp-idf-v5.1.1.zip progress: 0.00% (36.04 / 1104765.32) KB
esp-idf-v5.1.1.zip progress: 0.00% (37.38 / 1104765.32) KB
[... 493 lines cut ...]
xtensa-esp-elf-gdb-12.1_20221002-aarch64-apple-darwin21.1.tar.gz progress: 0.04% (10.77 / 26949.62) KB
xtensa-esp-elf-gdb-12.1_20221002-aarch64-apple-darwin21.1.tar.gz progress: 0.04% (12.11 / 26949.62) KB
xtensa-esp-elf-gdb-12.1_20221002-aarch64-apple-darwin21.1.tar.gz progress: 0.05% (13.46 / 26949.62) KB
xtensa-esp-elf-gdb-12.1_20221002-aarch64-apple-darwin21.1.tar.gz progress: 0.05% (14.33 / 26949.62) KB
xtensa-esp-elf-gdb-12.1_20221002-aarch64-apple-darwin21.1.tar.gz progress: 0.06% (15.67 / 26949.62) KB
xtensa-esp-elf-gdb-12.1_20221002-aarch64-apple-darwin21.1.tar.gz progress: 0.06% (17.02 / 26949.62) KB
ENOENT: no such file or directory, open '/home/mickey/.espressif/dist/xtensa-esp-elf-gdb-12.1_20221002-aarch64-apple-darwin21.1.tar.gz'
xtensa-esp-elf-gdb-12.1_20221002-aarch64-apple-darwin21.1.tar.gz progress: 0.07% (18.36 / 26949.62) KB
...
xtensa-esp-elf-gdb-12.1_20221002-aarch64-apple-darwin21.1.tar.gz progress: 0.07% (19.71 / 26949.62) KB
xtensa-esp-elf-gdb-12.1_20221002-aarch64-apple-darwin21.1.tar.gz progress: 0.08% (21.05 / 26949.62) KB
xtensa-esp-elf-gdb-12.1_20221002-aarch64-apple-darwin21.1.tar.gz progress: 0.08% (22.40 / 26949.62) KB
...
xtensa-esp-elf-gdb-12.1_20221002-aarch64-apple-darwin21.1.tar.gz progress: 100.00% (26949.62 / 26949.62) KB

At this point, the installation window reverts to the first screen.
```

### Other Steps to Reproduce

Brand new installation of macOS and Visual Studio Code.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2023-09-12:
You either have connection issues or folder permission issues based on 

> Failed download. Retrying...
> Error: Error: Error creating dist directory
> Waiting 8 seconds...

mickeyl, 2023-09-12:
Here is the extension log:

[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/12585535/esp_idf_vsc_ext.log)

And here a screen cap of the installation process (dropbox link because of size, sorry):

[
](https://www.dropbox.com/scl/fi/r5vh5l11xlhnt226pqsjo/esp-idf-screencast.mov?rlkey=ntq28ylrutix5xz0t4qq6no2h&dl=0)

brianignacio5, 2023-09-14:
It seems the issue happens when creating the list directory @mickeyl Maybe some operation changed the permission of `/home/mickey/.espressif/dist` ? I would suggest to delete this `dist` and check the `.espressif` permission.

mickeyl, 2023-09-14:
I always try from scratch, that includes removing the `.espressif` subdirectory. Even creating it upfront (including the `dist` subdirectory) didn't work ­– just tried. `$HOME/.espressif/dist` is still empty after the loop. Are you sure that the extension is picking up the correct directory? 

`{"message":"Error creating dist directory","stack":"Error: Error creating dist directory\n\tat /dist/extension.js:2:1309092","level":"error","timestamp":"2023-09-12T10:41:50.560Z"}`

is unfortunately not very clear about a) where it actually tries to create the directory and b) which concrete error did happen.

radurentea, 2023-09-18:
This seems like a permissions issue  in **$HOME/.espressif**

First, I would suggest to make a clean reinstall of the extension following this [guide](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/INSTALL.md#uninstalling-the-plugin) which ensures to delete everything related to our extension.  (deleting also extension folder inside .vscode)

Try configuring again and if the issue persists, try to check permissions with `ls -ld $HOME/.espressif`

And let us know what the output is.

closed 2023-10-09 by github-actions[bot]