espressif/vscode-esp-idf-extension #1040: Not able to install IDF any veriosn Python dependency error

v-3n0m, 2023-09-16:
### OS

Linux

### Operating System version

POP os 22.04

### Visual Studio Code version

1.6.4

### ESP-IDF version

4.4.5

### Python version

3.10.12

### Doctor command output

/home/nikhi/esp/esp-idf/tools/check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
The following Python requirements are not satisfied:
pyparsing>=2.0.3,<2.4.0
To install the missing packages, please run "/home/nikhi/esp/esp-idf/install.sh"
Diagnostic information:
    IDF_PYTHON_ENV_PATH: /home/nikhi/.espressif/python_env/idf4.4_py3.10_env
    Python interpreter used: /home/nikhi/.espressif/python_env/idf4.4_py3.10_env/bin/python

ESP-IDF v4.4.5-dirty

### Description

I am not able to install any versions of IDF. after removing my previouly instlled idf version

### Debug Message

```plain
Python requirements has been installed
The following Python requirements are not satisfied:
pyparsing>=2.0.3,<2.4.0
To install the missing packages, please run "/home/nikhi/esp/esp-idf/install.sh"
Diagnostic information:
    IDF_PYTHON_ENV_PATH: /home/nikhi/.espressif/python_env/idf4.4_py3.10_env
    Python interpreter used: /home/nikhi/.espressif/python_env/idf4.4_py3.10_env/bin/python
/home/nikhi/esp/esp-idf/tools/check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Command failed: "/home/nikhi/.espressif/python_env/idf4.4_py3.10_env/bin/python" "/home/nikhi/esp/esp-idf/tools/check_python_dependencies.py" -r "/home/nikhi/esp/esp-idf/requirements.txt"
/home/nikhi/esp/esp-idf/tools/check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Command failed: "/home/nikhi/.espressif/python_env/idf4.4_py3.10_env/bin/python" "/home/nikhi/esp/esp-idf/tools/check_python_dependencies.py" -r "/home/nikhi/esp/esp-idf/requirements.txt"
/home/nikhi/esp/esp-idf/tools/check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Spaider, 2023-09-17:
I see exactly the same issue. Looks like `idf-component-manager` has unbound upper version for `pyparsing` and it installs version 3.0.9 which is the latest at the moment. But I may be wrong since it was very quick look into the problem.

dobairoland, 2023-09-18:
I believe the fix removing that deprecation error is not in ESP-IDF v4.4.5. It will be part of the next bugfix release. This is the fix: https://github.com/espressif/esp-idf/commit/fd96dafc2bd200273b6d0ccf61ffce4525921a75

The workaround is to use an older setuptools version: https://github.com/espressif/esp-idf/issues/11712#issuecomment-1617564515

dobairoland, 2023-09-18:
> I am not able to install any versions of IDF. after removing my previouly instlled idf version

The previous installation might cause the problem. It is possible that the installer is still linked to the old environment and it is not updating the right one. It could help to clean up the system by removing from all permanently set environment variables. And only after that run the installer.

This command might help:
```python
/home/nikhi/.espressif/python_env/idf4.4_py3.10_env/bin/python -m pip install "pyparsing>=2.0.3,<2.4.0"
```
but you might hit another issue later on if you don't clean up your system.

lhxzui, 2023-09-20:
I have encountered the same problem.
It looks like that the file espidf.constraints.txt in path '/home/rfq/.vscode/extensions/espressif.esp-idf-extension-1.6.4' be used when install within vscode. The line 28 'pyparsing>=3.0.3,<3.1' in espidf.constraints.txt specifys the version of pyparsing. 
First I run './install.sh' in path esp-idf local. This step will install pyparsing which esp-idf-release-v4.4 requests in file '/home/rfq/esp/esp-idf-release-v4.4/requirements.txt' line 21 'pyparsing>=2.0.3,<2.4.0'. Second, disable The line 28 'pyparsing>=3.0.3,<3.1' in espidf.constraints.txt. Then install esp-idf use vscode-esp-idf-extension within vscode. It works normally.
So, why not use pyparsing which esp-idf-release-v4.4 requests in file '/home/rfq/esp/esp-idf-release-v4.4/requirements.txt' when select esp-idf-release-v4.4?

lhxzui, 2023-09-20:
> > I am not able to install any versions of IDF. after removing my previouly instlled idf version
> 
> The previous installation might cause the problem. It is possible that the installer is still linked to the old environment and it is not updating the right one. It could help to clean up the system by removing from all permanently set environment variables. And only after that run the installer.
> 
> This command might help:
> 
> ```shell
> /home/nikhi/.espressif/python_env/idf4.4_py3.10_env/bin/python -m pip install "pyparsing>=2.0.3,<2.4.0"
> ```
> 
> but you might hit another issue later on if you don't clean up your system.

This command is can't solve the problem.

Spaider, 2023-09-20:
Well, it kinda works for my installation.

First thing, you should replace `/home/nikhi/` with your home dir or with `$HOME`. Second, add `--force` command line param to force-replace installed version. 
Third, you have to run this from within IDF Python environment, not from global scope.

Those things might be obvious for some people, I noted them just in case ;)

radurentea, 2023-09-26:
Just added a fix. Configuring the extension should not throw any error now

lhxzui, 2023-09-26:
> Just added a fix. Configuring the extension should not throw any error now

How to do?

Spaider, 2023-09-26:
@radurentea this bug is specifically for version 4.4.5 but your commit fixes version 5.1 and above.
Is there a way to back-port it to 4.4.5?

radurentea, 2023-09-26:
Sorry for the confusion, I've updated my comment and  added the link for installing the version of the extension where this should be fixed, until we release a new version.

It doesn't have anything to do with the esp idf, so it should work if you use the version of the extension that i provided with the vsix file.

> > Just added a fix. Configuring the extension should not throw any error now
> 
> How to do?

Install the new  extension version from the vsix file I've edited in my prev reply, following these steps:

1. Open Visual Studio Code.
2. Go to Extensions:
3. Use the keyboard shortcut Ctrl+Shift+X (Windows/Linux) or Cmd+Shift+X (macOS).
4. Alternatively, you can click on the Extensions icon in the Activity Bar on the side of the window.
5. Install from VSIX:
6. In the Extensions view, click on the More button (...) in the top menu.
7. Select Install from VSIX... from the dropdown menu.
8. Navigate to the .vsix file and select it.
9. Click the Install button in the dialog that appears to confirm.

lhxzui, 2023-09-26:
> Sorry for the confusion, I've updated my comment and added the link for installing the version of the extension where this should be fixed, until we release a new version.
>
> It doesn't have anything [...]

Thanks.

AliMoal, 2023-09-26:
I need to downgrade to v4.3.6 but this problem exists...

radurentea, 2023-10-05:
@AliMoal , please follow the steps that I've provided, should fix the issue until we release the fix.

closed 2023-10-05 by radurentea

AliMoal, 2024-01-13:
@radurentea Hi again, 
Could you send the files to me again? GitHub has deleted the files you sent.
Thanks.