espressif/vscode-esp-idf-extension #1041: 网络环境不好的情况下，用idf 5.1.1的完整安装包安装后，无法成功配置VSCode扩展 (IDFGH-11077)

guoxiaoqiao, 2023-09-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1.1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

正常配置code的IDF扩展

### What is the actual behavior?

提示以下错误
"c:\idf511\tools\idf-python\3.11.2\python.exe -m pip" is not valid. (ERROR_INVALID_PIP)

### Steps to reproduce.

1、下载：https://dl.espressif.com/dl/idf-installer/esp-idf-tools-setup-offline-5.1.1.exe
2、断网安装？我的是联网的，但是网络环境不好，可能与python库服务器在国外有关

### Build or installation Logs.

```plain
idf_versions.txt has (1) KB
idf_versions.txt progress: 100.00% (0.08 / 0.08) KB
Installing ESP-IDF extension python packages in G:\esp32\idf-tools\python_env\idf5.1_py3.11_env\Scripts\python.exe ...

spawn C:\WINDOWS\system32\cmd.exe ENOENTidf_versions.txt has (1) KB
idf_versions.txt progress: 100.00% (0.08 / 0.08) KB
idf_versions.txt has (1) KB
idf_versions.txt progress: 100.00% (0.08 / 0.08) KB
Using existing c:\idf511\dist\idf-git-2.39.2-win64.zip
Installing c:\idf511\tools\idf-git\2.39.2 ...
Warning File c:\idf511\tools\idf-git\2.39.2\cmd\git.exe
                                        already exists and was not updated.
Warning File c:\idf511\tools\idf-git\2.39.2\cmd\scalar.exe
                                        already exists and was not updated.
Warning File c:\idf511\tools\idf-git\2.39.2\usr\bin\awk.exe
                                        already exists and was not updated.
Warning File c:\idf511\tools\idf-git\2.39.2\usr\bin\b2sum.exe
                                        already exists and was not updated.
Warning File c:\idf511\tools\idf-git\2.39.2\usr\bin\basename.exe
                                        already exists and was not updated.
Warning File c:\idf511\tools\idf-git\2.39.2\usr\bin\basenc.exe
                                        already exists and was not updated.
Warning File c:\idf511\tools\idf-git\2.39.2\usr\bin\cat.exe
                                        already exists and was not updated.
Warning File c:\idf511\tools\idf-git\2.39.2\usr\bin\chattr.exe
                                        already exists and was not updated.
Warning File c:\idf511\tools\idf-git\2.39.2\usr\bin\cmp.exe
                                        already exists and was not updated.
Warning File c:\idf511\tools\idf-git\2.39.2\usr\bin\comm.exe
                                        already exists and was not updated.
Warning File c:\idf511\tools\idf-git\2.39.2\usr\bin\cp.exe
[... 218 lines cut ...]
                                        already exists and was not updated.
Warning File c:\idf511\tools\idf-git\2.39.2\usr\share\licenses\openssh\LICENCE
                                        already exists and was not updated.
Warning File c:\idf511\tools\idf-git\2.39.2\usr\share\licenses\openssl\LICENSE
                                        already exists and was not updated.
Warning File c:\idf511\tools\idf-git\2.39.2\usr\share\licenses\zlib\LICENSE
                                        already exists and was not updated.
Warning File c:\idf511\tools\idf-git\2.39.2\usr\share\nano\git.nanorc
                                        already exists and was not updated.
Warning File c:\idf511\tools\idf-git\2.39.2\usr\ssl\ct_log_list.cnf
                                        already exists and was not updated.
Warning File c:\idf511\tools\idf-git\2.39.2\usr\ssl\ct_log_list.cnf.dist
                                        already exists and was not updated.
Warning File c:\idf511\tools\idf-git\2.39.2\usr\ssl\openssl.cnf
                                        already exists and was not updated.
Warning File c:\idf511\tools\idf-git\2.39.2\usr\ssl\openssl.cnf.dist
                                        already exists and was not updated.
Extracted c:\idf511\tools\idf-git\2.39.2 ...
Using existing c:\idf511\tools\idf-python\3.11.2
"c:\idf511\tools\idf-python\3.11.2\python.exe -m pip" is not valid. (ERROR_INVALID_PIP)
```

### More Information.

[image]

brianignacio5, 2023-09-18:
This is already fix on GitHub master`. You can try it with this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/workflows/ci/master/esp-idf-extension.vsix.zip)

larryli, 2023-09-20:
> 令人抓狂的dirty!!!!!! [image]

dirty 是在 windows 下的问题。restore submodule 就好了。

guoxiaoqiao, 2023-09-20:
[image]

[image]

更新了还是不行

brianignacio5, 2023-10-12:
I've tested with current master vsix [here](https://nightly.link/espressif/vscode-esp-idf-extension/workflows/ci/master/esp-idf-extension.vsix.zip) for ESP-IDF v4.4.6 and v5.1.1 and both seem to be working fine for a clean Windows 11 environment.

Can you try updated vsix ?

guoxiaoqiao, 2023-10-26:
[image]

[image]

[image]

[image]

I used the updated version of the plug-in market on October 13, but it still cannot be used. I guess it may be related to the fact that my Win11 is the latest preview version.

brianignacio5, 2023-11-01:
Can you share the extension log here:

Check log file which can be obtained from:
Windows: %USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log
Linux & MacOSX: $HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log

brianignacio5, 2023-11-13:
Closing this issue due to lack of activity. Comment here if you need more help.

closed 2023-11-13 by brianignacio5