espressif/vscode-esp-idf-extension #1043: SDK Configuration Editor (menuconfig): Failed to save to "sdkconfig_debug"

vdm97, 2023-09-20:
### OS

Windows

### Operating System version

Windows 10

### Visual Studio Code version

1.82.1

### ESP-IDF version

5.1.1

### Python version

3.11.2

### Doctor command output

```
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19045 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:<HOMEPATH>Program Files (x86)<HOMEPATH>Common Files<HOMEPATH>Oracle<HOMEPATH>Java<HOMEPATH>javapath;C:<HOMEPATH>Program Files<HOMEPATH>Python39<HOMEPATH>Scripts<HOMEPATH>;C:<HOMEPATH>Program Files<HOMEPATH>Python39<HOMEPATH>;C:<HOMEPATH>WINDOWS<HOMEPATH>system32;C:<HOMEPATH>WINDOWS;C:<HOMEPATH>WINDOWS<HOMEPATH>System32<HOMEPATH>Wbem;C:<HOMEPATH>WINDOWS<HOMEPATH>System32<HOMEPATH>WindowsPowerShell<HOMEPATH>v1.0<HOMEPATH>;C:<HOMEPATH>WINDOWS<HOMEPATH>System32<HOMEPATH>OpenSSH<HOMEPATH>;C:<HOMEPATH>Program Files<HOMEPATH>dotnet<HOMEPATH>;C:<HOMEPATH>Program Files<HOMEPATH>Microsoft VS Code<HOMEPATH>bin;C:<HOMEPATH>Program Files<HOMEPATH>Microsoft SQL Server<HOMEPATH>130<HOMEPATH>Tools<HOMEPATH>Binn<HOMEPATH>;C:<HOMEPATH>Program Files (x86)<HOMEPATH>GtkSharp<HOMEPATH>2.12<HOMEPATH>bin;C:<HOMEPATH>Program Files<HOMEPATH>Microsoft SQL Server<HOMEPATH>Client SDK<HOMEPATH>ODBC<HOMEPATH>170<HOMEPATH>Tools<HOMEPATH>Binn<HOMEPATH>;C:<HOMEPATH>Program Files<HOMEPATH>PuTTY<HOMEPATH>;C:<HOMEPATH>Users<HOMEPATH>user<HOMEPATH>AppData<HOMEPATH>Roaming<HOMEPATH>nvm;C:<HOMEPATH>Program Files<HOMEPATH>nodejs;C:<HOMEPATH>Program Files<HOMEPATH>TortoiseSVN<HOMEPATH>bin;C:<HOMEPATH>Program Files (x86)<HOMEPATH>Nordic Semiconductor<HOMEPATH>nrf-command-line-tools<HOMEPATH>bin<HOMEPATH>;C:<HOMEPATH>Program Files (x86)<HOMEPATH>Gpg4win<HOMEPATH>..<HOMEPATH>GnuPG<HOMEPATH>bin;C:<HOMEPATH>ProgramData<HOMEPATH>chocolatey<HOMEPATH>bin;C:<HOMEPATH>android<HOMEPATH>platform-tools;C:<HOMEPATH>Strawberry<HOMEPATH>c<HOMEPATH>bin;C:<HOMEPATH>Strawberry<HOMEPATH>perl<HOMEPATH>site<HOMEPATH>bin;C:<HOMEPATH>Strawberry<HOMEPATH>perl<HOMEPATH>bin;C:<HOMEPATH>Program Files<HOMEPATH>TortoiseGit<HOMEPATH>bin;C:<HOMEPATH>Program Files<HOMEPATH>Git<HOMEPATH>cmd;C:<HOMEPATH>Program Files (x86)<HOMEPATH>Windows Kits<HOMEPATH>10<HOMEPATH>Windows Performance Toolkit<HOMEPATH>;C:<HOMEPATH>Users<HOMEPATH>user<HOMEPATH>.cargo<HOMEPATH>bin;C:<HOMEPATH>Users<HOMEPATH>user<HOMEPATH>AppData<HOMEPATH>Local<HOMEPATH>Microsoft<HOMEPATH>WindowsApps;C:<HOMEPATH>Users<HOMEPATH>user<HOMEPATH>AppData<HOMEPATH>Roaming<HOMEPATH>nvm;C:<HOMEPATH>Program Files<HOMEPATH>nodejs;C:<HOMEPATH>svn<HOMEPATH>P_ARM<HOMEPATH>ESP32;C:<HOMEPATH>Users<HOMEPATH>user<HOMEPATH>.dotnet<HOMEPATH>tools;C:<HOMEPATH>Users<HOMEPATH>user<HOMEPATH>.dotnet<HOMEPATH>tools;C:<HOMEPATH>Users<HOMEPATH>user<HOMEPATH>AppData<HOMEPATH>Roaming<HOMEPATH>Python<HOMEPATH>Python39<HOMEPATH>Scripts;C:<HOMEPATH>Program Files<HOMEPATH>OpenSSL-Win64<HOMEPATH>bin; 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.82.1 
Visual Studio Code language en 
Visual Studio Code shell C:<HOMEPATH>WINDOWS<HOMEPATH>System32<HOMEPATH>WindowsPowerShell<HOMEPATH>v1.0<HOMEPATH>powershell.exe 
ESP-IDF Extension version 1.6.4 
Workspace folder c:<HOMEPATH>svn<HOMEPATH>P_ARM<HOMEPATH>ESP32 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:<HOMEPATH>Users<HOMEPATH>user<HOMEPATH>esp<HOMEPATH>esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) c:<HOMEPATH>Users<HOMEPATH>user<HOMEPATH>esp<HOMEPATH>esp-idf-tools<HOMEPATH>tools<HOMEPATH>xtensa-esp-elf-gdb<HOMEPATH>12.1_20221002<HOMEPATH>xtensa-esp-elf-gdb<HOMEPATH>bin;c:<HOMEPATH>Users<HOMEPATH>user<HOMEPATH>esp<HOMEPATH>esp-idf-tools<HOMEPATH>tools<HOMEPATH>riscv32-esp-elf-gdb<HOMEPATH>12.1_20221002<HOMEPATH>riscv32-esp-elf-gdb<HOMEPATH>bin;c:<HOMEPATH>Users<HOMEPATH>user<HOMEPATH>esp<HOMEPATH>esp-idf-tools<HOMEPATH>tools<HOMEPATH>xtensa-esp32-elf<HOMEPATH>esp-12.2.0_20230208<HOMEPATH>xtensa-esp32-elf<HOMEPATH>bin;c:<HOMEPATH>Users<HOMEPATH>user<HOMEPATH>esp<HOMEPATH>esp-idf-tools<HOMEPATH>tools<HOMEPATH>xtensa-esp32s2-elf<HOMEPATH>esp-12.2.0_20230208<HOMEPATH>xtensa-esp32s2-elf<HOMEPATH>bin;c:<HOMEPATH>Users<HOMEPATH>user<HOMEPATH>esp<HOMEPATH>esp-idf-tools<HOMEPATH>tools<HOMEPATH>xtensa-esp32s3-elf<HOMEPATH>esp-12.2.0_20230208<HOMEPATH>xtensa-esp32s3-elf<HOMEPATH>bin;c:<HOMEPATH>Users<HOMEPATH>user<HOMEPATH>esp<HOMEPATH>esp-idf-tools<HOMEPATH>tools<HOMEPATH>riscv32-esp-elf<HOMEPATH>esp-12.2.0_20230208<HOMEPATH>riscv32-esp-elf<HOMEPATH>bin;c:<HOMEPATH>Users<HOMEPATH>user<HOMEPATH>esp<HOMEPATH>esp-idf-tools<HOMEPATH>tools<HOMEPATH>esp32ulp-elf<HOMEPATH>2.35_20220830<HOMEPATH>esp32ulp-elf<HOMEPATH>bin;c:<HOMEPATH>Users<HOMEPATH>user<HOMEPATH>esp<HOMEPATH>esp-idf-tools<HOMEPATH>tools<HOMEPATH>cmake<HOMEPATH>3.24.0<HOMEPATH>bin;c:<HOMEPATH>Users<HOMEPATH>user<HOMEPATH>esp<HOMEPATH>esp-idf-tools<HOMEPATH>tools<HOMEPATH>openocd-esp32<HOMEPATH>v0.12.0-esp32-20230419<HOMEPATH>openocd-esp32<HOMEPATH>bin;c:<HOMEPATH>Users<HOMEPATH>user<HOMEPATH>esp<HOMEPATH>esp-idf-tools<HOMEPATH>tools<HOMEPATH>ninja<HOMEPATH>1.10.2;c:<HOMEPATH>Users<HOMEPATH>user<HOMEPATH>esp<HOMEPATH>esp-idf-tools<HOMEPATH>tools<HOMEPATH>idf-exe<HOMEPATH>1.0.3;c:<HOMEPATH>Users<HOMEPATH>user<HOMEPATH>esp<HOMEPATH>esp-idf-tools<HOMEPATH>tools<HOMEPATH>ccache<HOMEPATH>4.8<HOMEPATH>ccache-4.8-windows-x86_64;c:<HOMEPATH>Users<HOMEPATH>user<HOMEPATH>esp<HOMEPATH>esp-idf-tools<HOMEPATH>tools<HOMEPATH>dfu-util<HOMEPATH>0.11<HOMEPATH>dfu-util-0.11-win64;c:<HOMEPATH>Users<HOMEPATH>user<HOMEPATH>esp<HOMEPATH>esp-idf-tools<HOMEPATH>tools<HOMEPATH>esp-rom-elfs<HOMEPATH>20230320
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: c:<HOMEPATH>Users<HOMEPATH>user<HOMEPATH>esp<HOMEPATH>esp-idf-tools<HOMEPATH>tools<HOMEPATH>openocd-esp32<HOMEPATH>v0.12.0-esp32-20230419/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
    ESP_ROM_ELF_DIR: c:<HOMEPATH>Users<HOMEPATH>user<HOMEPATH>esp<HOMEPATH>esp-idf-tools<HOMEPATH>tools<HOMEPATH>esp-rom-elfs<HOMEPATH>20230320/
Virtual env Python Path (idf.pythonBinPath) c:<HOMEPATH>Users<HOMEPATH>user<HOMEPATH>esp<HOMEPATH>esp-idf-tools<HOMEPATH>python_env<HOMEPATH>idf5.1_py3.11_env<HOMEPATH>Scripts<HOMEPATH>python.exe
Serial port (idf.port) COM3
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,target/esp32.cfg,target/esp_common.cfg
ESP-IDF Tools Path (idf.toolsPath) c:<HOMEPATH>Users<HOMEPATH>user<HOMEPATH>esp<HOMEPATH>esp-idf-tools
Git Path (idf.gitPath) c:<HOMEPATH>Users<HOMEPATH>user<HOMEPATH>esp<HOMEPATH>esp-idf-tools<HOMEPATH>tools<HOMEPATH>idf-git<HOMEPATH>2.39.2<HOMEPATH>cmd<HOMEPATH>git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to c:<HOMEPATH>Users<HOMEPATH>user<HOMEPATH>esp<HOMEPATH>esp-idf-tools<HOMEPATH>tools<HOMEPATH>xtensa-esp-elf-gdb<HOMEPATH>12.1_20221002<HOMEPATH>xtensa-esp-elf-gdb<HOMEPATH>bin: true
Access to c:<HOMEPATH>Users<HOMEPATH>user<HOMEPATH>esp<HOMEPATH>esp-idf-tools<HOMEPATH>tools<HOMEPATH>riscv32-esp-elf-gdb<HOMEPATH>12.1_20221002<HOMEPATH>riscv32-esp-elf-gdb<HOMEPATH>bin: true
Access to c:<HOMEPATH>Users<HOMEPATH>user<HOMEPATH>esp<HOMEPATH>esp-idf-tools<HOMEPATH>tools<HOMEPATH>xtensa-esp32-elf<HOMEPATH>esp-12.2.0_20230208<HOMEPATH>xtensa-esp32-elf<HOMEPATH>bin: true
Access to c:<HOMEPATH>Users<HOMEPATH>user<HOMEPATH>esp<HOMEPATH>esp-idf-tools<HOMEPATH>tools<HOMEPATH>xtensa-esp32s2-elf<HOMEPATH>esp-12.2.0_20230208<HOMEPATH>xtensa-esp32s2-elf<HOMEPATH>bin: true
Access to c:<HOMEPATH>Users<HOMEPATH>user<HOMEPATH>esp<HOMEPATH>esp-idf-tools<HOMEPATH>tools<HOMEPATH>xtensa-esp32s3-elf<HOMEPATH>esp-12.2.0_20230208<HOMEPATH>xtensa-esp32s3-elf<HOMEPATH>bin: true
Access to c:<HOMEPATH>Users<HOMEPATH>user<HOMEPATH>esp<HOMEPATH>esp-idf-tools<HOMEPATH>tools<HOMEPATH>riscv32-esp-elf<HOMEPATH>esp-12.2.0_20230208<HOMEPATH>riscv32-esp-elf<HOMEPATH>bin: true
[... cut ...]
I assume that the i get the error because my `sdkconfig` is not named `sdkconfig` but `sdkconfig_debug`. Is there a way to fix this error? I'm afraid that at some point in the future the version 1.5.x will be incompatible with the newest ESP-IDF toolchain. Furthermore changing between 1.5.x and 1.6.x is annoying.

### Debug Message

```plain
Provided in Description.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2023-09-21:
Have you tried using the `idf.sdkconfigFilePath` settings in your project `.vscode/settings.json` ?

Maybe to something like `c:\path\to\project\sdkconfig`

vdm97, 2023-09-21:
Hi, thank you for the suggestion. I added `"idf.sdkconfigFilePath": "${workspaceFolder}/sdkconfig"` to `.vscode\settings.json`, but unfortunately the "error" ist still present. The following is the output:
```
[SDK Configuration Editor]
-- Project sdkconfig file C:/svn/path/to/sdkconfig_debug

[SDK Configuration Editor]
Loading defaults file C:/svn/path/to/sdkconfig_debug...

...

[SDK Configuration Editor]
{"version":2,"save":"\"sdkconfig_debug\""}
[SDK Configuration Editor]
Saving config to "sdkconfig_debug"...

[SDK Configuration Editor]
---------------------------ERROR--------------------------
Error: Failed to save to "sdkconfig_debug": [Errno 22] Invalid argument: '"sdkconfig_debug"'

-----------------------END OF ERROR-----------------------
[SDK Configuration Editor]
{"version": 2, "values": {}, "ranges": {}, "visible": {}, "error": ["Failed to save to \"sdkconfig_debug\": [Errno 22] Invalid argument: '\"sdkconfig_debug\"'"]}
```

Somehow the setting is ignored?

brianignacio5, 2024-02-06:
Sorry for late reply. I've looked at the code of CMakeLists.txt again and there is something odd

You are setting SDKCONFIG first to debug or release and then using SDKCONFIG to set SDKCONFIG_DEFAULTS as `set(SDKCONFIG_DEFAULTS ${SDKCONFIG})` but if you look at the [esp-idf multi config example](https://github.com/espressif/esp-idf/tree/master/examples/build_system/cmake/multi_config) what you set to debug or release is SDKCONFIG_DEFAULTS first and keep the sdkconfig in a build relative location.

For example SDKCONFIG_DEFAULTS="sdkconfig.prod_common;sdkconfig.prod1" will yield sdkconfig for product 1 and SDKCONFIG_DEFAULTS="sdkconfig.prod_common;sdkconfig.prod2" for product 2.

In your use case you will have something like:

1. SDKCONFIG_DEFAULTS="sdkconfig_debug" for debug case and SDKCONFIG_DEFAULTS="sdkconfig_release" for release. 

SDKCONFIG_DEFAULTS are used to generate sdkconfig files for distinct scenarios. You want a single sdkconfig (as far as I can understand) but generate for 2 different cases. More info [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#custom-sdkconfig-defaults)

You probably want to set multiple profiles for these. Take a look at our [project configuration tutorial](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/project_configuration.md) to a similar example as [esp-idf multi config example](https://github.com/espressif/esp-idf/tree/master/examples/build_system/cmake/multi_config) but done with our vscode extension profiles.

closed 2024-02-06 by brianignacio5