espressif/vscode-esp-idf-extension #1044: Compilation fails with IDF 4.4.4 and pyparsing v3.0.9 (but pyparsing v2.3.1 works!)

gonzabrusco, 2023-09-20:
### OS

Windows

### Operating System version

Windows 10

### ESP-IDF version

4.4.4

### Python version

3.8

### Doctor command output

-

### Description

Hello. I've found a bug in the extension. I installed the IDF environment using this extension in one PC (using this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/workflows/ci/master/esp-idf-extension.vsix.zip) but from three weeks ago) and everything worked perfectly.

We tried to install it again today (downloading the vsix again from the same url) in another PC with Windows 10 and installed successfuly but compilation failed with this message:

```
FAILED: esp-idf/esp_system/ld/sections.ld C:/Users/Julian/Documents/Firmware/powermeter-smart-pic/build/esp-idf/esp_system/ld/sections.ld 
cmd.exe /C "cd /D C:\Users\Julian\Documents\Firmware\powermeter-smart-pic\build\esp-idf\esp_system && c:\Espressif\4.4.4\python_env\idf4.4_py3.8_env\Scripts\python.exe C:/Espressif/4.4.4/esp-idf/tools/ldgen/ldgen.py --config C:/Users/Julian/Documents/Firmware/powermeter-smart-pic/sdkconfig --fragments C:/Espressif/4.4.4/esp-idf/components/esp_ringbuf/linker.lf      C:/Espressif/4.4.4/esp-idf/components/driver/linker.lf  C:/Espressif/4.4.4/esp-idf/components/esp_pm/linker.lf  C:/Espressif/4.4.4/esp-idf/components/spi_flash/linker.lf       C:/Espressif/4.4.4/esp-idf/components/esp_gdbstub/linker.lf     C:/Espressif/4.4.4/esp-idf/components/espcoredump/linker.lf     C:/Espressif/4.4.4/esp-idf/components/esp_phy/linker.lf C:/Espressif/4.4.4/esp-idf/components/esp_system/linker.lf      C:/Espressif/4.4.4/esp-idf/components/esp_system/app.lf C:/Espressif/4.4.4/esp-idf/components/hal/linker.lf     C:/Espressif/4.4.4/esp-idf/components/esp_event/linker.lf       C:/Espressif/4.4.4/esp-idf/components/esp_wifi/linker.lf        C:/Espressif/4.4.4/esp-idf/components/lwip/linker.lf    C:/Espressif/4.4.4/esp-idf/components/log/linker.lf     C:/Espressif/4.4.4/esp-idf/components/heap/linker.lf    C:/Espressif/4.4.4/esp-idf/components/soc/linker.lf     C:/Espressif/4.4.4/esp-idf/components/esp_hw_support/linker.lf  C:/Espressif/4.4.4/esp-idf/components/xtensa/linker.lf  C:/Espressif/4.4.4/esp-idf/components/esp_common/common.lf      C:/Espressif/4.4.4/esp-idf/components/esp_common/soc.lf C:/Espressif/4.4.4/esp-idf/components/freertos/linker.lf        C:/Espressif/4.4.4/esp-idf/components/newlib/newlib.lf  C:/Espressif/4.4.4/esp-idf/components/newlib/system_libs.lf     C:/Espressif/4.4.4/esp-idf/components/app_trace/linker.lf       C:/Espressif/4.4.4/esp-idf/components/bt/linker.lf      C:/Espressif/4.4.4/esp-idf/components/esp_lcd/linker.lf --input C:/Espressif/4.4.4/esp-idf/components/esp_system/ld/esp32/sections.ld.in --output C:/Users/Julian/Documents/Firmware/powermeter-smart-pic/build/esp-idf/esp_system/ld/sections.ld --kconfig C:/Espressif/4.4.4/esp-idf/Kconfig --env-file C:/Users/Julian/Documents/Firmware/powermeter-smart-pic/build/config.env --libraries-file C:/Users/Julian/Documents/Firmware/powermeter-smart-pic/build/ldgen_libraries --objdump C:/Espressif/4.4.4/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-objdump.exe"
linker script generation failed for C:/Espressif/4.4.4/esp-idf/components/esp_system/ld/esp32/sections.ld.in
ERROR: failed to parse C:/Espressif/4.4.4/esp-idf/components/esp_phy/linker.lf
Expected end of text, found 'if'  (at char 0), (line:1, col:1)
ninja: build stopped: subcommand failed.

 *  The terminal process "C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe -Command ninja " terminated with exit code: 1.
```

Comparing both python environments (on both PCs) we found that the culprit was pyparsing v3.0.9.

These are the packages installed on the NON working environment (pyparsing==3.0.9):

```
(idf4.4_py3.8_env) PS C:\Espressif\4.4.4\python_env\idf4.4_py3.8_env> pip freeze --local
bitstring==3.1.9
Brotli==1.1.0
cachecontrol==0.13.1
certifi==2023.7.22
cffi==1.15.1
charset-normalizer==3.2.0
click==8.0.4
colorama==0.4.6
construct==2.10.54
contextlib2==21.6.0
cryptography==41.0.4
ecdsa==0.18.0
esp-debug-backend==1.0.3
esp-windows-curses @ file://c/:\Espressif\4.4.4\esp-idf/tools/kconfig_new/esp-windows-curses
filelock==3.12.4
Flask==0.12.5
Flask-Compress==1.14
Flask-SocketIO==2.9.6
future==0.18.3
gcovr==6.0
gdbgui==0.13.2.0
gevent==1.5.0
greenlet==2.0.2
idf-component-manager==1.4.0
idna==3.4
itsdangerous==2.0.1
Jinja2==3.0.3
kconfiglib==13.7.1
lxml==4.9.3
MarkupSafe==2.1.3
msgpack==1.0.5
packaging==23.1
pycparser==2.21
pyelftools==0.30
pygdbmi==0.9.0.2
Pygments==2.16.1
pyparsing==3.0.9
pyserial==3.5
python-engineio==3.14.2
python-socketio==4.6.1
pywin32==306
PyYAML==6.0.1
reedsolo==1.5.4
requests==2.31.0
requests-file==1.5.1
requests-toolbelt==1.0.0
schema==0.7.5
six==1.16.0
tqdm==4.66.1
urllib3==1.26.16
websocket-client==1.6.3
Werkzeug==0.16.1
windows-curses==2.3.1
```

And these are the packages on the working environment (pyparsing==2.3.1):

```
(idf4.4_py3.8_env) PS C:\Espressif\4.4.4\python_env\idf4.4_py3.8_env> pip freeze --local
bitstring==3.1.9
Brotli==1.0.9
cachecontrol==0.13.1
certifi==2023.7.22
cffi==1.15.1
charset-normalizer==3.2.0
click==8.0.4
colorama==0.4.6
construct==2.10.54
contextlib2==21.6.0
cryptography==41.0.3
ecdsa==0.18.0
esp-debug-backend==1.0.3
esp-windows-curses @ file://c:\Espressif\4.4.4\esp-idf/tools/kconfig_new/esp-windows-curses
filelock==3.12.3
Flask==0.12.5
Flask-Compress==1.13
Flask-SocketIO==2.9.6
future==0.18.3
gcovr==6.0
gdbgui==0.13.2.0
gevent==1.5.0
greenlet==2.0.2
idf-component-manager==1.3.2
idna==3.4
itsdangerous==2.0.1
Jinja2==3.0.3
kconfiglib==13.7.1
lxml==4.9.3
MarkupSafe==2.1.3
msgpack==1.0.5
packaging==23.1
psutil==5.9.5
pycparser==2.21
pyelftools==0.29
pygdbmi==0.9.0.2
Pygments==2.16.1
pyparsing==2.3.1
pyserial==3.5
python-engineio==3.14.2
python-socketio==4.6.1
pywin32==306
PyYAML==6.0.1
reedsolo==1.5.4
requests==2.31.0
requests-file==1.5.1
requests-toolbelt==1.0.0
schema==0.7.5
six==1.16.0
tqdm==4.66.1
typing-extensions==4.7.1
urllib3==1.26.16
websocket-client==1.6.2
Werkzeug==0.16.1
windows-curses==2.3.1
xmlrunner==1.7.7
```

Reverting pyparsing to v2.3.1 fixed the compilation issue!!

### Debug Message

```plain
-
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

eddieliao, 2023-09-22:
Do you still have the old vsix installer? I'm having the same issue and I'd be interested in seeing if the older vsix installer would work on my end.

From what I can tell, this is the same error as described in #1040. Manually reinstalling pyparsing 2.3.1 worked for me as well, but I'd rather have it work straight from the installer. Interested in seeing what the difference is with the new nightly build compared to the previous one.

gonzabrusco, 2023-09-22:
Unfortunately I don't have it anymore. But I downloaded it by the time this was posted: https://github.com/espressif/vscode-esp-idf-extension/issues/1015#issuecomment-1702439045

PR #1048 (2023-09-26): Remove idf component manager
## Description

Removes the idf component manager from extension's requirements.txt

Fixes #1044

## Type of change

Please delete options that are not relevant.
- Bug fix (non-breaking change which fixes an issue)

## Steps to test this pull request

Provide a list of steps to test changes in this PR and required output

1. Run ESP-IDF: Configure ESP-IDF Extension
2. Install 4.4.5 idf version
3. No issues during setup

## How has this been tested?

Please describe the tests that you ran to verify your changes. Provide instructions so we can reproduce. Please also  [... cut]
closed 2023-09-26 by radurentea