espressif/vscode-esp-idf-extension #1045: CMD encounters unexpected UNC paths

ELNL, 2023-09-22:
### OS

Windows

### Operating System version

windows 10

### Visual Studio Code version

1.82.2

### ESP-IDF version

1.6.4

### Python version

3.11.2

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19045 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files (x86)\Exact Software\bin;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;C:\Program Files (x86)\Windows Kits\10\Windows Performance Toolkit\;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.82.2 
Visual Studio Code language en 
Visual Studio Code shell C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.6.4 
Workspace folder v:\EmbeddedProjects\lvgl_test 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) v:\ESP-IDF-Container\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) v:\ESP-IDF-Tools\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin;v:\ESP-IDF-Tools\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin;v:\ESP-IDF-Tools\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin;v:\ESP-IDF-Tools\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin;v:\ESP-IDF-Tools\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin;v:\ESP-IDF-Tools\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin;v:\ESP-IDF-Tools\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;v:\ESP-IDF-Tools\tools\cmake\3.24.0\bin;v:\ESP-IDF-Tools\tools\openocd-esp32\v0.12.0-esp32-20230419\openocd-esp32\bin;v:\ESP-IDF-Tools\tools\ninja\1.10.2;v:\ESP-IDF-Tools\tools\idf-exe\1.0.3;v:\ESP-IDF-Tools\tools\ccache\4.8\ccache-4.8-windows-x86_64;v:\ESP-IDF-Tools\tools\dfu-util\0.11\dfu-util-0.11-win64;v:\ESP-IDF-Tools\tools\esp-rom-elfs\20230320
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: v:\ESP-IDF-Tools\tools\openocd-esp32\v0.12.0-esp32-20230419/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
    ESP_ROM_ELF_DIR: v:\ESP-IDF-Tools\tools\esp-rom-elfs\20230320/
Virtual env Python Path (idf.pythonBinPath) v:\ESP-IDF-Tools\python_env\idf5.1_py3.11_env\Scripts\python.exe
Serial port (idf.port) COM1
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,target/esp32s3.cfg
ESP-IDF Tools Path (idf.toolsPath) v:\ESP-IDF-Tools
Git Path (idf.gitPath) v:\ESP-IDF-Tools\tools\idf-git\2.39.2\cmd\git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to v:\ESP-IDF-Tools\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin: true
Access to v:\ESP-IDF-Tools\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin: true
Access to v:\ESP-IDF-Tools\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin: true
Access to v:\ESP-IDF-Tools\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin: true
Access to v:\ESP-IDF-Tools\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin: true
Access to v:\ESP-IDF-Tools\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin: true
Access to v:\ESP-IDF-Tools\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin: true
Access to v:\ESP-IDF-Tools\tools\cmake\3.24.0\bin: true
Access to v:\ESP-IDF-Tools\tools\openocd-esp32\v0.12.0-esp32-20230419\openocd-esp32\bin: true
Access to v:\ESP-IDF-Tools\tools\ninja\1.10.2: true
Access to v:\ESP-IDF-Tools\tools\idf-exe\1.0.3: true
Access to v:\ESP-IDF-Tools\tools\ccache\4.8\ccache-4.8-windows-x86_64: true
Access to v:\ESP-IDF-Tools\tools\dfu-util\0.11\dfu-util-0.11-win64: true
Access to v:\ESP-IDF-Tools\tools\esp-rom-elfs\20230320: true
[... 139 lines cut ...]
### Debug Message

```plain
AILED: esp-idf/esp_system/ld/memory.ld //EL-WIN10/VisualStudioCode/EmbeddedProjects/lvgl_test/build/esp-idf/esp_system/ld/memory.ld 
cmd.exe /C "cd /D \\EL-WIN10\VisualStudioCode\EmbeddedProjects\lvgl_test\build\esp-idf\esp_system && V:\ESP-IDF-Tools\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin\xtensa-esp32s3-elf-gcc.exe -C -P -x c -E -o //EL-WIN10/VisualStudioCode/EmbeddedProjects/lvgl_test/build/esp-idf/esp_system/ld/memory.ld -I //EL-WIN10/VisualStudioCode/EmbeddedProjects/lvgl_test/build/config -I V:/ESP-IDF-Container/esp-idf/components/esp_system/ld V:/ESP-IDF-Container/esp-idf/components/esp_system/ld/esp32s3/memory.ld.in"
'\\EL-WIN10\VisualStudioCode\EmbeddedProjects\lvgl_test\build\esp-idf\esp_system'
CMD does not support UNC paths as current directories.
ninja: build stopped: subcommand failed.
```

### Other Steps to Reproduce

Install VSC.
Install ESP-IDF
Create project hello_world

Full Clean 

Then the build system interprets some paths as UNC  paths.

The system is completely installed on a mapped drive V:  Alle paths are accessible with regular paths starting with V:. But somehow UNC paths are introduced. (The current computer name is EL-WIN10)

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

georgik, 2023-10-02:
UNC paths are not supported. The problem is not only in CMD, but also in other tools like CMake, Git or CCache. They does not work well with UNC or long paths - see limitation: https://github.com/espressif/idf-installer/issues/103

One possible workaround is to use `subst` command to map long path to new drive letter:
```
subst "R:" "my_super_long_path"
```

ELNL, 2023-10-15:
Thank you this response @georgik .
To shorten the paths I already used a 'drive mapping' in Windows 10, using the Windows UI. I am not sure is 'subst' works the same way. I will try this. (It will take a few weeks perhaps to do it...)

arantius, 2024-10-09:
I still have exactly this problem.  I can build the project (just "hello world" example for now), but I can't run menuconfig.  My project is already on a mapped drive, and that's the path I gave when creating the project.  But when I try to menuconfig, it uses the UNC path equivalent, instead of the drive letter.

arantius, 2024-10-09:
Actually, as soon as I've tried to do menuconfig, I can't build anymore for the same UNC path error.  If I create a new project and never try to menuconfig it then I can build all I like.

brianignacio5, 2024-10-28:
It might be related to some default settings that are resolving `${workspaceFolder}` to the real project path instead of the mapped location. For example, try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/workflows/ci/master/esp-idf-extension.vsix.zip) and set `idf.buildPathWin` and `idf.sdkconfigFilePath` in `<your project>\.vscode\settings.json` to build and sdkconfig paths using the mapped drive.

brianignacio5, 2025-08-05:
I'm closing this issue as it is not specific to this vscode esp-idf extension.