espressif/vscode-esp-idf-extension #1049: ESP-MATTER install in windows: error shile executing activate.sh

Suxsem, 2023-09-26:
### OS

Windows

### Operating System version

Windows 10

### ESP-IDF version

5.1.1

### Python version

3.11.4

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19045 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Program Files\Common Files\Oracle\Java\javapath;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\Microsoft SQL Server\140\Tools\Binn\;C:\Program Files\Microsoft SQL Server\150\Tools\Binn\;C:\Program Files\nodejs\;C:\Program Files\Git\cmd;C:\Program Files\Microsoft SQL Server\Client SDK\ODBC\170\Tools\Binn\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\dotnet\;C:\Program Files (x86)\AOMEI\AOMEI Backupper\6.9.2;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files (x86)\ZeroTier\One\;C:\Program Files\Docker\Docker\resources\bin;C:\Program Files\Cloudflare\Cloudflare WARP\;C:\Program Files\Microsoft SQL Server\140\DTS\Binn\;C:\Program Files\Microsoft SQL Server\150\DTS\Binn\;C:\Program Files\Microsoft SQL Server\160\DTS\Binn\;C:\Program Files\usbipd-win\;C:\Users\Public\Documents\RAD Studio\12.0\Bpl;C:<HOMEPATH>\AppData\Local\Programs\Python\Python311\Scripts\;C:<HOMEPATH>\AppData\Local\Programs\Python\Python311\;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin;C:<HOMEPATH>\AppData\Roaming\npm;C:\Program Files\OpenSSL-Win64\bin;C:<HOMEPATH>\.dotnet\tools;C:<HOMEPATH>\AppData\Local\Programs\Fiddler;C:\Program Files\mitmproxy\bin;C:<HOMEPATH>\mingw64\bin;C:<HOMEPATH>\flutter\bin;C:<HOMEPATH>\.dotnet\tools 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.82.2 
Visual Studio Code language en 
Visual Studio Code shell C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.6.4 
Workspace folder c:<HOMEPATH>\Desktop\temp 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:<HOMEPATH>\esp\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) C:<HOMEPATH>\.espressif\esp-matter
Custom extra paths (idf.customExtraPaths) C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin;C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:<HOMEPATH>\.espressif\tools\cmake\3.24.0\bin;C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230419\openocd-esp32\bin;C:<HOMEPATH>\.espressif\tools\ninja\1.10.2;C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3;C:<HOMEPATH>\.espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64;C:<HOMEPATH>\.espressif\tools\dfu-util\0.11\dfu-util-0.11-win64;C:<HOMEPATH>\.espressif\tools\esp-rom-elfs\20230320
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230419/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
    ESP_ROM_ELF_DIR: C:<HOMEPATH>\.espressif\tools\esp-rom-elfs\20230320/
Virtual env Python Path (idf.pythonBinPath) C:<HOMEPATH>\.espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe
Serial port (idf.port) COM1
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) C:<HOMEPATH>\.espressif
Git Path (idf.gitPath) C:\Program Files\Git\cmd\git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) true
Access to ESP-IDF Custom extra paths
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\cmake\3.24.0\bin: true
Access to C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230419\openocd-esp32\bin: true
Access to C:<HOMEPATH>\.espressif\tools\ninja\1.10.2: true
Access to C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3: true
Access to C:<HOMEPATH>\.espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64: true
Access to C:<HOMEPATH>\.espressif\tools\dfu-util\0.11\dfu-util-0.11-win64: true
Access to C:<HOMEPATH>\.espressif\tools\esp-rom-elfs\20230320: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) true
[... 93 lines cut ...]

Thank you

### Debug Message

```plain
*  Executing task: source C:\Users\semer\.espressif\esp-matter\connectedhomeip\connectedhomeip\scripts\activate.sh 

source : Termine 'source' non riconosciuto come nome di cmdlet, funzione, 
programma eseguibile o file script. Controllare l'ortografia del nome o 
verificare che il percorso sia incluso e corretto, quindi riprovare.
In riga:1 car:1
+ source C:\Users\semer\.espressif\esp-matter\connectedhomeip\connected ...       
+ ~~~~~~
    + CategoryInfo          : ObjectNotFound: (source:String) [], CommandNotFound 
   Exception
    + FullyQualifiedErrorId : CommandNotFoundException
 

 *  The terminal process "C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -Command source C:\Users\semer\.espressif\esp-matter\connectedhomeip\connectedhomeip\scripts\activate.sh" terminated with exit code: 1.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Suxsem, 2023-09-26:
@brianignacio5 why there is idf.espMatterPathWin setting? 

https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/additional_frameworks.md

brianignacio5, 2023-10-03:
Removed the settings for Windows and added validation to avoid Windows users to show ESP-Matter commands.

closed 2023-10-11 by brianignacio5