espressif/vscode-esp-idf-extension #1053: I want to know how to use ESP-IDF debugger

Ashutosh-017, 2023-09-28:
### OS

Windows

### Operating System version

Windows 10

### Visual Studio Code version

1.82.1

### ESP-IDF version

5.1.1

### Python version

3.8.7

### Doctor command output

DEBUG_ADAPTER_STARTED
2023-09-28 14:33:28,451 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STARTED

[Debug Adapter]
DEBUG_ADAPTER_READY2CONNECT
2023-09-28 14:33:28,451 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_READY2CONNECT

[Debug Adapter]
DEBUG_ADAPTER_STOPPED
2023-09-28 14:33:43,453 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STOPPED

[Stopped] : ESP-IDF Debug Adapter

### Extension

Output of debug console 
Debug adapter -> Extension: DEBUG_ADAPTER_STARTED
Debug adapter -> Extension: DEBUG_ADAPTER_READY2CONNECT

### Description

When ever I try to run debugger then this message is pop up out 
"Fail to init debug Adapter"

### Debug Message

```plain
Debug adapter -> Extension: DEBUG_ADAPTER_STARTED
Debug adapter -> Extension: DEBUG_ADAPTER_READY2CONNECT
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2023-09-28:
Have you read the tutorials and setup and the extension ?

1. https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/install.md
2. https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/basic_use.md
3. https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/debugging.md

Ashutosh-017, 2023-09-28:
@brianignacio5 Yes I do read all these documents I am able build code run code. I have created a MQTTS related project and a BLE related project everything work fine. I only want to know how to use debugger please help me

brianignacio5, 2023-09-28:
What about the (3) ? Is about debugging ? 

In your launch.json, change logLevel to 5 to increase the debug output see what is wrong. Check also the OpenOCD output in the ESP-IDF Output Channel. Any errors ? Did you select the right openOCD config files ?

Ashutosh-017, 2023-09-28:
Thank you so much for this quick response 
Please have a look on data I provide you. 
**launch.json file content** 
{
  "version": "0.2.0",
  "configurations": [
    {
      "type": "espidf",
      "name": "Launch",
      "request": "launch",
      "logLevel": 5
    }
  ]
}
**settings.json** file content 
{
    "C_Cpp.intelliSenseEngine": "default",
    "idf.adapterTargetName": "esp32c3",
    "idf.customExtraPaths": "C:\\Users\\ashutosh\\.espressif\\tools\\xtensa-esp-elf-gdb\\11.2_20220823\\xtensa-esp-elf-gdb\\bin;C:\\Users\\ashutosh\\.espressif\\tools\\riscv32-esp-elf-gdb\\11.2_20220823\\riscv32-esp-elf-gdb\\bin;C:\\Users\\ashutosh\\.espressif\\tools\\xtensa-esp32-elf\\esp-2022r1-11.2.0\\xtensa-esp32-elf\\bin;C:\\Users\\ashutosh\\.espressif\\tools\\xtensa-esp32s2-elf\\esp-2022r1-11.2.0\\xtensa-esp32s2-elf\\bin;C:\\Users\\ashutosh\\.espressif\\tools\\xtensa-esp32s3-elf\\esp-2022r1-11.2.0\\xtensa-esp32s3-elf\\bin;C:\\Users\\ashutosh\\.espressif\\tools\\riscv32-esp-elf\\esp-2022r1-11.2.0\\riscv32-esp-elf\\bin;C:\\Users\\ashutosh\\.espressif\\tools\\esp32ulp-elf\\2.35_20220830\\esp32ulp-elf\\bin;C:\\Users\\ashutosh\\.espressif\\tools\\cmake\\3.24.0\\bin;C:\\Users\\ashutosh\\.espressif\\tools\\openocd-esp32\\v0.11.0-esp32-20221026\\openocd-esp32\\bin;C:\\Users\\ashutosh\\.espressif\\tools\\ninja\\1.10.2;C:\\Users\\ashutosh\\.espressif\\tools\\idf-exe\\1.0.3;C:\\Users\\ashutosh\\.espressif\\tools\\ccache\\4.6.2\\ccache-4.6.2-windows-x86_64;C:\\Users\\ashutosh\\.espressif\\tools\\dfu-util\\0.9\\dfu-util-0.9-win64;C:\\Users\\ashutosh\\.espressif\\tools\\esp-rom-elfs\\20220823",
    "idf.customExtraVars": {
        "OPENOCD_SCRIPTS": "C:\\Users\\ashutosh\\.espressif\\tools\\openocd-esp32\\v0.11.0-esp32-20221026/openocd-esp32/share/openocd/scripts",
        "IDF_CCACHE_ENABLE": "1",
        "ESP_ROM_ELF_DIR": "C:\\Users\\ashutosh\\.espressif\\tools\\esp-rom-elfs\\20220823/"
    },
    "idf.espIdfPathWin": "C:\\Users\\ashutosh\\esp\\esp-idf",
    "idf.openOcdConfigs": [
        "board/esp32c3-bridge.cfg"
    ],
    "idf.portWin": "COM3",
    "idf.pythonBinPathWin": "C:\\Users\\ashutosh\\.espressif\\python_env\\idf5.0_py3.8_env\\Scripts\\python.exe",
    "idf.toolsPathWin": "C:\\Users\\ashutosh\\.espressif",
    "idf.flashType": "UART"
}
**Output on output window** 
C:\Users\ashutosh\.espressif\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\bin\openocd.exe

C:\Users\ashutosh\.espressif\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\bin\openocd.exe

[OpenOCD]
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1002
Info : esp_usb_jtag: capabilities descriptor set to 0x30a
Info : esp_usb_jtag: target chip id set to 5
Warn : Transport "jtag" was already selected
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections

[OpenOCD]
❌ Error: esp_usb_jtag: could not find or open device!

[OpenOCD]
❌ 
Error: Unsupported xlen: -1
Error: Unknown target arch!

[OpenOCD]
[Stopped] : OpenOCD Server
[Debug Adapter]
2023-09-28 17:17:26,788 - Debug Adapter (main) - DEBUG - START_TIME_2023_09_28_17-17-26
2023-09-28 17:17:26,788 - Debug Adapter (main) - DEBUG - Init errors:None
2023-09-28 17:17:26,788 - Debug Adapter (main) - INFO - Working directory: C:\Users\ashutosh\AppData\Local\Programs\Microsoft VS Code

[Debug Adapter]
2023-09-28 17:17:26,788 - Debug Adapter (main) - INFO - Arguments: 
{   'app_flash_off': 65536,
    'board_type': None,
    'cmdfile': None,
    'core_file': (),
    'debug': 5,
    'developer_mode': 'none',
[... 237 lines cut ...]
        "adapterID": "espidf",
        "clientID": "vscode",
        "clientName": "Visual Studio Code",
        "columnsStartAt1": true,
        "linesStartAt1": true,
        "locale": "en",
        "pathFormat": "path",
        "supportsArgsCanBeInterpretedByShell": true,
        "supportsInvalidatedEvent": true,
        "supportsMemoryEvent": true,
        "supportsMemoryReferences": true,
        "supportsProgressReporting": true,
        "supportsRunInTerminalRequest": true,
        "supportsStartDebuggingRequest": true,
        "supportsVariablePaging": true,
        "supportsVariableType": true
    },
    "command": "initialize",
    "seq": 1,
    "type": "request"
}
Created gdb object of type: GdbEsp32c3

Help me how to select right openOCD config file.
I share everything with you please help me

brianignacio5, 2023-09-28:
See the log 

```
[OpenOCD]
❌ Error: esp_usb_jtag: could not find or open device!

[OpenOCD]
❌
Error: Unsupported xlen: -1
Error: Unknown target arch!
```

there is an openOCD configuration error. Probably because

```json
"idf.openOcdConfigs": [
"board/esp32c3-bridge.cfg"
],
```

is not the correct board for you. Try a different openOCD configuration and make sure openOCD works correctly.

Ashutosh-017, 2023-09-29:
I am working with ESP32-C3 Mini please tell me how to change openOCD configuration and make openOCD works correctly. 
Thank you

Ashutosh-017, 2023-09-29:
@brianignacio5 On this page https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/debugging.md
**I read this line** 
Choosing esp32c3 as IDF_TARGET will set idf.openOCDConfigs to ["board/esp32c3-builtin.cfg"] if using built-in usb jtag or ["board/esp32c3-ftdi.cfg"] if using ESP-PROG-JTAG. 
So apply this but still I face the same error please help bro.

Ashutosh-017, 2023-10-05:
I connected following pins
GPIO19 with USB D+
GPIO18 with USB D-
5V to VCC
Ground to Ground 
Then,
I am facing this error 
[OpenOCD]
❌ Error: libusb_open() failed with LIBUSB_
[OpenOCD]
❌ ERROR_NOT_SUPPORTED
Error: libusb_open() failed with LIBUSB_ERROR_NOT_FOUND
Error: esp_usb_jtag: could not find or open device!
Please provide support

closed 2023-10-31 by github-actions[bot]