espressif/vscode-esp-idf-extension #1054: ESP-IDF installation downloading forbidden pyparsing package 3.0.9

KonssnoK, 2023-09-28:
### OS

Windows

### Operating System version

Win11

### ESP-IDF version

4.4.6

### Python version

3.11

### Doctor command output

pip version: 23.1.2
pycparser version: 2.21
pyelftools version: 0.29
pygdbmi version: 0.9.0.2
Pygments version: 2.15.1
pyparsing version: 3.0.9
pyserial version: 3.5
pywin32 version: 306
PyYAML version: 6.0
reedsolo version: 1.5.4
requests version: 2.31.0

### Description

```
ERROR: failed to parse C:/Users/r/esp/ESP-IDF/components/esp_event/linker.lf
Expected end of text, found 'i'  (at char 0), (line:1, col:1)
ninja: build stopped: subcommand failed.
```

Couldn't link anymore, found 
https://github.com/espressif/vscode-esp-idf-extension/issues/32

Check packages with
` C:\src\espressif\.espressif\python_env\idf4.4_py3.11_env\Scripts\pip.exe  list`

find out that pyparsing is at 3.0.9.

Try to run 
`ESP-IDF : Configure ESP-IDF extension `

Doesn't start because 

```
The following Python requirements are not satisfied:
Requirement 'pyparsing<2.4.0,>=2.0.3' was not met. Installed version: 3.0.9
To install the missing packages, please run "C:/src/esp-idf/install.bat"
Diagnostic information:
```

Manually rollback the package

```
C:\src\espressif\.espressif\python_env\idf4.4_py3.11_env\Scripts\pip.exe uninstall pyparsing
C:\src\espressif\.espressif\python_env\idf4.4_py3.11_env\Scripts\pip.exe install pyparsing==2.3.0
```

Restart VSCODE,
Run 
`ESP-IDF : Configure ESP-IDF extension `

now the extension is rendered.

Click on "use current installation"

[image]

Script runs and in the terminal i see

```
Requirement already satisfied: MarkupSafe>=2.0 in c:\src\espressif\.espressif\python_env\idf4.4_py3.11_env\lib\site-packages (from jinja2->gcovr->-r c:\Users\NR6085\.vscode\extensions\espressif.esp-idf-extension-1.6.4\requirements.txt (line 1)) (2.1.3)
Installing collected packages: pyparsing
Attempting uninstall: pyparsing
Found existing installation: pyparsing 2.3.0
Uninstalling pyparsing-2.3.0:
Successfully uninstalled pyparsing-2.3.0
Successfully installed pyparsing-3.0.9
```

[image]

Somebody updated packages without checking for regressions

### Debug Message

```plain
?
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

radurentea, 2023-10-02:
Hi @KonssnoK,

This issue has been fixed in this [PR](https://github.com/espressif/vscode-esp-idf-extension/pull/1048) and it will be available on next release of the extension.

Meanwhile, you can try installing the extension using one of these:
[esp-idf-extension.vsix.zip](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/944521762.zip)
[esp-idf-extension-open.vsix.zip](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/944521764.zip)

For more background and guidance how to do it, check this #1040

brianignacio5, 2023-10-13:
Any updates @KonssnoK ?

KonssnoK, 2023-10-13:
@brianignacio5 was the new release released?
what i did was to manually replace the package, then i didnt have time to test the zips.

radurentea, 2023-10-26:
Hi @KonssnoK,

It has been released

closed 2023-10-26 by KonssnoK