espressif/vscode-esp-idf-extension #1059: Failed to find/open ESP32-H2 device

parfnick, 2023-10-14:
### OS

Windows

### Operating System version

Windos 10

### Visual Studio Code version

1.83.1

### ESP-IDF version

5.1.1

### Python version

3.1

### Doctor command output

[report.txt](https://github.com/espressif/vscode-esp-idf-extension/files/12907028/report.txt)

### Description

I use ESP32-H2-DevKitM-1.
When run debug session **from VSCODE** openocd say, that it can not found the device. OpenOCD detect(see log) wrong(1002) PID, device manager show PID(1001) for board.

**When run openocd from command line**, like this:
_openocd.exe  -f   ...\esp32h2-builtin.cfg_

device is found correctly, **no errors**.

### Debug Message

```plain
Debug: 6 8 log.c:236 handle_log_output_command(): set log_output to "openocd_log.txt"
Debug: 7 8 options.c:233 add_default_dirs(): bindir=/builds/idf/openocd-esp32/_build/../openocd-esp32/bin
Debug: 8 8 options.c:234 add_default_dirs(): pkgdatadir=/builds/idf/openocd-esp32/_build/../openocd-esp32/share/openocd
Debug: 9 8 options.c:235 add_default_dirs(): exepath=C:/Espressif/tools/openocd-esp32/v0.12.0-esp32-20230419/openocd-esp32/bin
Debug: 10 8 options.c:236 add_default_dirs(): bin2data=../share/openocd
Debug: 11 8 configuration.c:33 add_script_search_dir(): adding C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230419/openocd-esp32/share/openocd/scripts
Debug: 12 8 configuration.c:33 add_script_search_dir(): adding C:/Users/parfn/AppData/Roaming/OpenOCD
Debug: 13 9 configuration.c:33 add_script_search_dir(): adding C:\Users\parfn/.config/openocd
Debug: 14 9 configuration.c:33 add_script_search_dir(): adding C:\Users\parfn/.openocd
Debug: 15 9 configuration.c:33 add_script_search_dir(): adding C:/Espressif/tools/openocd-esp32/v0.12.0-esp32-20230419/openocd-esp32/bin/../share/openocd/site
Debug: 16 9 configuration.c:33 add_script_search_dir(): adding C:/Espressif/tools/openocd-esp32/v0.12.0-esp32-20230419/openocd-esp32/bin/../share/openocd/scripts
Debug: 17 9 command.c:152 script_debug(): command - ocd_find board/esp32h2-bridge.cfg
Debug: 18 9 configuration.c:88 find_file(): found C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230419/openocd-esp32/share/openocd/scripts/board/esp32h2-bridge.cfg
Debug: 19 11 command.c:152 script_debug(): command - ocd_find interface/esp_usb_bridge.cfg
Debug: 20 11 configuration.c:88 find_file(): found C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230419/openocd-esp32/share/openocd/scripts/interface/esp_usb_bridge.cfg
Debug: 21 11 command.c:152 script_debug(): command - adapter driver esp_usb_jtag
Info : 22 11 transport.c:107 allow_transports(): only one transport option; autoselecting 'jtag'
Debug: 23 12 command.c:152 script_debug(): command - espusbjtag vid_pid 0x303a 0x1002
Info : 24 12 esp_usb_jtag.c:894 esp_usb_jtag_vid_pid(): esp_usb_jtag: VID set to 0x303a and PID to 0x1002
Debug: 25 12 command.c:152 script_debug(): command - espusbjtag caps_descriptor 0x030A
Info : 26 12 esp_usb_jtag.c:905 esp_usb_jtag_caps_descriptor(): esp_usb_jtag: capabilities descriptor set to 0x30a
Debug: 27 12 command.c:152 script_debug(): command - adapter speed 40000
Debug: 28 12 adapter.c:253 adapter_config_khz(): handle adapter khz
Debug: 29 12 adapter.c:217 adapter_khz_to_speed(): convert khz to adapter specific speed value
Debug: 30 12 adapter.c:217 adapter_khz_to_speed(): convert khz to adapter specific speed value
Debug: 31 12 command.c:152 script_debug(): command - espusbjtag chip_id 16
Info : 32 12 esp_usb_jtag.c:916 esp_usb_jtag_chip_id(): esp_usb_jtag: target chip id set to 16
Debug: 33 12 command.c:152 script_debug(): command - ocd_find target/esp32h2.cfg
Debug: 34 12 configuration.c:88 find_file(): found C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230419/openocd-esp32/share/openocd/scripts/target/esp32h2.cfg
Debug: 35 12 command.c:152 script_debug(): command - transport select jtag
Warn : 36 12 transport.c:280 handle_transport_select(): Transport "jtag" was already selected
Debug: 37 12 command.c:152 script_debug(): command - ocd_find bitsbytes.tcl
Debug: 38 13 configuration.c:88 find_file(): found C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230419/openocd-esp32/share/openocd/scripts/bitsbytes.tcl
[... 87 lines cut ...]
Debug: 105 19 command.c:152 script_debug(): command - riscv set_ebreaku on
Info : 106 20 server.c:298 add_service(): Listening on port 6666 for tcl connections
Info : 107 20 server.c:298 add_service(): Listening on port 4444 for telnet connections
Debug: 108 20 command.c:152 script_debug(): command - init
Debug: 109 20 command.c:152 script_debug(): command - target init
Debug: 110 21 command.c:152 script_debug(): command - target names
Debug: 111 21 command.c:152 script_debug(): command - esp32h2 cget -event gdb-flash-erase-start
Debug: 112 21 command.c:152 script_debug(): command - esp32h2 configure -event gdb-flash-erase-start reset init
Debug: 113 21 command.c:152 script_debug(): command - esp32h2 cget -event gdb-flash-write-end
Debug: 114 21 command.c:152 script_debug(): command - esp32h2 configure -event gdb-flash-write-end reset halt
Debug: 115 21 command.c:152 script_debug(): command - esp32h2 cget -event gdb-attach
Debug: 116 21 target.c:1657 handle_target_init_command(): Initializing targets...
Debug: 117 21 riscv.c:444 riscv_init_target(): riscv_init_target()
Debug: 118 21 semihosting_common.c:107 semihosting_common_init():  
Error: 119 1492 esp_usb_jtag.c:646 esp_usb_jtag_init(): esp_usb_jtag: could not find or open device!
Debug: 120 1492 command.c:542 run_command(): Command 'init' failed with error code -4
User : 121 1492 command.c:613 command_run_line(): C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230419/openocd-esp32/share/openocd/scripts/target/esp_common.cfg:4: Error: 
at file "C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230419/openocd-esp32/share/openocd/scripts/target/esp_common.cfg", line 4
Debug: 122 1492 riscv.c:490 riscv_deinit_target(): riscv_deinit_target()
Debug: 123 1492 target.c:2223 target_free_all_working_areas_restore(): freeing all working areas
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2023-10-15:
Can you share the debug log output from extension debug ? In launch.json change logLevel to 4 and share the output from generated project-directory debug.log file

parfnick, 2023-10-18:
[debug.log](https://github.com/espressif/vscode-esp-idf-extension/files/12996264/debug.log)

parfnick, 2023-10-19:
Problem can be fixed if **explicitly type interface config file**.
_{
  "C_Cpp.intelliSenseEngine": "Tag Parser",
  "idf.adapterTargetName": "esp32h2",
  "idf.customAdapterTargetName": "esp32h2",
  "idf.openOcdDebugLevel": 4,
  "idf.openOcdLaunchArgs": ["-l openocd_log.txt"],
  "cmake.configureOnOpen": false,
  "idf.openOcdConfigs": [
    "interface/esp_usb_jtag.cfg",
    "target/esp32h2.cfg"
  ],
  "idf.flashType": "JTAG"
}_

Maybe, main problem in openocd feature, but plugin do not take it into account, and set only board config file.

parfnick, 2023-10-19:
I use settings
_{
  "C_Cpp.intelliSenseEngine": "Tag Parser",
  "idf.adapterTargetName": "esp32h2",
  "idf.customAdapterTargetName": "esp32h2",
  "idf.openOcdDebugLevel": 4,
  "idf.openOcdLaunchArgs": ["-l openocd_log.txt"],
  "cmake.configureOnOpen": false,
  "idf.openOcdConfigs": [
    **"board/esp32h2-builtin.cfg"**
  ],
  "idf.flashType": "JTAG"
}_

Why openocd use esp32h2_bridge.cfg ?

_Debug: 17 9 command.c:152 script_debug(): command - ocd_find board/esp32h2-bridge.cfg
Debug: 18 9 configuration.c:88 find_file(): found C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230419/openocd-esp32/share/openocd/scripts/board/esp32h2-bridge.cfg
Debug: 19 11 command.c:152 script_debug(): command - **ocd_find interface/esp_usb_bridge.cfg**
Debug: 20 11 configuration.c:88 find_file(): found C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230419/openocd-esp32/share/openocd/scripts/interface/esp_usb_bridge.cfg
Debug: 21 11 command.c:152 script_debug(): command - adapter driver esp_usb_jtag
Info : 22 11 transport.c:107 allow_transports(): only one transport option; autoselecting 'jtag'
Debug: 23 12 command.c:152 script_debug(): command - espusbjtag vid_pid 0x303a 0x1002
Info : 24 12 esp_usb_jtag.c:894 esp_usb_jtag_vid_pid(): esp_usb_jtag: VID set to 0x303a and PID to 0x1002_

parfnick, 2023-10-19:
Where(in source code) plugin run openocd for the first time?

radurentea, 2023-10-26:
Hi @parfnick,

Try running **ESP-IDF: SDK Configuration Editor (menuconfig)** command first from command palette and let me know if the issue persists.

brianignacio5, 2023-11-01:
When you run openOCD from the extension you should see the command line used for output. Based on settings you should see something like:

`openocd.exe -f board/esp32h2-bridge.cfg`

Is the openocd.exe from the extension the same you are using in the terminal? is the OPENOCD_SCRIPTS the same directory ?

brianignacio5, 2023-11-13:
Closing this issue for lack of activity. Comment here if you need more help.

closed 2023-11-13 by brianignacio5