espressif/vscode-esp-idf-extension #1060: I cannot get debugging to work with ESP32S3 on one machine

fitch22, 2023-10-16:
### OS

Windows

### Operating System version

Windows 11 Pro

### Visual Studio Code version

1.83.1

### ESP-IDF version

5.1.1

### Python version

3.11.2

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.22621 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:<HOMEPATH>\.espressif\python_env\idf5.1_py3.11_env\Scripts;C:<HOMEPATH>\.espressif\python_env\idf5.1_py3.11_env;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files (x86)\VMware\VMware Workstation\bin\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\dotnet\;C:\Program Files (x86)\Common Files\Intuit\QBPOSSDKRuntime;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;C:\Program Files\Git\cmd;C:\Program Files\WireGuard\;C:\Program Files\nodejs\;C:\Program Files\TortoiseGit\bin;C:<HOMEPATH>\.cargo\bin;C:\Program Files (x86)\Arm GNU Toolchain arm-none-eabi\12.2 rel1\bin;C:<HOMEPATH>\.dotnet\tools;C:<HOMEPATH>\AppData\Local\gitkraken\bin;C:<HOMEPATH>\.dotnet\tools;C:<HOMEPATH>\AppData\Roaming\npm;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.83.1 
Visual Studio Code language en 
Visual Studio Code shell C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps\Microsoft.PowerShell_8wekyb3d8bbwe\pwsh.exe 
ESP-IDF Extension version 1.6.5 
Workspace folder c:<HOMEPATH>\Documents\esp_experiments\blink 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) c:/users/bill/esp/esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin;C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:<HOMEPATH>\.espressif\tools\cmake\3.24.0\bin;C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230419\openocd-esp32\bin;C:<HOMEPATH>\.espressif\tools\ninja\1.10.2;C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3;C:<HOMEPATH>\.espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64;C:<HOMEPATH>\.espressif\tools\dfu-util\0.11\dfu-util-0.11-win64;C:<HOMEPATH>\.espressif\tools\esp-rom-elfs\20230320
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230419/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
    ESP_ROM_ELF_DIR: C:<HOMEPATH>\.espressif\tools\esp-rom-elfs\20230320/
Virtual env Python Path (idf.pythonBinPath) C:<HOMEPATH>\.espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe
Serial port (idf.port) COM20
OpenOCD Configs (idf.openOcdConfigs) board/esp32s3-builtin.cfg
ESP-IDF Tools Path (idf.toolsPath) C:<HOMEPATH>\.espressif
Git Path (idf.gitPath) C:<HOMEPATH>\.espressif\tools\idf-git\2.39.2\cmd\git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\cmake\3.24.0\bin: true
Access to C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230419\openocd-esp32\bin: true
Access to C:<HOMEPATH>\.espressif\tools\ninja\1.10.2: true
Access to C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3: true
Access to C:<HOMEPATH>\.espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64: true
Access to C:<HOMEPATH>\.espressif\tools\dfu-util\0.11\dfu-util-0.11-win64: true
Access to C:<HOMEPATH>\.espressif\tools\esp-rom-elfs\20230320: true
[... 160 lines cut ...]
  File "c:\Users\Bill\.vscode\extensions\espressif.esp-idf-extension-1.6.5\esp_debug_adapter\debug_adapter\command_processor.py", line 70, in on_initialize_request
    self.da.adapter_init()
  File "c:\Users\Bill\.vscode\extensions\espressif.esp-idf-extension-1.6.5\esp_debug_adapter\debug_adapter\debug_adapter.py", line 329, in adapter_init
    raise e
  File "c:\Users\Bill\.vscode\extensions\espressif.esp-idf-extension-1.6.5\esp_debug_adapter\debug_adapter\debug_adapter.py", line 325, in adapter_init
    self.start_gdb()
  File "c:\Users\Bill\.vscode\extensions\espressif.esp-idf-extension-1.6.5\esp_debug_adapter\debug_adapter\debug_adapter.py", line 678, in start_gdb
    raise e
  File "c:\Users\Bill\.vscode\extensions\espressif.esp-idf-extension-1.6.5\esp_debug_adapter\debug_adapter\debug_adapter.py", line 666, in start_gdb
    self._gdb.exec_file_set(elf)
  File "C:\Users\Bill\.espressif\python_env\idf5.1_py3.11_env\Lib\site-packages\esp_debug_backend\gdb.py", line 283, in exec_file_set
    res, _ = self._mi_cmd_run('-file-exec-and-symbols %s' % local_file_path)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Bill\.espressif\python_env\idf5.1_py3.11_env\Lib\site-packages\esp_debug_backend\gdb.py", line 206, in _mi_cmd_run
    raise DebuggerTargetStateTimeoutError(
esp_debug_backend.defs.DebuggerTargetStateTimeoutError: Failed to wait for completion of command "-file-exec-and-symbols c:/Users/Bill/Documents/esp_experiments/blink/build/blink.elf" / 25!
```

### Other Steps to Reproduce

Works on a different machine, see above.  Have tried re-installing several times, never any change.  Tried the latest build of openocd, no change.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

fitch22, 2023-10-18:
I have installed vscode and esp-idf extension on a 3rd computer, also windows 11 pro.  Like the 2nd computer, this works fine.

I have tried comparing anything different between machines and I cannot find anything that makes any sense.  I have tried clearing the PATH to eliminate any differences in installed software but that made no difference.

Back to the non-working computer.  It works with the microsoft debugger.  This again confirms to me that the driver is installed correctly, and that both openocd and gdb are working.  It looks to me like something in the espressif debug adapter.

How can I debug the espressif debug adapter?  Is there any more information that I can provide?  Sure hope someone has some ideas.

By the way, here is the complete debug.log file from my first post, I could not attach the whole file because it was too large.

[debug.log](https://github.com/espressif/vscode-esp-idf-extension/files/13034538/debug.log)

fitch22, 2023-10-19:
I increased tmoScaleFactor and got things to work.  7 and higher work, but 6 or less do not.

Question: What is a safe value for this?  Why would I need to set it so high?  Is it because it is running on a very fast computer?

Thanks.

radurentea, 2023-10-26:
A safe value would be the default one, which is 1, but it seems in this case 7 works best for your setup. If it doesn't cause any side effects than there is no reason to worry.

There are a bunch of reasons that I can't really pinpoint such as:

- Hardware Delays: Your hardware might be taking longer to respond, perhaps due to its own internal processes or other connected peripherals.
- Software Overheads: some softwares might introduce some latencies.
- Communication Delays: If the extension communicates with external devices or servers, network or serial communication 
- Race Conditions: Faster systems might process things out of order if not properly synchronized
- Overclocking Issues: If your system is overclocked, it might introduce instability in certain processes.

closed 2023-11-16 by github-actions[bot]