espressif/vscode-esp-idf-extension #1066: Fresh installation does not configure PATH

rkit1, 2023-10-22:
### OS

Windows

### Operating System version

Windows 11

### ESP-IDF version

5.1.1

### Python version

???

### Doctor command output

'riscv32-esp-elf-gcc' is not recognized as an internal or external command,
operable program or batch file.
Command failed: riscv32-esp-elf-gcc --version
'riscv32-esp-elf-gcc' is not recognized as an internal or external command,
operable program or batch file.
Error checking riscv32-esp-elf version
Error: Command failed: riscv32-esp-elf-gcc --version
'riscv32-esp-elf-gcc' is not recognized as an internal or external command,
operable program or batch file.

'xtensa-esp32s3-elf-gcc' is not recognized as an internal or external command,
operable program or batch file.
Command failed: xtensa-esp32s3-elf-gcc --version
'xtensa-esp32s3-elf-gcc' is not recognized as an internal or external command,
operable program or batch file.
Error checking xtensa-esp32s3-elf version
Error: Command failed: xtensa-esp32s3-elf-gcc --version
'xtensa-esp32s3-elf-gcc' is not recognized as an internal or external command,
operable program or batch file.

'xtensa-esp32s2-elf-gcc' is not recognized as an internal or external command,
operable program or batch file.
Command failed: xtensa-esp32s2-elf-gcc --version
'xtensa-esp32s2-elf-gcc' is not recognized as an internal or external command,
operable program or batch file.
Error checking xtensa-esp32s2-elf version
Error: Command failed: xtensa-esp32s2-elf-gcc --version
'xtensa-esp32s2-elf-gcc' is not recognized as an internal or external command,
operable program or batch file.

'xtensa-esp32-elf-gcc' is not recognized as an internal or external command,
operable program or batch file.
Command failed: xtensa-esp32-elf-gcc --version
'xtensa-esp32-elf-gcc' is not recognized as an internal or external command,
operable program or batch file.
Error checking xtensa-esp32-elf version
Error: Command failed: xtensa-esp32-elf-gcc --version
'xtensa-esp32-elf-gcc' is not recognized as an internal or external command,
operable program or batch file.

'riscv32-esp-elf-gdb-no-python' is not recognized as an internal or external command,
operable program or batch file.
Command failed: riscv32-esp-elf-gdb-no-python --version
'riscv32-esp-elf-gdb-no-python' is not recognized as an internal or external command,
operable program or batch file.
Error checking riscv32-esp-elf-gdb version
Error: Command failed: riscv32-esp-elf-gdb-no-python --version
'riscv32-esp-elf-gdb-no-python' is not recognized as an internal or external command,
operable program or batch file.

'xtensa-esp-elf-gdb-no-python' is not recognized as an internal or external command,
operable program or batch file.
Command failed: xtensa-esp-elf-gdb-no-python --version
'xtensa-esp-elf-gdb-no-python' is not recognized as an internal or external command,
operable program or batch file.
Error checking xtensa-esp-elf-gdb version
Error: Command failed: xtensa-esp-elf-gdb-no-python --version
'xtensa-esp-elf-gdb-no-python' is not recognized as an internal or external command,
operable program or batch file.

'esp32ulp-elf-as' is not recognized as an internal or external command,
operable program or batch file.
Command failed: esp32ulp-elf-as --version
'esp32ulp-elf-as' is not recognized as an internal or external command,
operable program or batch file.
Error checking esp32ulp-elf version
Error: Command failed: esp32ulp-elf-as --version
'esp32ulp-elf-as' is not recognized as an internal or external command,
operable program or batch file.

'openocd' is not recognized as an internal or external command,
operable program or batch file.
Command failed: openocd --version
'openocd' is not recognized as an internal or external command,
operable program or batch file.
Error checking openocd-esp32 version
Error: Command failed: openocd --version
'openocd' is not recognized as an internal or external command,
operable program or batch file.

'cmake' is not recognized as an internal or external command,
operable program or batch file.
Command failed: cmake --version
'cmake' is not recognized as an internal or external command,
operable program or batch file.
Error checking cmake version
Error: Command failed: cmake --version
'cmake' is not recognized as an internal or external command,
operable program or batch file.

'ninja' is not recognized as an internal or external command,
operable program or batch file.
Command failed: ninja --version
'ninja' is not recognized as an internal or external command,
operable program or batch file.
Error checking ninja version
Error: Command failed: ninja --version
'ninja' is not recognized as an internal or external command,
operable program or batch file.

'idf.py.exe' is not recognized as an internal or external command,
operable program or batch file.
Command failed: idf.py.exe -v
'idf.py.exe' is not recognized as an internal or external command,
operable program or batch file.
Error checking idf-exe version
Error: Command failed: idf.py.exe -v
'idf.py.exe' is not recognized as an internal or external command,
operable program or batch file.

'ccache.exe' is not recognized as an internal or external command,
operable program or batch file.
Command failed: ccache.exe --version
'ccache.exe' is not recognized as an internal or external command,
operable program or batch file.
Error checking ccache version
Error: Command failed: ccache.exe --version
'ccache.exe' is not recognized as an internal or external command,
operable program or batch file.

'dfu-util' is not recognized as an internal or external command,
operable program or batch file.
Command failed: dfu-util --version
'dfu-util' is not recognized as an internal or external command,
operable program or batch file.
Error checking dfu-util version
Error: Command failed: dfu-util --version
'dfu-util' is not recognized as an internal or external command,
operable program or batch file.

### Description

Fresh install. Configured using express option.
Trying to use menuconfig, getting this message.
```
[SDK Configuration Editor]
---------------------------ERROR--------------------------

ESP-IDF v5.1.1

-----------------------END OF ERROR-----------------------
[SDK Configuration Editor]
---------------------------ERROR--------------------------

"cmake" must be available on the PATH to use idf.py

-----------------------END OF ERROR-----------------------
[SDK Configuration Editor]
Executing action: confserver

[SDK Configuration Editor]
---------------------------ERROR--------------------------

SDK Configuration editor confserver process exited with code: 2
-----------------------END OF ERROR-----------------------
```
Also see doctor output. 

### Debug Message

```plain
1
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

radurentea, 2023-10-23:
Hi @rkit1,

The output above doesn't seem to be the whole output from the Doctor Command. Can you run the command and paste the whole output here, or in a .txt file? After you run the command, the message will be copied to your clipboard.

rkit1, 2023-10-23:
That's all I get.
https://www.youtube.com/watch?v=eU75yTLeANc

rkit1, 2023-10-23:
Sorry, it is in clipboard
```
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.22621 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\PuTTY\;C:\Program Files (x86)\Bitvise SSH Client;C:\Program Files\nodejs\;C:\Program Files\Git\cmd;C:\Program Files\Docker\Docker\resources\bin;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin;C:<HOMEPATH>\AppData\Roaming\npm 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.83.1 
Visual Studio Code language en 
Visual Studio Code shell C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.6.5 
Workspace folder z:\wrk\electro\esp32-bt-serial 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:<HOMEPATH>\esp\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) C:\Espressif\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin;C:\Espressif\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin;C:\Espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin;C:\Espressif\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin;C:\Espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin;C:\Espressif\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin;C:\Espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:\Espressif\tools\cmake\3.24.0\bin;C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230419\openocd-esp32\bin;C:\Espressif\tools\ninja\1.10.2;C:\Espressif\tools\idf-exe\1.0.3;C:\Espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64;C:\Espressif\tools\dfu-util\0.11\dfu-util-0.11-win64;C:\Espressif\tools\esp-rom-elfs\20230320
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230419/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
    ESP_ROM_ELF_DIR: C:\Espressif\tools\esp-rom-elfs\20230320/
Virtual env Python Path (idf.pythonBinPath) C:\Espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe
Serial port (idf.port) COM1
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) C:\Espressif
Git Path (idf.gitPath) C:\Espressif\tools\idf-git\2.39.2\cmd\git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to C:\Espressif\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin: true
Access to C:\Espressif\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin: true
Access to C:\Espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin: true
Access to C:\Espressif\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin: true
Access to C:\Espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin: true
Access to C:\Espressif\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin: true
Access to C:\Espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin: true
Access to C:\Espressif\tools\cmake\3.24.0\bin: true
Access to C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230419\openocd-esp32\bin: true
Access to C:\Espressif\tools\ninja\1.10.2: true
Access to C:\Espressif\tools\idf-exe\1.0.3: true
Access to C:\Espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64: true
Access to C:\Espressif\tools\dfu-util\0.11\dfu-util-0.11-win64: true
Access to C:\Espressif\tools\esp-rom-elfs\20230320: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) true
-------------------------------------------------------- Configurations has spaces -------------------------------------------------------------
Spaces in system environment Path true
Spaces in ESP-ADF Path (idf.espAdfPath) false
Spaces in ESP-IDF Path (idf.espIdfPath) false
Spaces in ESP-MDF Path (idf.espMdfPath) false
Spaces in ESP-Matter Path (idf.espMatterPath) false
Spaces in ESP-IDF Custom extra paths
Spaces in C:\Espressif\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin: false
Spaces in C:\Espressif\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin: false
Spaces in C:\Espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin: false
Spaces in C:\Espressif\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin: false
Spaces in C:\Espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin: false
Spaces in C:\Espressif\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin: false
[... cut ...]
---------------------------------------------------- Check ESP-IDF debug adapter requirements.txt ------------------------------------------
Check Debug AdapterPython packages Python requirements are satisfied.
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at Unknown error in ESP-IDF doctor command
--------------------------------------------------------------------------------------------------------------------------------------------

```

radurentea, 2023-10-26:
This is very strange, maybe there you have some settings in a project that overrides the global settings? In a `.vscode/settings.json in your project?
`
When you configured the extension did you have any errors?

Also, when you say fresh install, is it the first time using the extension? Cause sometimes users don't follow the uninstall steps and when they reinstall the extensions some errors might occur. [Uninstall guide](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/INSTALL.md#uninstalling-the-plugin)

rkit1, 2023-10-29:
Fully uninstalling and reinstalling fixed the problem. Although that was my first time using the extension.

rkit1, 2023-10-29:
Problem returned after restarting VS Code.

radurentea, 2023-10-31:
It seems the extension configuration gets overwritten by other settings. Vscode stores its settings in a JSON file. There are two types of settings:

**User Settings**: These are global settings that apply to any instance of VSCode you open. User settings are stored in a settings.json file located in the user's Visual Studio Code configuration directory. The location of this file varies by operating system:

Windows: %APPDATA%\Code\User\settings.json
macOS: ~/Library/Application Support/Code/User/settings.json
Linux: ~/.config/Code/User/settings.json

**Workspace Settings** (the ones I've mentioned in my prev reply): These are settings that apply only to the specific project you are working on. Workspace settings are stored in a .vscode folder within your project's root directory, in a file named settings.json.

Can you paste both the json files here to check them?

rkit1, 2023-11-01:
1. Create new Windows 11 user.
2. Install VS Code for that user only.
3. Install the extension. Don't install anything else.
4. Configure the extension using the express option.
5. Create new project from template.
6. SDK configuration works.
7. Close and reopen VS Code.
8. SDK configuration does not work.

rkit1, 2023-11-01:
```
{
    "idf.gitPathWin": "C:\\Users\\test\\.espressif\\tools\\idf-git\\2.39.2\\cmd\\git.exe",
    "idf.espIdfPathWin": "C:\\Users\\test\\esp\\esp-idf",
    "idf.pythonBinPathWin": "C:\\Users\\test\\.espressif\\python_env\\idf5.1_py3.11_env\\Scripts\\python.exe",
    "idf.toolsPathWin": "C:\\Users\\test\\.espressif",
    "idf.customExtraPaths": "C:\\Users\\test\\.espressif\\tools\\xtensa-esp-elf-gdb\\12.1_20221002\\xtensa-esp-elf-gdb\\bin;C:\\Users\\test\\.espressif\\tools\\riscv32-esp-elf-gdb\\12.1_20221002\\riscv32-esp-elf-gdb\\bin;C:\\Users\\test\\.espressif\\tools\\xtensa-esp32-elf\\esp-12.2.0_20230208\\xtensa-esp32-elf\\bin;C:\\Users\\test\\.espressif\\tools\\xtensa-esp32s2-elf\\esp-12.2.0_20230208\\xtensa-esp32s2-elf\\bin;C:\\Users\\test\\.espressif\\tools\\xtensa-esp32s3-elf\\esp-12.2.0_20230208\\xtensa-esp32s3-elf\\bin;C:\\Users\\test\\.espressif\\tools\\riscv32-esp-elf\\esp-12.2.0_20230208\\riscv32-esp-elf\\bin;C:\\Users\\test\\.espressif\\tools\\esp32ulp-elf\\2.35_20220830\\esp32ulp-elf\\bin;C:\\Users\\test\\.espressif\\tools\\cmake\\3.24.0\\bin;C:\\Users\\test\\.espressif\\tools\\openocd-esp32\\v0.12.0-esp32-20230419\\openocd-esp32\\bin;C:\\Users\\test\\.espressif\\tools\\ninja\\1.10.2;C:\\Users\\test\\.espressif\\tools\\idf-exe\\1.0.3;C:\\Users\\test\\.espressif\\tools\\ccache\\4.8\\ccache-4.8-windows-x86_64;C:\\Users\\test\\.espressif\\tools\\dfu-util\\0.11\\dfu-util-0.11-win64;C:\\Users\\test\\.espressif\\tools\\esp-rom-elfs\\20230320",
    "idf.customExtraVars": {
        "OPENOCD_SCRIPTS": "C:\\Users\\test\\.espressif\\tools\\openocd-esp32\\v0.12.0-esp32-20230419/openocd-esp32/share/openocd/scripts",
        "IDF_CCACHE_ENABLE": "1",
        "ESP_ROM_ELF_DIR": "C:\\Users\\test\\.espressif\\tools\\esp-rom-elfs\\20230320/"
    }
}
```

keck9939, 2023-11-04:
I have what seems to be exactly the same issue. Any progress on solving it?

keck9939, 2023-11-04:
After posting the above, I tried creating a new user and installed VS Code and the IDF extension in that. I had initially the same issues, but noticed the fact that IDF_PYTHON_ENV_PATH was shown as undefined. So I set that environment variable to point to the IDF plugin Python location and that fixed it.
Then I went back to my original account, completely uninstalled the plugin and re-installed and it worked even though IDF_PYTHON_ENV_PATH was still shown (correctly) as undefined. In view of the fact that the Python path is specified in the user settings.json file it is not clear why it should have needed IDF_PYTHON_ENV_PATH, why it didn't work when I first installed it in my regular account and then it did work when I reinstalled. Something is wrong with the way the extension is handling this.

keck9939, 2023-12-09:
I am glad to see this has been reopened. After working for quite a while, once again the VS Code Esp IDE extension has stopped working because it cannot find cmake or ninja. I suspect the reason it broke is that VS Code recently was automatically updated, although I can't be sure. For what its worth, the Doctor report is attached.
[report.txt](https://github.com/espressif/vscode-esp-idf-extension/files/13622260/report.txt)

radurentea, 2023-12-18:
@keck9939, can you give it another try using the master branch?

keck9939, 2023-12-18:
I tried the master branch and for the first time it actually found and allowed me to use the standalone ESP-IDF install that was already on the machine, so that is very promising. It built my project without issues. Hopefully, it will continue to work and not mysteriously break again.

radurentea, 2024-01-04:
Fixed by the following PR: https://github.com/espressif/vscode-esp-idf-extension/pull/1093

closed 2024-01-04 by radurentea