espressif/vscode-esp-idf-extension #1067: The 'Flash device' button does not support encrypted mode

owen63, 2023-10-24:
### OS

Windows

### Operating System version

Windows 10

### Visual Studio Code version

1.6.5

### ESP-IDF version

5.0.1

### Python version

3.11.1

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19045 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\Git\cmd;C:\Ruby31-x64\bin;C:\Program Files\PuTTY\;C:\ST\STM32CubeCLT;C:\ST\STM32CubeCLT\STM32CubeProgrammer\bin;C:\ST\STM32CubeCLT\STLink-gdb-server\bin;C:\ST\STM32CubeCLT\GNU-tools-for-STM32\bin;C:\Program Files (x86)\STMicroelectronics\stlink_server;C:\Program Files\Python311\Scripts\;C:\Program Files\Python311\;C:\Program Files\TortoiseGit\bin;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\AppData\Roaming\Python\Python311\Scripts 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.83.1 
Visual Studio Code language en 
Visual Studio Code shell C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.6.5 
Workspace folder c:<HOMEPATH>\FTN\streetdog_ccu 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) c:<HOMEPATH>\FTN\esp-idf-v5.0.1
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32s2-elf\esp-2022r1-11.2.0\xtensa-esp32s2-elf\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32s3-elf\esp-2022r1-11.2.0\xtensa-esp32s3-elf\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-2022r1-11.2.0\riscv32-esp-elf\bin;C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:<HOMEPATH>\.espressif\tools\cmake\3.24.0\bin;C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\bin;C:<HOMEPATH>\.espressif\tools\ninja\1.10.2;C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3;C:<HOMEPATH>\.espressif\tools\ccache\4.6.2\ccache-4.6.2-windows-x86_64;C:<HOMEPATH>\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64;C:<HOMEPATH>\.espressif\tools\esp-rom-elfs\20220823
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.11.0-esp32-20221026/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
    ESP_ROM_ELF_DIR: C:<HOMEPATH>\.espressif\tools\esp-rom-elfs\20220823/
Virtual env Python Path (idf.pythonBinPath) C:<HOMEPATH>\.espressif\python_env\idf5.0_py3.8_env\Scripts\python.exe
Serial port (idf.port) COM5
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) C:<HOMEPATH>\.espressif
Git Path (idf.gitPath) C:<HOMEPATH>\.espressif\tools\idf-git\2.30.1\cmd\git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32s2-elf\esp-2022r1-11.2.0\xtensa-esp32s2-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32s3-elf\esp-2022r1-11.2.0\xtensa-esp32s3-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-2022r1-11.2.0\riscv32-esp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\cmake\3.24.0\bin: true
Access to C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\bin: true
Access to C:<HOMEPATH>\.espressif\tools\ninja\1.10.2: true
Access to C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3: true
Access to C:<HOMEPATH>\.espressif\tools\ccache\4.6.2\ccache-4.6.2-windows-x86_64: true
Access to C:<HOMEPATH>\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64: true
Access to C:<HOMEPATH>\.espressif\tools\esp-rom-elfs\20220823: true
[... 162 lines cut ...]
    "flash_files" : {
        "0x0" : "bootloader/bootloader.bin",
        "0x20000" : "app.bin",
        "0x8000" : "partition_table/partition-table.bin"
    },
    "bootloader" : { "offset" : "0x0", "file" : "bootloader/bootloader.bin", "encrypted" : "true" },
    "app" : { "offset" : "0x20000", "file" : "app.bin", "encrypted" : "true" },
    "partition-table" : { "offset" : "0x8000", "file" : "partition_table/partition-table.bin", "encrypted" : "true" },
    "extra_esptool_args" : {
        "after"  : "no_reset",
        "before" : "default_reset",
        "stub"   : true,
        "chip"   : "esp32s3"
    }
}

### Debug Message

```plain
*  Executing task: C:\xxxxx\.espressif\python_env\idf5.0_py3.8_env\Scripts\python.exe c:\xxxxxx\esp-idf-v5.0.1\components\esptool_py\esptool\esptool.py -p COM5 -b 460800 --before default_reset --after no_reset --chip esp32s3 write_flash --flash_mode dio --flash_freq 80m --flash_size 16MB 0x0 bootloader/bootloader.bin 0x20000 app.bin 0x8000 partition_table/partition-table.bin
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

chipweinberger, 2023-12-31:
related https://github.com/espressif/vscode-esp-idf-extension/issues/853

radurentea, 2024-01-18:
Hi @owen63,

It seems that flash encryption does work, but in order to flash with that --encrypt flash you would have to use the command "ESP-IDF: Encrypt and flash your project".

Make sure to understand the following about flash encryption:
1) Enable flash encryption in sdk config (you did this)
2) Flash the device normally (without --encrypt flag)
3) After normal flashing is done, simply press and release the reset button. This action triggers a hardware reset, causing the device to reboot.
4) Flash device using "ESP-IDF: Encrypt and flash your project" (which will add the --encrypt flag to the flash command)

I'm currently working on an implementation which will be more intuitive for users, not having to worry about the method they chose to flash the device.

@chipweinberger , I will try to fix your request from #853 as well in this PR.

owen63, 2024-01-18:
Hi Radu,

Thanks, I was not aware of the "ESP-IDF: Encrypt and flash your project"
command.

closed 2024-11-19 by radurentea

themindfactory, 2024-11-22:
How does it know what key file to use to encrypt the files? .bin is usual extension....