espressif/vscode-esp-idf-extension #1070: why i cant use [SDK Configuration Editor] with vscode ? (IDFGH-11323)

zhou88mang, 2023-10-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

i can use idf.py menuconfig by command line , and i hope to use SDK Configuration Editor with   vscode (in the ubuntu18.04)

### What is the actual behavior?

i can build my project with vscode ( ESP-IDF Setup is success), but i cant  use SDK Configuration Editor with   vscode

### Steps to reproduce.

 1. Step1：Click on the gear icon on vscode

### Build or installation Logs.

```plain
[SDK Configuration Editor]
---------------------------ERROR--------------------------

ERROR: ./bin/python doesn't exist! Please run the install script or "idf_tools.py install-python-env" in order to create it

-----------------------END OF ERROR-----------------------
[SDK Configuration Editor]
---------------------------ERROR--------------------------

ESP-IDF v5.1.1-dirty

-----------------------END OF ERROR-----------------------
[SDK Configuration Editor]
---------------------------ERROR--------------------------

SDK Configuration editor confserver process exited with code: 1
-----------------------END OF ERROR-----------------------
```

### More Information.

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS linux x64 5.4.0-150-generic 
System environment variable IDF_PYTHON_ENV_PATH 
 <HOMEPATH>/.espressif/python_env/idf5.1_py3.10_env 
System environment variable PATH 
 <HOMEPATH>/esp/esp-idf/components/espcoredump:<HOMEPATH>/esp/esp-idf/components/partition_table:<HOMEPATH>/esp/esp-idf/components/app_update:<HOMEPATH>/esp/esp-idf/components/espcoredump:<HOMEPATH>/esp/esp-idf/components/partition_table:<HOMEPATH>/esp/esp-idf/components/app_update:<HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/12.1_20221002/xtensa-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/12.1_20221002/riscv32-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin:<HOMEPATH>/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin:<HOMEPATH>/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin:<HOMEPATH>/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:<HOMEPATH>/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230419/openocd-esp32/bin:<HOMEPATH>/.espressif/python_env/idf5.1_py3.10_env/bin:<HOMEPATH>/esp/esp-idf/tools:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:<HOMEPATH>/zm_work/compile_tools/gcc-arm-none-eabi-8-2019-q3-update/bin:<HOMEPATH>/zm_work/compile_tools/gcc-arm-none-eabi-8-2019-q3-update/bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.83.1 
Visual Studio Code language en 
Visual Studio Code shell /bin/bash 
ESP-IDF Extension version 1.6.5 
[... 128 lines cut ...]
            "name": "ESP-IDF",
            "compilerPath": "${config:idf.toolsPath}/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc",
            "includePath": [
                "${config:idf.espIdfPath}/components/**",
                "${config:idf.espIdfPathWin}/components/**",
                "${config:idf.espAdfPath}/components/**",
                "${config:idf.espAdfPathWin}/components/**",
                "${workspaceFolder}/**"
            ],
            "browse": {
                "path": [
                    "${config:idf.espIdfPath}/components",
                    "${config:idf.espIdfPathWin}/components",
                    "${config:idf.espAdfPath}/components/**",
                    "${config:idf.espAdfPathWin}/components/**",
                    "${workspaceFolder}"
                ],
                "limitSymbolsToIncludedHeaders": false
            }
        }
    ],
    "version": 4
}
 
--------------------------------------------------------------------------------------------------------------------------------------------

brianignacio5, 2023-11-01:
You need to fix the value in `idf.pythonBinPath` in your settings.json. Right now is only `Virtual env Python Path (idf.pythonBinPath) python`

zhou88mang, 2023-11-01:
thanks, I have solved this problem ！

closed 2023-11-01 by brianignacio5

MarkoMk, 2024-09-01:
> You need to fix the value in `idf.pythonBinPath` in your settings.json. Right now is only `Virtual env Python Path (idf.pythonBinPath) python`

I have the same issue, where it just loads and doesn't ever open. Any advice? Using ubuntu 22.04 if that changes anything. And all of the paths are correct. I can buld projects and flash them to the board, I just can't open the SDK Configuration Editor.

DANVEZ, 2025-05-01:
i have the similar problem with vscode.  May i ask how u solve this problem? When I click the button"SDK configuration editor(menuconfig)", the problem appeared.

The problem is blow :

[SDK Configuration Editor]
---------------------------ERROR--------------------------

WARNING: Git version unavailable, reading from source

-----------------------END OF ERROR-----------------------
---------------------------ERROR--------------------------

ESP-IDF v5.4.1
"cmake" must be available on the PATH to use idf.py

-----------------------END OF ERROR-----------------------
Executing action: confserver

---------------------------ERROR--------------------------

SDK Configuration editor confserver process exited with code: 2
-----------------------END OF ERROR-----------------------

ESP-IDF v5.4.1
board is esp32s3