espressif/vscode-esp-idf-extension #1078: Cant execute any of esp-idf commands within vscode

S-A-M-J, 2023-11-14:
### OS

MacOS

### Operating System version

macOs Sonoma

### Visual Studio Code version

1.74.2

### ESP-IDF version

5.1.1

### Python version

3.9

### Doctor command output

cant execute doctor command: returns command not found

### Extension

This file doesnt exist.

### Description

Vscode doesn't seem to find any of the executables or variables linked to esp-idf. I even added IDF_PATH globally to my zhsrc file but vscode just doesnt want to see it. Result is that esp-idf files are not identified as such, none of the commands from the command palette work and cmake complains that $env{IDF_PATH} is empty. 
I tried reinstalling the extension multiple times but no dice. 

### Debug Message

```plain
Command 'ESP-IDF: Configure ESP-IDF extension' resulted in an error (command 'espIdf.setup.start' not found)
```

### Other Steps to Reproduce

Don't know how to reproduce it. It suddenly appeared. Unfortunately i don't have a second mac to test on.
Similar issues apparently were encountered[here](https://github.com/espressif/vscode-esp-idf-extension/issues/907). Apparently fixed back then but issue reappeared. 

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2023-11-15:
I'm not sure what is happening in your setup and you provide no troubleshooting information to reproduce it.

Can you install the master branch [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/workflows/ci/master/esp-idf-extension.vsix.zip) ?

luukWeFab, 2023-11-23:
I'm having the same problem but then on windows. I tried to look for "%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log" but the file just doesn't exist. I installed the linked vsix installer, but this did not solve the problem. There is also nothing printed in the output console of ESP-IDF with loglevel 6