espressif/vscode-esp-idf-extension #1080: Extension never gets past "installation" stage

xobs, 2023-11-20:
### OS

Windows

### Operating System version

Windows 11 22H2 (22621.2428)

### Visual Studio Code version

1.84.1

### ESP-IDF version

a7fbf452fa181452f266b16c49063e2ff069f119

### Python version

3.10.11

### Doctor command output

There is no output when I run the command. However, I see from the log file that it is copied to the clipboard. This should be communicated to the user.

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.22621 
System environment variable IDF_PYTHON_ENV_PATH 
 C:<HOMEPATH>\.espressif\python_env\idf5.2_py3.10_env 
System environment variable PATH 
 C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:<HOMEPATH>\.espressif\tools\cmake\3.24.0\bin;C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin;C:<HOMEPATH>\.espressif\tools\ninja\1.11.1\;C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3\;C:<HOMEPATH>\.espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64;C:<HOMEPATH>\.espressif\tools\dfu-util\0.11\dfu-util-0.11-win64;C:<HOMEPATH>\.espressif\python_env\idf5.2_py3.10_env\Scripts;C:<HOMEPATH>\esp\esp-idf\tools;C:\Program Files\PowerShell\7;;C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v12.0\bin;C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v12.0\libnvvp;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\dotnet\;C:\Program Files (x86)\MATLAB\MATLAB Runtime\v85\runtime\win32;C:\Program Files\Microsoft Network Monitor 3\;C:\Program Files (x86)\Windows Kits\10\Microsoft Application Virtualization\Sequencer\;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files\usbipd-win\;C:\Program Files\NVIDIA Corporation\Nsight Compute 2022.4.0\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files (x86)\dotnet\;C:\Program Files\Git\cmd;C:\Program Files\ojdkbuild\java-17-openjdk-17.0.3.0.6-1\bin;C:\Program Files (x86)\Windows Kits\10\Windows Performance Toolkit\;C:\Program Files\CMake\bin;C:\Program Files\Docker\Docker\resources\bin;C:\Program Files\Go\bin;C:\Program Files\nodejs\;E:\Software\Liquid Technologies\Liquid Studio\XmlDataBinder20\Redist20\cpp\win32\bin;E:\Software\Liquid Technologies\Liquid Studio\XmlDataBinder20\Redist20\cpp\win64\bin;C:\Program Files\PowerShell\7\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:<HOMEPATH>\.rustup\toolchains\esp\riscv32-esp-elf\esp-13.2.0_20230928\riscv32-esp-elf\bin;C:<HOMEPATH>\.rustup\toolchains\esp\xtensa-esp-elf\esp-13.2.0_20230928\xtensa-esp-elf\bin;C:<HOMEPATH>\.rustup\toolchains\esp\xtensa-esp32-elf-clang\esp-16.0.0-20230516\esp-clang\bin;C:\Program Files\PowerShell\7;;C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v12.0\bin;C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v12.0\libnvvp;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\dotnet\;C:\Program Files (x86)\MATLAB\MATLAB Runtime\v85\runtime\win32;C:\Program Files\Microsoft Network Monitor 3\;C:\Program Files (x86)\Windows Kits\10\Microsoft Application Virtualization\Sequencer\;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files\usbipd-win\;C:\Program Files\NVIDIA Corporation\Nsight Compute 2022.4.0\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files (x86)\dotnet\;C:\Program Files\Git\cmd;C:\Program Files\ojdkbuild\java-17-openjdk-17.0.3.0.6-1\bin;C:\Program Files (x86)\Windows Kits\10\Windows Performance Toolkit\;C:\Program Files\CMake\bin;C:\Program Files\Docker\Docker\resources\bin;C:\Program Files\Go\bin;C:\Program Files\nodejs\;E:\Software\Liquid Technologies\Liquid Studio\XmlDataBinder20\Redist20\cpp\win32\bin;E:\Software\Liquid Technologies\Liquid Studio\XmlDataBinder20\Redist20\cpp\win64\bin;C:\Program Files\PowerShell\7\;C:<HOMEPATH>\.cargo\bin;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin;C:<HOMEPATH>\.dotnet\tools;C:\Program Files (x86)\Nmap;C:<HOMEPATH>\AppData\Local\Microsoft\WinGet\Links;E:\Software\Renode\bin;C:<HOMEPATH>\go\bin;E:\Software\flutter\bin;C:<HOMEPATH>\AppData\Roaming\npm;E:\Software\Microsoft VS Code Insiders\bin;E:\Software\xpack-riscv-none-elf-gcc-12.2.0-3\bin\;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.84.1 
Visual Studio Code language en 
Visual Studio Code shell C:\Program Files\PowerShell\7\pwsh.exe 
ESP-IDF Extension version 1.6.5 
Workspace folder e:\Code\Farpatch\farpatch 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:<HOMEPATH>\esp\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\12.1_20231023\xtensa-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\12.1_20231023\riscv32-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf\esp-13.2.0_20230928\xtensa-esp-elf\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-13.2.0_20230928\riscv32-esp-elf\bin;C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:<HOMEPATH>\.espressif\tools\cmake\3.24.0\bin;C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin;C:<HOMEPATH>\.espressif\tools\ninja\1.11.1;C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3;C:<HOMEPATH>\.espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64;C:<HOMEPATH>\.espressif\tools\dfu-util\0.11\dfu-util-0.11-win64;C:<HOMEPATH>\.espressif\tools\esp-rom-elfs\20230320
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230921/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
    ESP_ROM_ELF_DIR: C:<HOMEPATH>\.espressif\tools\esp-rom-elfs\20230320/
Virtual env Python Path (idf.pythonBinPath) C:<HOMEPATH>\.espressif\python_env\idf5.0_py3.10_env\Scripts\python.exe
Serial port (idf.port) COM14
OpenOCD Configs (idf.openOcdConfigs) board/esp32s3-builtin.cfg
ESP-IDF Tools Path (idf.toolsPath) C:<HOMEPATH>\.espressif
Git Path (idf.gitPath) C:<HOMEPATH>\.espressif\tools\idf-git\2.39.2\cmd\git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\12.1_20231023\xtensa-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\12.1_20231023\riscv32-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf\esp-13.2.0_20230928\xtensa-esp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-13.2.0_20230928\riscv32-esp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\cmake\3.24.0\bin: true
Access to C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin: true
Access to C:<HOMEPATH>\.espressif\tools\ninja\1.11.1: true
Access to C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3: true
Access to C:<HOMEPATH>\.espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64: true
Access to C:<HOMEPATH>\.espressif\tools\dfu-util\0.11\dfu-util-0.11-win64: true
Access to C:<HOMEPATH>\.espressif\tools\esp-rom-elfs\20230320: true
[... 40 lines cut ...]
--------------------------------------------------------------------------------------------------------------------------------------------

### Extension

[This text cannot be included here because it is longer than 65536 characters]

### Description

[... cut ...] ck to the Setup screen without giving an error.

### Debug Message

```plain
N/A
```

### Other Steps to Reproduce

No idea. My solution right now is to ignore the `esp-idf` extension and do everything via Terminal, since the extension just isn't working. This used to work, but some update somewhere has broken it.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2023-11-20:
Can you share the extension log lines related to the setup ? There should be something related to python packages install based on your description.

Seems to me you have conflicting settings in your project and the extension. See that Doctor command has `idf.pythonBinPath` to different location from what you are mentioning in the setup.

xobs, 2023-11-20:
Here is the file, compressed as a zip file: [esp_idf_vsc_ext.zip](https://github.com/espressif/vscode-esp-idf-extension/files/13412116/esp_idf_vsc_ext.zip)

There is no error message that appears -- it just keeps sending me back to the same setup screen. Perhaps one should be added?

To the best of my knowledge, I haven't updated my `settings.json` file with any ESP-specific configuration, so if there's an inconsistency there then it is a bug caused by esp-idf.

I do see something has set `idf.pythonBinPathWin` in my `settings.json` to `C:\\Users\\Sean\\.espressif\\python_env\\idf5.1_py3.8_env\\Scripts\\python.exe` which is a valid path. What could have set that variable, and how should it be fixed?

brianignacio5, 2023-12-07:
What is the output when you run this in terminal:

`C:\\Users\\Sean\\.espressif\\python_env\\idf5.1_py3.8_env\\Scripts\\python.exe\" \"C:\\Users\\Sean\\esp\\esp-idf\\tools\\check_python_dependencies.py\" -r \"C:\\Users\\Sean\\esp\\esp-idf\\tools\\requirements\\requirements.core.txt\"`

What is the output of installation stage ? Check menu View -> Output -> Select ESP-IDF from dropdown. What is the output from setup?

xobs, 2023-12-07:
Something must have changed. When I started up `vscode-esp-idf-extension` today it said this:

```
idf_versions.txt has (1) KB
idf_versions.txt progress: 100.00% (0.08 / 0.08) KB
The following Python requirements are not satisfied:
'esp-idf-panic-decoder' - was not found and is required by the application

'pyclang' - was not found and is required by the application
To install the missing packages, please run "install.bat"
Diagnostic information:
    IDF_PYTHON_ENV_PATH: C:\Users\Sean\.espressif\python_env\idf5.0_py3.10_env
    Python interpreter used: C:\Users\Sean\.espressif\python_env\idf5.1_py3.8_env\Scripts\python.exe
    Warning: python interpreter not running from IDF_PYTHON_ENV_PATH
    PATH: C:\Users\Sean\esp\esp-idf\components\esptool_py\esptool;C:\Users\Sean\esp\esp-idf\components\espcoredump;C:\Users\Sean\esp\esp-idf\components\partition_table;C:\Users\Sean\.espressif\tools\xtensa-esp-elf-gdb\12.1_20231023\xtensa-esp-elf-gdb\bin;C:\Users\Sean\.espressif\tools\riscv32-esp-elf-gdb\12.1_20231023\riscv32-esp-elf-gdb\bin;C:\Users\Sean\.espressif\tools\xtensa-esp-elf\esp-13.2.0_20230928\xtensa-esp-elf\bin;C:\Users\Sean\.espressif\tools\riscv32-esp-elf\esp-13.2.0_20230928\riscv32-esp-elf\bin;C:\Users\Sean\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:\Users\Sean\.espressif\tools\cmake\3.24.0\bin;C:\Users\Sean\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin;C:\Users\Sean\.espressif\tools\ninja\1.11.1;C:\Users\Sean\.espressif\tools\idf-exe\1.0.3;C:\Users\Sean\.espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64;C:\Users\Sean\.espressif\tools\dfu-util\0.11\dfu-util-0.11-win64;C:\Users\Sean\.espressif\tools\esp-rom-elfs\20230320;C:\Users\Sean\.espressif\python_env\idf5.0_py3.10_env\Scripts;C:\Users\Sean\esp\esp-idf\tools;C:\Users\Sean\.espressif\tools\idf-git\2.30.1\cmd;C:\Program Files\PowerShell\7;;C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v12.0\bin;C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v12.0\libnvvp;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\dotnet\;C:\Program Files (x86)\MATLAB\MATLAB Runtime\v85\runtime\win32;C:\Program Files\Microsoft Network Monitor 3\;C:\Program Files (x86)\Windows Kits\10\Microsoft Application Virtualization\Sequencer\;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files\usbipd-win\;C:\Program Files\NVIDIA Corporation\Nsight Compute 2022.4.0\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files (x86)\dotnet\;C:\Program Files\Git\cmd;C:\Program Files\ojdkbuild\java-17-openjdk-17.0.3.0.6-1\bin;C:\Program Files (x86)\Windows Kits\10\Windows Performance Toolkit\;C:\Program Files\CMake\bin;C:\Program Files\Docker\Docker\resources\bin;C:\Program Files\Go\bin;C:\Program Files\nodejs\;E:\Software\Liquid Technologies\Liquid Studio\XmlDataBinder20\Redist20\cpp\win32\bin;E:\Software\Liquid Technologies\Liquid Studio\XmlDataBinder20\Redist20\cpp\win64\bin;C:\Program Files\PowerShell\7\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Users\Sean\.rustup\toolchains\esp\riscv32-esp-elf\esp-13.2.0_20230928\riscv32-esp-elf\bin;C:\Users\Sean\.rustup\toolchains\esp\xtensa-esp-elf\esp-13.2.0_20230928\xtensa-esp-elf\bin;C:\Users\Sean\.rustup\toolchains\esp\xtensa-esp32-elf-clang\esp-16.0.0-20230516\esp-clang\bin;C:\Program Files\PowerShell\7;C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v12.0\bin;C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v12.0\libnvvp;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\dotnet\;C:\Program Files (x86)\MATLAB\MATLAB Runtime\v85\runtime\win32;C:\Program Files\Microsoft Network Monitor 3\;C:\Program Files (x86)\Windows Kits\10\Microsoft Application Virtualization\Sequencer\;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files\usbipd-win\;C:\Program Files\NVIDIA Corporation\Nsight Compute 2022.4.0\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files (x86)\dotnet\;C:\Program Files\Git\cmd;C:\Program Files\ojdkbuild\java-17-openjdk-17.0.3.0.6-1\bin;C:\Program Files (x86)\Windows Kits\10\Windows Performance Toolkit\;C:\Program Files\CMake\bin;C:\Program Files\Docker\Docker\resources\bin;C:\Program Files\Go\bin;C:\Program Files\nodejs\;E:\Software\Liquid Technologies\Liquid Studio\XmlDataBinder20\Redist20\cpp\win32\bin;C:\Program Files (x86)\GnuWin32\bin;C:\Program Files\PowerShell\7\;C:\Users\Sean\.cargo\bin;C:\Users\Sean\AppData\Local\Microsoft\WindowsApps;C:\Users\Sean\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\Sean\.dotnet\tools;C:\Program Files (x86)\Nmap;C:\Users\Sean\AppData\Local\Microsoft\WinGet\Links;E:\Software\Renode\bin;C:\Users\Sean\go\bin;E:\Software\flutter\bin;C:\Users\Sean\AppData\Roaming\npm;E:\Software\Microsoft VS Code Insiders\bin;E:\Software\xpack-riscv-none-elf-gcc-12.2.0-3\bin\;E:\Software\xpack-arm-none-eabi-gcc-11.2.1-1.2\bin;
Command failed: "C:\Users\Sean\.espressif\python_env\idf5.1_py3.8_env\Scripts\python.exe" "C:\Users\Sean\esp\esp-idf\tools\check_python_dependencies.py" -r "C:\Users\Sean\esp\esp-idf\tools\requirements\requirements.core.txt"
Installing ESP-IDF extension python packages in C:\Users\Sean\.espressif\python_env\idf5.1_py3.8_env\Scripts\python.exe ...

...
```

Then it proceeded to install as normal.

It seems like it might have been missing a module called `esp-idf-panic-decoder`, and that has since been remedied? If there was an improvement to be made it might be to check all dependencies before installing. But for now it "works for me". Please close the issue unless you would like to do more forensics.

closed 2023-12-18 by brianignacio5