espressif/vscode-esp-idf-extension #1083: Unable to resolve configuration with compilerPath

boris-gu, 2023-11-23:
### OS

Windows

### Operating System version

Windows 11

### Visual Studio Code version

1.84.2

### ESP-IDF version

5.1.2

### Python version

IDK, installed via ESP-IDF Setup

### Doctor command output

```
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.22621 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\Git\cmd;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\LLVM\bin;C:\Program Files\MATLAB\MATLAB Runtime\v95\runtime\win64;C:<HOMEPATH>\AppData\Local\Programs\Python\Python311\Scripts\;C:<HOMEPATH>\AppData\Local\Programs\Python\Python311\;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.84.2 
Visual Studio Code language en 
Visual Studio Code shell C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.6.5 
Workspace folder d:\Projects\ESP32\sample_project 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:<HOMEPATH>\esp\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin;C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:<HOMEPATH>\.espressif\tools\cmake\3.24.0\bin;C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin;C:<HOMEPATH>\.espressif\tools\ninja\1.10.2;C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3;C:<HOMEPATH>\.espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64;C:<HOMEPATH>\.espressif\tools\dfu-util\0.11\dfu-util-0.11-win64;C:<HOMEPATH>\.espressif\tools\esp-rom-elfs\20230320
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230921/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
    ESP_ROM_ELF_DIR: C:<HOMEPATH>\.espressif\tools\esp-rom-elfs\20230320/
Virtual env Python Path (idf.pythonBinPath) C:<HOMEPATH>\.espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe
Serial port (idf.port) COM1
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,target/esp32.cfg
ESP-IDF Tools Path (idf.toolsPath) C:<HOMEPATH>\.espressif
Git Path (idf.gitPath) C:<HOMEPATH>\.espressif\tools\idf-git\2.39.2\cmd\git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\cmake\3.24.0\bin: true
Access to C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin: true
Access to C:<HOMEPATH>\.espressif\tools\ninja\1.10.2: true
Access to C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3: true
Access to C:<HOMEPATH>\.espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64: true
Access to C:<HOMEPATH>\.espressif\tools\dfu-util\0.11\dfu-util-0.11-win64: true
[... 134 lines cut ...]

### Extension

I couldn't insert the log because "Comment is too long (maximum is 65536 characters)"

### Description

The project is building normally, but when running vs code, C/C++ issues a message
```
[11/23/2023, 9:02:41 AM] Unable to resolve configuration with compilerPath "${env:IDF_TOOLS_PATH}\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe".  Using "cl.exe" instead.
[11/23/2023, 9:02:41 AM] Unable to resolve configuration with compilerPath "${env:IDF_TOOLS_PATH}\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe".  Using "cl.exe" instead.
```
Is this normal behavior? I would like to remove these messages because they are very annoying.

[image]

### Debug Message

```plain
-
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2023-11-23:
Change `config:idf.toolsPath` to `config:idf.toolsPathWin`

boris-gu, 2023-11-23:
Thank you, it helped.
Is there anyway to configure it so that I don't have to change this text yourself?

boris-gu, 2023-11-23:
I tried installing the esp-id on another computer and there I had another problem with the text. When creating a project (clean or from an example), the problem line looks like this:  
`"compilerPath": "${config:idf.toolsPath}undefined",`  
Instead of the path to the compiler, it writes `undefined`. Why did this happen and how can I fix it?

_I do not know if I should create a new issue to solve this problem, so I wrote here_

brianignacio5, 2023-11-23:
> Thank you, it helped.
> Is there anyway to configure it so that I don't have to change this text myself?

I'll make a bug fix for this a new PR.

> I tried installing the esp-id on another computer and there I had another problem with this line. When creating a project (clean or from an example), the problem line looks like this: `"compilerPath": "${config:idf.toolsPath}undefined",` Instead of the path to the compiler, it writes `undefined`. Why did this happen and how can I fix it?
> 
> _I do not know if I should create a new issue to solve this problem, so I wrote here_

Seems your extension might not be configured since the last part is giving undefined. Will check out too so it doesn't use this value.

boris-gu, 2023-11-23:
I found a solution to the second problem: for some reason the username was capitalized. I don't remember changing anything in this field. Maybe I forgot or the problem is due to the synchronization of vs code (on one computer I have a user Boris and on the other boris)

[image]

brianignacio5, 2023-11-27:
It could be settings sync. I would recommend to disable if settings across machines have different locations.

Benjamin45, 2023-11-27:
> Change `config:idf.toolsPath` to `config:idf.toolsPathWin`

I have same issue, where I can change this settings please ?

boris-gu, 2023-11-27:
.vscode/c_cpp_properties.json in your project

closed 2024-01-03 by github-actions[bot]

Brandon-Ballard25, 2024-07-21:
Im getting the same error           "compilerPath": "${config:idf.toolsPathWin}undefined",

brianignacio5, 2024-07-22:
> Im getting the same error "compilerPath": "${config:idf.toolsPathWin}undefined",

This might be related to your extension not being configured properly at the time you created the project. Check the `ESP-IDF: Doctor Command` output. If extension is configured correctly you can re-create this file again with `ESP-IDF: Add vscode configuration folder` after deleting the original file.

PR #1084 (2024-01-08): fix duplicate python list add win compilerPath
## Description

Make sure python paths list does not have duplicated in `ESP-IDF: Configure ESP-IDF extension` command.

Fix c_cpp_properties.json compilerPath command on Windows replacement when IDF_TARGET is changed.

Fixes #1083

Fixes #1073 

## Type of change

Please delete options that are not relevant.
- Bug fix (non-breaking change which fixes an issue)

## Steps to test this pull request

Provide a list of steps to test changes in this PR and required output

1. Click on "ESP-IDF: Configure ESP-IDF extension" and check that there is no du [... cut]