espressif/vscode-esp-idf-extension #1086: Errors in the configuration of the extension

boris-gu, 2023-11-25:
### OS

Windows

### Operating System version

Windows 11

### Visual Studio Code version

1.84.2

### ESP-IDF version

5.1.2

### Python version

IDK, installed via ESP-IDF Setup

### Doctor command output

This command doesn't work for me. The message "ESP-IDF: Preparing ESP-IDF extension report" appears, but then nothing happens

### Extension

[esp_idf_vsc_ext.log.txt](https://github.com/espressif/vscode-esp-idf-extension/files/13463509/esp_idf_vsc_ext.log.txt)
I added .txt to the file name, because otherwise github refused to add it

### Description

When I configure the extension, there are errors in the output. There are no errors on the other computer.

```
spawn C:\Windows\system32\cmd.exe ENOENT
error: unable to create symlink porting/npl/riot/include/npl_syscfg/npl_sycfg.h: Permission denied
error: Submodule 'components/bt/host/nimble/nimble' could not be updated.
error: unable to create symlink examples/build_system/cmake/idf_as_lib/build-esp32.sh: Permission denied
error: unable to create symlink examples/build_system/cmake/idf_as_lib/build-esp32c2.sh: Permission denied
error: unable to create symlink examples/build_system/cmake/idf_as_lib/build-esp32c3.sh: Permission denied
error: unable to create symlink examples/build_system/cmake/idf_as_lib/build-esp32c6.sh: Permission denied
fatal: Could not reset index file to revision 'HEAD'.
Command failed: "C:\Users\Boris\.espressif\tools\idf-git\2.39.2\cmd\git.exe" reset --hard --recurse-submodule
```

These are just some mistakes. I saved the full output to a file
[esp uotput.txt](https://github.com/espressif/vscode-esp-idf-extension/files/13463527/esp.uotput.txt)

### Debug Message

```plain
-
```

### Other Steps to Reproduce

Steps to Reproduce:
1. Remove the extension
2. Delete folders `.espressif` `esp` and `.vscode\extensions\espressif.esp-idf-extension-1.6.5`
3. Install the extension
4. Open "configure ASP-IDF extension"
5. Start express installation

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

boris-gu, 2023-11-25:
I was able to open Doctor command output in a new vs code window

```
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.22621 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\Git\cmd;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\LLVM\bin;C:\Program Files\MATLAB\MATLAB Runtime\v95\runtime\win64;C:<HOMEPATH>\AppData\Local\Programs\Python\Python311\Scripts\;C:<HOMEPATH>\AppData\Local\Programs\Python\Python311\;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.84.2 
Visual Studio Code language en 
Visual Studio Code shell C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.6.5 
Workspace folder No workspace folder is open 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:<HOMEPATH>\esp\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin;C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:<HOMEPATH>\.espressif\tools\cmake\3.24.0\bin;C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin;C:<HOMEPATH>\.espressif\tools\ninja\1.10.2;C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3;C:<HOMEPATH>\.espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64;C:<HOMEPATH>\.espressif\tools\dfu-util\0.11\dfu-util-0.11-win64;C:<HOMEPATH>\.espressif\tools\esp-rom-elfs\20230320
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230921/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
    ESP_ROM_ELF_DIR: C:<HOMEPATH>\.espressif\tools\esp-rom-elfs\20230320/
Virtual env Python Path (idf.pythonBinPath) C:<HOMEPATH>\.espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe
Serial port (idf.port) COM1
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) C:<HOMEPATH>\.espressif
Git Path (idf.gitPath) C:<HOMEPATH>\.espressif\tools\idf-git\2.39.2\cmd\git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\cmake\3.24.0\bin: true
Access to C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin: true
Access to C:<HOMEPATH>\.espressif\tools\ninja\1.10.2: true
Access to C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3: true
Access to C:<HOMEPATH>\.espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64: true
Access to C:<HOMEPATH>\.espressif\tools\dfu-util\0.11\dfu-util-0.11-win64: true
Access to C:<HOMEPATH>\.espressif\tools\esp-rom-elfs\20230320: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) true
-------------------------------------------------------- Configurations has spaces -------------------------------------------------------------
Spaces in system environment Path true
Spaces in ESP-ADF Path (idf.espAdfPath) false
Spaces in ESP-IDF Path (idf.espIdfPath) false
Spaces in ESP-MDF Path (idf.espMdfPath) false
Spaces in ESP-Matter Path (idf.espMatterPath) false
Spaces in ESP-IDF Custom extra paths
[... cut ...]
Check Debug AdapterPython packages Python requirements are satisfied.
---------------------------------------------------- Visual Studio Code launch.json --------------------------------------------------------
No workspace folder is opened 
---------------------------------------------------- Visual Studio Code c_cpp_properties.json ----------------------------------------------
No workspace folder is opened 
--------------------------------------------------------------------------------------------------------------------------------------------

```

brianignacio5, 2023-11-27:
Based on the doctor command output and the log you shared the setup seems to work fine.

boris-gu, 2023-11-27:
I tried to build a simple LED flashing project and it works well.

But I'm not sure that everything will always go well. As it was written in the errors, I have no files  
```
esp\esp-idf\components\bt\host\nimble\nimble\porting\npl\riot\include\npl_sycfg.h  
esp\esp-idf\examples\build_system\cmake\idf_as_lib\build-esp32.sh  
esp\esp-idf\examples\build_system\cmake\idf_as_lib\build-esp32c2.sh  
esp\esp-idf\examples\build_system\cmake\idf_as_lib\build-esp32c3.sh  
esp\esp-idf\examples\build_system\cmake\idf_as_lib\build-esp32c6.sh  
esp\esp-idf\examples\build_system\cmake\idf_as_lib\build-esp32h2.sh  
esp\esp-idf\examples\build_system\cmake\idf_as_lib\build-esp32s2.sh  
esp\esp-idf\examples\build_system\cmake\idf_as_lib\build-esp32s3.sh  
esp\esp-idf\examples\build_system\cmake\idf_as_lib\run-esp32c2.sh  
esp\esp-idf\examples\build_system\cmake\idf_as_lib\run-esp32c3.sh  
esp\esp-idf\examples\build_system\cmake\idf_as_lib\run-esp32c6.sh  
esp\esp-idf\examples\build_system\cmake\idf_as_lib\run-esp32h2.sh  
esp\esp-idf\examples\build_system\cmake\idf_as_lib\run-esp32s2.sh  
esp\esp-idf\examples\build_system\cmake\idf_as_lib\run-esp32s3.sh  
```

I'm sorry, I do not know what these files are for or if they are needed at all. I only know that errors come out during installation

brianignacio5, 2023-11-27:
These kinds of errors seems to me that are related to directory write permission and git cloning issues:

> spawn C:\Windows\system32\cmd.exe ENOENT
> error: unable to create symlink porting/npl/riot/include/npl_syscfg/npl_sycfg.h: Permission denied
> error: Submodule 'components/bt/host/nimble/nimble' could [...]

Does your user has write permission to the esp-idf directory ?

boris-gu, 2023-11-27:
My Windows user has write access to the \esp and \.espressif folders. I also tried using command `git init` in these folders and everything went well

brianignacio5, 2023-11-27:
What happens if you run `C:\Users\Boris\.espressif\tools\idf-git\2.39.2\cmd\git.exe" reset --hard --recurse-submodule` in a cmd.exe terminal inside esp-idf directory ?

boris-gu, 2023-11-27:
[image]

boris-gu, 2023-11-27:
I searched a bit for a solution to the problem and this answer helped me (enabling developer mode). It might be worth adding this to your Troubleshooting

https://github.com/dtolnay/cxx/issues/357#issuecomment-1312161007

PR #1160 (2024-04-10): Docs/update cmds
## Description

This PR address a few different issues:

- Rename NVS Partition Editor and Partition Editor commands names.
- Add button to enable custom partition if disabled when running Partition Editor command. This will update sdkconfig in the back.
- Add `venv` validation in Setup Wizard.
- Pass current workspace folder to setup wizard save settings.

Fixes #1061
Fixes #1086

## Type of change

Please delete options that are not relevant.
- Bug fix (non-breaking change which fixes an issue)

## Steps to test this pull request

Provide a list of steps to test changes in t [... cut]
closed 2024-01-03 by github-actions[bot]