espressif/vscode-esp-idf-extension #1088: Issue with ESP debugger within VS code

MukulSosaleKeerthi-eaton, 2023-11-27:
### OS

Windows

### Operating System version

Windows 10

### Visual Studio Code version

1.84.2

### ESP-IDF version

5.1.1

### Python version

3.10

### Doctor command output

$ ESP-IDF Doctor command
bash: ESP-IDF: command not found

Using ESP IDF as a VS code plugin.

### Description

Trying to run the debugger with VS code environment. Setup the debugger with the below configuration. Getting the error message related to could not find files for given pattern (msg attached below).

{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        
        {
            "name": "ESP-IDF Debug: Launch",
            "type": "espidf",
            "request": "launch"
        }
    ]
}

### Debug Message

```plain
[Debug Adapter]
No Python at 'c:\Users\E07
[Debug Adapter]
02940\edge-rtos-v2-blil-demo\esp32_common\esp-idf\tools\tools\idf-python\3.8.7\python.exe'

[Stopped] : ESP-IDF Debug Adapter
INFO: Could not find files for the given pattern(s).
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2023-11-28:
To run the doctor command:

View -> Command Palette -> type `ESP-IDF: Doctor command` . Paste the output in clipboard in the GitHub issue.

https://github.com/espressif/vscode-esp-idf-extension#troubleshooting Take a look at the troubleshooting information. In your launch.json set `logLevel: 4` to see more output from information.

At simple view I would say that python path is incorrect c:\Users\E07
02940\edge-rtos-v2-blil-demo\esp32_common\esp-idf\tools\tools\idf-python\3.8.7\python.exe but need doctor command output.

MukulSosaleKeerthi-eaton, 2023-11-28:
Check ESP-IDF Python packages Python requirements from <HOMEPATH>/esp/esp-idf/tools/esp-idf-v4.4/requirements.txt are satisfied.
---------------------------------------------------- Check extension requirements.txt ------------------------------------------------------
Check Extension Python packages Python requirements from <HOMEPATH>/.vscode/extensions/espressif.esp-idf-extension-1.6.5/requirements.txt are satisfied.
---------------------------------------------------- Check ESP-IDF debug adapter requirements.txt ------------------------------------------
Check Debug AdapterPython packages Error
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at Unknown error in ESP-IDF doctor command
--------------------------------------------------------------------------------------------------------------------------------------------

MukulSosaleKeerthi-eaton, 2023-11-28:
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19045 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:<HOMEPATH>\Python310;C:<HOMEPATH>\Python310\Scripts;C:\Program Files (x86)\VMware\VMware Workstation\bin\;C:\Program Files\Common Files\Oracle\Java\javapath;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\1E\Client\Extensibility\NomadBranch;C:\Program Files\Docker\Docker\resources\bin;C:\ProgramData\DockerDesktop\version-bin;C:\Program Files\MATLAB\R2022a\runtime\win64;C:\Program Files\MATLAB\R2022a\bin;C:\Program Files\MATLAB\R2019b\runtime\win64;C:\Program Files\MATLAB\R2019b\bin;C:\ProgramData\chocolatey\bin;C:<HOMEPATH>\AppData\Roaming\Microsoft\Windows\Start Menu\Programs\Python 3.10;C:<HOMEPATH>\New folder\Lib\venv\scripts\nt;C:<HOMEPATH>\New folder\Scripts;C:\Windows;C:\Program Files\dotnet\;C:\Program Files\doxygen\bin;C:\Program Files\PuTTY\;C:<HOMEPATH>\AppData\Roaming\Python\Python311\Scripts;C:\Program Files\CMake\bin;C:<HOMEPATH>\Scripts\;C:<HOMEPATH>\;C:<HOMEPATH>\Python310;C:<HOMEPATH>\Python310\Scripts;C:<HOMEPATH>\Downloads\MinGW\bin;C:<HOMEPATH>\AppData\Local\Programs\Git\cmd;C:\Microsoft VS Code\bin;C:<HOMEPATH>\.dotnet\tools;C:<HOMEPATH>\New folder\Lib\venv\scripts\nt; 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.84.2 
Visual Studio Code language en 
Visual Studio Code shell C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.6.5 
Workspace folder c:<HOMEPATH>\edge-rtos-v2-blil-demo 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) c:<HOMEPATH>\edge-rtos-v2-blil-demo\esp32_common\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) c:<HOMEPATH>\edge-rtos-v2-blil-demo\esp32_common\esp-idf\tools\tools\xtensa-esp32-elf\esp-2021r1-8.4.0\xtensa-esp32-elf\bin;c:<HOMEPATH>\edge-rtos-v2-blil-demo\esp32_common\esp-idf\tools\tools\xtensa-esp32s2-elf\esp-2021r1-8.4.0\xtensa-esp32s2-elf\bin;c:<HOMEPATH>\edge-rtos-v2-blil-demo\esp32_common\esp-idf\tools\tools\xtensa-esp32s3-elf\esp-2021r1-8.4.0\xtensa-esp32s3-elf\bin;c:<HOMEPATH>\edge-rtos-v2-blil-demo\esp32_common\esp-idf\tools\tools\riscv32-esp-elf\esp-2021r1-8.4.0\riscv32-esp-elf\bin;c:<HOMEPATH>\edge-rtos-v2-blil-demo\esp32_common\esp-idf\tools\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;c:<HOMEPATH>\edge-rtos-v2-blil-demo\esp32_common\esp-idf\tools\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;c:<HOMEPATH>\edge-rtos-v2-blil-demo\esp32_common\esp-idf\tools\tools\cmake\3.16.4\bin;c:<HOMEPATH>\edge-rtos-v2-blil-demo\esp32_common\esp-idf\tools\tools\openocd-esp32\v0.10.0-esp32-20210401\openocd-esp32\bin;c:<HOMEPATH>\edge-rtos-v2-blil-demo\esp32_common\esp-idf\tools\tools\ninja\1.10.2;c:<HOMEPATH>\edge-rtos-v2-blil-demo\esp32_common\esp-idf\tools\tools\idf-exe\1.0.1;c:<HOMEPATH>\edge-rtos-v2-blil-demo\esp32_common\esp-idf\tools\tools\ccache\3.7;c:<HOMEPATH>\edge-rtos-v2-blil-demo\esp32_common\esp-idf\tools\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: c:<HOMEPATH>\edge-rtos-v2-blil-demo\esp32_common\esp-idf\tools\tools\openocd-esp32\v0.10.0-esp32-20210401/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 
Virtual env Python Path (idf.pythonBinPath) c:<HOMEPATH>\edge-rtos-v2-blil-demo\esp32_common\esp-idf\tools\python_env\idf4.3_py3.8_env\Scripts\python.exe
Serial port (idf.port) COM13
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) c:<HOMEPATH>\edge-rtos-v2-blil-demo\esp32_common\esp-idf\tools
Git Path (idf.gitPath) c:<HOMEPATH>\edge-rtos-v2-blil-demo\esp32_common\esp-idf\tools\tools\idf-git\2.39.2\cmd\git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to c:<HOMEPATH>\edge-rtos-v2-blil-demo\esp32_common\esp-idf\tools\tools\xtensa-esp32-elf\esp-2021r1-8.4.0\xtensa-esp32-elf\bin: true
Access to c:<HOMEPATH>\edge-rtos-v2-blil-demo\esp32_common\esp-idf\tools\tools\xtensa-esp32s2-elf\esp-2021r1-8.4.0\xtensa-esp32s2-elf\bin: true
Access to c:<HOMEPATH>\edge-rtos-v2-blil-demo\esp32_common\esp-idf\tools\tools\xtensa-esp32s3-elf\esp-2021r1-8.4.0\xtensa-esp32s3-elf\bin: true
Access to c:<HOMEPATH>\edge-rtos-v2-blil-demo\esp32_common\esp-idf\tools\tools\riscv32-esp-elf\esp-2021r1-8.4.0\riscv32-esp-elf\bin: true
Access to c:<HOMEPATH>\edge-rtos-v2-blil-demo\esp32_common\esp-idf\tools\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin: true
[... cut ...]
---------------------------------------------------- Check ESP-IDF debug adapter requirements.txt ------------------------------------------
Check Debug AdapterPython packages undefined
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at Unknown error in ESP-IDF doctor command
--------------------------------------------------------------------------------------------------------------------------------------------

brianignacio5, 2023-12-07:
What about the extension log ? 

Windows: %USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log
Linux & MacOSX: $HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log

Seems there is some issue in PATH maybe because checking git and other executable versions doesn't seems to be working also python packages.

MukulSosaleKeerthi-eaton, 2023-12-07:
The issue was fixed and had to do with the ESP IDF version 4.4.1 which I managed to install and that fixed the issue.

brianignacio5, 2023-12-07:
Oh I see. Closing this issue then.

closed 2023-12-07 by brianignacio5