espressif/vscode-esp-idf-extension #1091: Unable to start OpenOCD server to set up Debug Adapter

ionlabdev, 2023-12-08:
### OS

Windows

### Operating System version

macOS 14.1.1

### Visual Studio Code version

1.85.0

### ESP-IDF version

1.6.5

### Python version

3.11.2

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS darwin arm64 23.1.0 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 /usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/opt/homebrew/bin:/opt/homebrew/bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.85.0 
Visual Studio Code language en 
Visual Studio Code shell /bin/zsh 
ESP-IDF Extension version 1.6.5 
Workspace folder <HOMEPATH>/development/blink 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) <HOMEPATH>/development/esp/esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) <HOMEPATH>/development/.espressif/tools/xtensa-esp-elf-gdb/12.1_20221002/xtensa-esp-elf-gdb/bin:<HOMEPATH>/development/.espressif/tools/riscv32-esp-elf-gdb/12.1_20221002/riscv32-esp-elf-gdb/bin:<HOMEPATH>/development/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin:<HOMEPATH>/development/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin:<HOMEPATH>/development/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin:<HOMEPATH>/development/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin:<HOMEPATH>/development/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:<HOMEPATH>/development/.espressif/tools/cmake/3.24.0/CMake.app/Contents/bin:<HOMEPATH>/development/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin:<HOMEPATH>/development/.espressif/tools/ninja/1.10.2:<HOMEPATH>/development/.espressif/tools/esp-rom-elfs/20230320
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: <HOMEPATH>/development/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/share/openocd/scripts
    ESP_ROM_ELF_DIR: <HOMEPATH>/development/.espressif/tools/esp-rom-elfs/20230320/
Virtual env Python Path (idf.pythonBinPath) <HOMEPATH>/development/.espressif/python_env/idf5.1_py3.11_env/bin/python
Serial port (idf.port) /dev/cu.usbserial-110
OpenOCD Configs (idf.openOcdConfigs) board/esp32s2-bridge.cfg
ESP-IDF Tools Path (idf.toolsPath) <HOMEPATH>/development/.espressif
Git Path (idf.gitPath) git
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to <HOMEPATH>/development/.espressif/tools/xtensa-esp-elf-gdb/12.1_20221002/xtensa-esp-elf-gdb/bin: true
Access to <HOMEPATH>/development/.espressif/tools/riscv32-esp-elf-gdb/12.1_20221002/riscv32-esp-elf-gdb/bin: true
Access to <HOMEPATH>/development/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin: true
Access to <HOMEPATH>/development/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin: true
Access to <HOMEPATH>/development/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin: true
Access to <HOMEPATH>/development/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin: true
Access to <HOMEPATH>/development/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin: true
Access to <HOMEPATH>/development/.espressif/tools/cmake/3.24.0/CMake.app/Contents/bin: true
Access to <HOMEPATH>/development/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin: true
Access to <HOMEPATH>/development/.espressif/tools/ninja/1.10.2: true
Access to <HOMEPATH>/development/.espressif/tools/esp-rom-elfs/20230320: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) true
[... 77 lines cut ...]
mainThreadExtensionService.ts:81 [espressif.esp-idf-extension]OpenOCD server failed to start because of Error: 477 14 esp_usb_jtag.c:646 esp_usb_jtag_init(): esp_usb_jtag: could not find or open device!
$onExtensionRuntimeError @ mainThreadExtensionService.ts:81
mainThreadExtensionService.ts:82 Error: OpenOCD server failed to start because of Error: 477 14 esp_usb_jtag.c:646 esp_usb_jtag_init(): esp_usb_jtag: could not find or open device!
	at Socket.<anonymous> (/dist/extension.js:2:1226582)
	at Socket.emit (node:events:513:28)
	at addChunk (node:internal/streams/readable:324:12)
	at readableAddChunk (node:internal/streams/readable:297:9)
	at Readable.push (node:internal/streams/readable:234:10)
	at Pipe.onStreamRead (node:internal/stream_base_commons:190:23)
$onExtensionRuntimeError @ mainThreadExtensionService.ts:82
mainThreadExtensionService.ts:81 [espressif.esp-idf-extension]OpenOCD server failed to start because of Error: 
$onExtensionRuntimeError @ mainThreadExtensionService.ts:81
mainThreadExtensionService.ts:82 Error: OpenOCD server failed to start because of Error: 
	at Socket.<anonymous> (/dist/extension.js:2:1226582)
	at Socket.emit (node:events:513:28)
	at addChunk (node:internal/streams/readable:324:12)
	at readableAddChunk (node:internal/streams/readable:297:9)
	at Readable.push (node:internal/streams/readable:234:10)
	at Pipe.onStreamRead (node:internal/stream_base_commons:190:23)
$onExtensionRuntimeError @ mainThreadExtensionService.ts:82
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2023-12-11:
How are you connecting your board to JTAG ? An external JTAG ?

Based on the output, it seems that openOCD can't find your device. Make sure the connection is right and the openOCD board selected is the right one. If using external jtag you can try with `["interface/ftdi/esp32_devkitj_v1.cfg", "target/esp32s2.cfg"]`

Also look at openOCD troubleshooting here: https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ

ionlabdev, 2023-12-12:
> How are you connecting your board to JTAG ? An external JTAG ?
>
> Based on the output, it seems that openOCD can't find your device. Make sure the connection is right and the openOCD board selected is [...]

Thank you for your response. I am using the internal JTAG interface of the ESP32s2.

- [x]  I checked that in the "ESP-IDF: Select OpenOCD Board Configuraiton" and device target are correct.
- [x]  I checked the openOCD troubleshooting guide for possible solutions. The cases mentioned in the troubleshooting guide do either not apply to my case or there is no troubleshooting solution for macOS.

Whats most puzzling to me is that previously, while following the [basic turtorial](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/basic_use.md) for the blink example project, I was able to start the openOCD server. Immediatly afterwards, I tried setting up the debug adapter. Since then, I cannot start up the openOCD server in any way. I did not manually change the settings between those two attempts to start up the openOCD server.

Do you have any further suggestions of what I should do to solve this problem?

brianignacio5, 2023-12-14:
If you were able to start openOCD server before and now you can't. It might be related to some configuration settings that are not correctly configured. `idf.openOcdConfigs` seems to me a good start.  ESP-IDF: Select OpenOCD Board Configuration to change this setting, and check the OpenOCD tagged output in menu View -> Output -> ESP-IDF

brianignacio5, 2023-12-14:
What is the hardware board you are using ? Maybe try using Kaluga openOCD config files ?

ionlabdev, 2023-12-14:
> What is the hardware board you are using ? Maybe try using Kaluga openOCD config files ?

I am using the ESP32-S2-DevKitM-1 board, so I would guess using the Kaluga config files won't work with this board. 

> If you were able to start openOCD server before and now you can't. It might be related to some configuration settings that are not correctly configured. `idf.openOcdConfigs` seems to me a good [...]

I have looked at the idf.openOcdConfigs. The value that I have set for it is `"idf.openOcdConfigs": ["board/esp32-bridge.cfg"]` as the ESP32-S2-DevKitM-1 board has an USB bridge. I have checked that the device target matches the openOCD board configuration settings.

brianignacio5, 2023-12-15:
Can you try this setting:

```
"idf.openOcdConfigs": ["interface/ftdi/esp32_devkitj_v1.cfg", "target/esp32s2.cfg"]
```

ionlabdev, 2023-12-15:
> ["interface/ftdi/esp32_devkitj_v1.cfg", "target/esp32s2.cfg"]

I changed this manually in the settings, but unfortunatly openOCD still throws the same errors.

brianignacio5, 2023-12-18:
I think you should try to refer to https://github.com/espressif/openocd-esp32 repository for your device.

From the extension side I don't see the problem. Maybe one of the maintainers can help you further.

closed 2023-12-18 by ionlabdev