espressif/vscode-esp-idf-extension #1092: installed ESP-IDF tools are not detected

dyarkovoy, 2023-12-10:
### OS

Windows

### Operating System version

Microsoft Windows [Version 10.0.22631.2715] 

### Visual Studio Code version

1.85.0

### ESP-IDF version

v5.1.2,master

### Python version

3.11.3

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.22631 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Program Files\Common Files\Oracle\Java\javapath;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\PuTTY\;C:\Program Files (x86)\Microsoft Emulator Manager\1.0\;C:\Program Files\CMake\bin;C:\Program Files\dotnet\;C:\Program Files\Git\cmd;C:\Program Files\nodejs\;C:\Program Files (x86)\Bitvise SSH Client;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin;C:\Program Files (x86)\Nmap;C:<HOMEPATH>\AppData\Local\JetBrains\Toolbox\scripts;C:<HOMEPATH>\AppData\Local\Programs\Python\Python311;C:<HOMEPATH>\AppData\Local\gitkraken\bin;C:<HOMEPATH>\AppData\Local\GitHubDesktop\bin;c:\work\tools;C:<HOMEPATH>\AppData\Roaming\npm; 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.85.0 
Visual Studio Code language en 
Visual Studio Code shell C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.6.5 
Workspace folder c:\work\clients\alm\alm-x2 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:\work\esp32\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) c:\work\esp32\.espressif\tools\xtensa-esp-elf-gdb\12.1_20231023\xtensa-esp-elf-gdb\bin;c:\work\esp32\.espressif\tools\riscv32-esp-elf-gdb\12.1_20231023\riscv32-esp-elf-gdb\bin;c:\work\esp32\.espressif\tools\xtensa-esp-elf\esp-13.2.0_20230928\xtensa-esp-elf\bin;c:\work\esp32\.espressif\tools\riscv32-esp-elf\esp-13.2.0_20230928\riscv32-esp-elf\bin;c:\work\esp32\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;c:\work\esp32\.espressif\tools\cmake\3.24.0\bin;c:\work\esp32\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin;c:\work\esp32\.espressif\tools\ninja\1.11.1;c:\work\esp32\.espressif\tools\idf-exe\1.0.3;c:\work\esp32\.espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64;c:\work\esp32\.espressif\tools\dfu-util\0.11\dfu-util-0.11-win64;c:\work\esp32\.espressif\tools\esp-rom-elfs\20230320
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: c:\work\esp32\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230921/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
    ESP_ROM_ELF_DIR: c:\work\esp32\.espressif\tools\esp-rom-elfs\20230320/
Virtual env Python Path (idf.pythonBinPath) C:\work\esp32\.espressif\python_env\idf5.3_py3.11_env\Scripts\python.exe
Serial port (idf.port) COM4
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,target/esp32s3.cfg
ESP-IDF Tools Path (idf.toolsPath) c:\work\esp32\.espressif
Git Path (idf.gitPath) c:\work\esp32\.espressif\tools\idf-git\2.39.2\cmd\git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to c:\work\esp32\.espressif\tools\xtensa-esp-elf-gdb\12.1_20231023\xtensa-esp-elf-gdb\bin: true
Access to c:\work\esp32\.espressif\tools\riscv32-esp-elf-gdb\12.1_20231023\riscv32-esp-elf-gdb\bin: true
Access to c:\work\esp32\.espressif\tools\xtensa-esp-elf\esp-13.2.0_20230928\xtensa-esp-elf\bin: true
Access to c:\work\esp32\.espressif\tools\riscv32-esp-elf\esp-13.2.0_20230928\riscv32-esp-elf\bin: true
Access to c:\work\esp32\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin: true
Access to c:\work\esp32\.espressif\tools\cmake\3.24.0\bin: true
Access to c:\work\esp32\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin: true
Access to c:\work\esp32\.espressif\tools\ninja\1.11.1: true
Access to c:\work\esp32\.espressif\tools\idf-exe\1.0.3: true
Access to c:\work\esp32\.espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64: true
Access to c:\work\esp32\.espressif\tools\dfu-util\0.11\dfu-util-0.11-win64: true
Access to c:\work\esp32\.espressif\tools\esp-rom-elfs\20230320: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH undefined
[... 129 lines cut ...]
[image]
[image]

### Debug Message

```plain
Not relevant
```

### Other Steps to Reproduce

I tracked it down to this line: https://github.com/dyarkovoy/vscode-esp-idf-extension/blob/f549deda8116673b1564d0f2e50cd77904344cca/src/idfToolsManager.ts#L207

My `process.env` PATH variable has all letters in upper case. For some reason, this line assumes that on Windows PATH variable is "Path", which is not true, at least not on my Windows. Even though `process.env` keys are case-insensitive, the actual glitch occurs on this line: https://github.com/dyarkovoy/vscode-esp-idf-extension/blob/f549deda8116673b1564d0f2e50cd77904344cca/src/idfToolsManager.ts#L217

where it attempts to find "Path" variable in the `modifiedEnv`, which is a regular object with case-sensitive keys.

The proposed solution is to extract actual `PATH` key name from the `process.env` using this simple expression:
`Object.keys(process.env).find(k=>k.toUpperCase()=="PATH")`, rather than assuming that PATH is always "Path" on Windows.

I tested the proposed solution and it solves this issue. I will submit PR shortly.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

dyarkovoy, 2023-12-10:
I believe this should also fix issue #1066

closed 2023-12-12 by brianignacio5

brianignacio5, 2024-01-11:
Could you check the output in the extension log  @txf- ?

txf-, 2024-01-11:
> Could you check the output in the extension log @txf- ?
How do I do that? Is that the Extension Host log?

brianignacio5, 2024-01-11:
> https://github.com/espressif/vscode-esp-idf-extension#troubleshooting

Check log file which can be obtained from:
Windows: %USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log
Linux & MacOSX: $HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log

txf-, 2024-01-11:
right I'm attaching it to this comment 
[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/13903335/esp_idf_vsc_ext.log)

brianignacio5, 2024-01-11:
You mention it wouldn't extract the tools in the directory ? When you are doing the install, what is the message in the menu View -> Output and choose ESP-IDF from drop down.

Also how are you installing the tools ?

txf-, 2024-01-11:
Here it is:

```
idf_versions.txt has (1) KB
idf_versions.txt progress: 100.00% (0.10 / 0.10) KB
/usr/bin/cmake

Downloading GDB for Xtensa
Found xtensa-esp-elf-gdb in /home/tiago/.espressif5x/dist
Downloading GDB for RISC-V
Found riscv32-esp-elf-gdb in /home/tiago/.espressif5x/dist
Downloading Toolchain for Xtensa (ESP32) based on GCC
Found xtensa-esp32-elf in /home/tiago/.espressif5x/dist
Downloading Toolchain for Xtensa (ESP32-S2) based on GCC
Found xtensa-esp32s2-elf in /home/tiago/.espressif5x/dist
Downloading Toolchain for Xtensa (ESP32-S3) based on GCC
Found xtensa-esp32s3-elf in /home/tiago/.espressif5x/dist
Downloading Toolchain for 32-bit RISC-V based on GCC
Found riscv32-esp-elf in /home/tiago/.espressif5x/dist
Downloading Toolchain for ESP32 ULP coprocessor
Found esp32ulp-elf in /home/tiago/.espressif5x/dist
Downloading OpenOCD for ESP32
Found openocd-esp32 in /home/tiago/.espressif5x/dist
Downloading Ninja build system
Found ninja in /home/tiago/.espressif5x/dist
Downloading ESP ROM ELFs
Found esp-rom-elfs in /home/tiago/.espressif5x/dist

Using existing GDB for Xtensa in /home/tiago/.espressif5x/tools/xtensa-esp-elf-gdb/12.1_20231023
Using existing GDB for RISC-V in /home/tiago/.espressif5x/tools/riscv32-esp-elf-gdb/12.1_20231023
/bin/sh: 1: xtensa-esp32-elf-gcc: not found
Command failed: xtensa-esp32-elf-gcc --version
/bin/sh: 1: xtensa-esp32-elf-gcc: not found
Error checking xtensa-esp32-elf version
Error: Command failed: xtensa-esp32-elf-gcc --version
/bin/sh: 1: xtensa-esp32-elf-gcc: not found

Installing package Toolchain for Xtensa (ESP32) based on GCC
Current system platform: linux-amd64
Installing tools: 

/bin/sh: 1: xtensa-esp32s2-elf-gcc: not found
Command failed: xtensa-esp32s2-elf-gcc --version
/bin/sh: 1: xtensa-esp32s2-elf-gcc: not found
Error checking xtensa-esp32s2-elf version
Error: Command failed: xtensa-esp32s2-elf-gcc --version
/bin/sh: 1: xtensa-esp32s2-elf-gcc: not found

Installing package Toolchain for Xtensa (ESP32-S2) based on GCC
Current system platform: linux-amd64
Installing tools: 

/bin/sh: 1: xtensa-esp32s3-elf-gcc: not found
Command failed: xtensa-esp32s3-elf-gcc --version
/bin/sh: 1: xtensa-esp32s3-elf-gcc: not found
Error checking xtensa-esp32s3-elf version
Error: Command failed: xtensa-esp32s3-elf-gcc --version
/bin/sh: 1: xtensa-esp32s3-elf-gcc: not found

Installing package Toolchain for Xtensa (ESP32-S3) based on GCC
Current system platform: linux-amd64
Installing tools: 

/bin/sh: 1: riscv32-esp-elf-gcc: not found
Command failed: riscv32-esp-elf-gcc --version
/bin/sh: 1: riscv32-esp-elf-gcc: not found
Error checking riscv32-esp-elf version
Error: Command failed: riscv32-esp-elf-gcc --version
/bin/sh: 1: riscv32-esp-elf-gcc: not found

[... 86 lines cut ...]
Error: Command failed: riscv32-esp-elf-gcc --version
/bin/sh: 1: riscv32-esp-elf-gcc: not found

/bin/sh: 1: xtensa-esp32s3-elf-gcc: not found
Command failed: xtensa-esp32s3-elf-gcc --version
/bin/sh: 1: xtensa-esp32s3-elf-gcc: not found
Error checking xtensa-esp32s3-elf version
Error: Command failed: xtensa-esp32s3-elf-gcc --version
/bin/sh: 1: xtensa-esp32s3-elf-gcc: not found

/bin/sh: 1: xtensa-esp32s2-elf-gcc: not found
Command failed: xtensa-esp32s2-elf-gcc --version
/bin/sh: 1: xtensa-esp32s2-elf-gcc: not found
Error checking xtensa-esp32s2-elf version
Error: Command failed: xtensa-esp32s2-elf-gcc --version
/bin/sh: 1: xtensa-esp32s2-elf-gcc: not found

/bin/sh: 1: xtensa-esp32-elf-gcc: not found
Command failed: xtensa-esp32-elf-gcc --version
/bin/sh: 1: xtensa-esp32-elf-gcc: not found
Error checking xtensa-esp32-elf version
Error: Command failed: xtensa-esp32-elf-gcc --version
/bin/sh: 1: xtensa-esp32-elf-gcc: not found

```

mahboobkarimian, 2024-01-25:
I have a problem in Linux with vscode 1.85.2. I also tested with the nightly build of the extension, the same problem.
```
Python requirements has been installed
/bin/sh: 1: riscv32-esp-elf-gcc: not found
Command failed: riscv32-esp-elf-gcc --version
/bin/sh: 1: riscv32-esp-elf-gcc: not found
Error checking riscv32-esp-elf version
Error: Command failed: riscv32-esp-elf-gcc --version
/bin/sh: 1: riscv32-esp-elf-gcc: not found

/bin/sh: 1: xtensa-esp-elf-gcc: not found
Command failed: xtensa-esp-elf-gcc --version
/bin/sh: 1: xtensa-esp-elf-gcc: not found
Error checking xtensa-esp-elf version
Error: Command failed: xtensa-esp-elf-gcc --version
/bin/sh: 1: xtensa-esp-elf-gcc: not found
```

brianignacio5, 2024-01-25:
I've made a fix related to this @mahboobkarimian  @txf- in [here](https://github.com/espressif/vscode-esp-idf-extension/pull/1118)

You can try installing the extension [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/1194391402.zip) from this PR to test it out

PR #1093 (2023-12-12): Fix for issue #1092 (VSC-1257)
## Description

Please include a summary of the change and which issue is fixed.

Fixes #1092 

## Type of change

Please delete options that are not relevant.
- Bug fix (non-breaking change which fixes an issue)

## Steps to test this pull request

Configure extension with the existing ESP-IDF + installed tools on Windows 11, there should be no errors complaining that the tools are not found at the specified correct paths

## How has this been tested?

This has been tested on my VSCode with the existing ESP-IDF + installed tools on Windows 11

## Dependent compo [... cut]