espressif/vscode-esp-idf-extension #1098: Error: "Make sure you have the esptool.py installed and set in $PATH with proper permission" and more

Peugot206, 2023-12-21:
### OS

Windows

### Operating System version

Windows 10

### Visual Studio Code version

1.85.1

### ESP-IDF version

5.1.2

### Python version

3.11.2

### Doctor command output

```
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19045 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:<HOMEPATH>\AppData\Local\Programs\Python\Python39;C:\Program Files\Git\cmd;C:\Windows\system32;C:\Espressif;C:<HOMEPATH>\AppData\Local\Programs\Python\Python39\Scripts;C:<HOMEPATH>\esp\esp-idf;C:<HOMEPATH>\esp\esp-idf\components\esptool_py\esptool;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\AppData\Local\JetBrains\Toolbox\scripts;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin;C:<HOMEPATH>\.dotnet\tools;"C:<HOMEPATH>\AppData\Local\Programs\Python\Python39;C:<HOMEPATH>\AppData\Local\Programs\Python\Python39\Scripts"; 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.85.1 
Visual Studio Code language en 
Visual Studio Code shell C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.6.5 
Workspace folder c:\ESP_IDF_PROJECTS\vipbox_iv\Src 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) ${env:IDF_PATH}
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) C:\Espressif\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin;C:\Espressif\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin;C:\Espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin;C:\Espressif\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin;C:\Espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin;C:\Espressif\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin;C:\Espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:\Espressif\tools\cmake\3.24.0\bin;C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin;C:\Espressif\tools\ninja\1.10.2;C:\Espressif\tools\idf-exe\1.0.3;C:\Espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64;C:\Espressif\tools\dfu-util\0.11\dfu-util-0.11-win64;C:\Espressif\tools\esp-rom-elfs\20230320
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230921/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
    ESP_ROM_ELF_DIR: C:\Espressif\tools\esp-rom-elfs\20230320/
Virtual env Python Path (idf.pythonBinPath) C:\Espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe
Serial port (idf.port) COM15
OpenOCD Configs (idf.openOcdConfigs) board/esp32-wrover-kit-1.8v.cfg
ESP-IDF Tools Path (idf.toolsPath) C:\Espressif
Git Path (idf.gitPath) C:\Espressif\tools\idf-git\2.39.2\cmd\git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) false
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to C:\Espressif\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin: true
Access to C:\Espressif\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin: true
Access to C:\Espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin: true
Access to C:\Espressif\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin: true
Access to C:\Espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin: true
Access to C:\Espressif\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin: true
Access to C:\Espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin: true
Access to C:\Espressif\tools\cmake\3.24.0\bin: true
Access to C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin: true
Access to C:\Espressif\tools\ninja\1.10.2: true
Access to C:\Espressif\tools\idf-exe\1.0.3: true
Access to C:\Espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64: true
Access to C:\Espressif\tools\dfu-util\0.11\dfu-util-0.11-win64: true
[... 145 lines cut ...]
```

### Description

I've been trying to get ESP IDF running in VSCode for a while now and keep running into issues.

First I installed the Windows .exe installer for ESP-IDF and used this existing installation in VSCode. I got so many problems here that I decided to just install ESP-IDF via the VSCode ESP-IDF extension. This worked much better, however I run into some issues:

- Commands like menuconfig, open terminal, setting target **won't do anything**. On the example projects these work fine.
- When I try to flash the device I get the error: 
```
[Flash]
Make sure you have the esptool.py installed and set in $PATH with proper permission
```
What could be causing this? Flashing also works fine on the example projects..

### Debug Message

```plain
None.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2023-12-21:
Can you provide the extension log ? It seems that the extension was not properly configured in the python packages part of the setup.

Windows: `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log`
Linux & MacOSX: `$HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log`

What steps do you perform to configure the extension ? Did you see any error in the ESP-IDF output ?

Check our [troubleshooting information](https://github.com/espressif/vscode-esp-idf-extension#troubleshooting)

Peugot206, 2023-12-21:
Thanks for your quick response @brianignacio5 !

Here is the extension log (with my username replaced to `MY_USERNAME`). 

[esp_idf_vsc_ext_nameEdited.log](https://github.com/espressif/vscode-esp-idf-extension/files/13742089/esp_idf_vsc_ext_nameEdited.log).

I installed ESP-IDF through the extension in VS Code _(even though I first tried with the ESP IDF .exe installer, but later uninstalled it due to the many issues)_. I then ran the `ESP-IDF: Configure` extension and used `EXPRESS`. It all succeeded without problems. The examples all work fine too, I can compile, flash, monitor, menuconfig, etc it. 
On this project however I can't. Stuff like set target device, menuconfig and monitor does not work. When it's at the end of building I got an error that it couldn't find `idf_size.py` then it can't find `idf_monitor.py` etc. So something is happening here that it can't find the path `C:\Users\MY_USERNAME\esp\esp-idf\tools` or something. But the examples somehow can and have no issues at all. FYI I set up the extension globally, not for a workspace.

brianignacio5, 2023-12-22:
I can observe that many paths in this project are broken like: `components\\esptool_py\\esptool\\esptool.py` and `"tools\\idf_monitor.py is not defined`

Looking at the doctor command you have idf.espIdfPath set to `${env:IDF_PATH}` which yields `Access to ESP-IDF Path (idf.espIdfPath) false` so maybe need to review your environment IDF_PATH and check that is doesn't have any spaces in it ?

Peugot206, 2023-12-22:
Thanks for the reply.

My Esp Idf Path Win is configured as following: `C:\Users\my.username\esp\esp-idf`, no spaces in there..

The Esp Idf Path is left default to `${env:IDF_PATH}`, but I assumes it takes the configured Win path.

brianignacio5, 2023-12-22:
Do you have any idf.* settings in `c:\ESP_IDF_PROJECTS\vipbox_iv\Src\.vscode\settings.json` or workspace settings?

I'm not sure if the . is breaking the path but it could be. the ESP-IDF is not found by the extension.

Peugot206, 2023-12-22:
Yes:

```
{
    "idf.espIdfPathWin": "${env:IDF_PATH}",
    "files.associations": {
        "bus.h": "c",
        "constants.h": "c",
        "monitor.h": "c"
    },
    "idf.adapterTargetName": "esp32",
    "idf.portWin": "COM15",
    "idf.flashType": "UART",
    "idf.openOcdConfigs": [
        "board/esp32-wrover-kit-1.8v.cfg"
    ],
}
```

brianignacio5, 2023-12-22:
This one is overriding your User settings  in this project. Update it with `C:\Users\my.username\esp\esp-idf` and try again

Peugot206, 2023-12-22:
Thank you SO much! This seems to have solved it.

closed 2023-12-22 by Peugot206