espressif/vscode-esp-idf-extension #1100: Error: libusb_open() failed with LIBUSB_ERROR_ACCESS in Docker

ag00se, 2023-12-21:
### OS

Windows

### Operating System version

Windows 11 Build 22621

### Visual Studio Code version

1.85.1

### ESP-IDF version

5.1.2

### Python version

3.8.10

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS linux x64 5.15.133.1-microsoft-standard-WSL2 
System environment variable IDF_PYTHON_ENV_PATH 
 /opt/esp/python_env/idf5.1_py3.8_env 
System environment variable PATH 
 /vscode/vscode-server/bin/linux-x64/0ee08df0cf4527e40edc9aa28f4b5bd38bbff2b2/bin/remote-cli:/opt/esp/idf/components/espcoredump:/opt/esp/idf/components/partition_table:/opt/esp/idf/components/app_update:/opt/esp/tools/xtensa-esp-elf-gdb/12.1_20221002/xtensa-esp-elf-gdb/bin:/opt/esp/tools/riscv32-esp-elf-gdb/12.1_20221002/riscv32-esp-elf-gdb/bin:/opt/esp/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin:/opt/esp/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin:/opt/esp/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin:/opt/esp/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin:/opt/esp/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/opt/esp/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin:/opt/esp/python_env/idf5.1_py3.8_env/bin:/opt/esp/idf/tools:/opt/qemu/bin:/opt/qemu/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.85.1 
Visual Studio Code language en 
Visual Studio Code shell /usr/bin/bash 
ESP-IDF Extension version 1.6.5 
Workspace folder /workspaces/blink 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) /opt/esp/idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) 
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: d:\Tools\esp_idf_tools\tools\openocd-esp32\v0.12.0-esp32-20230921/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
    ESP_ROM_ELF_DIR: d:\Tools\esp_idf_tools\tools\esp-rom-elfs\20230320/
Virtual env Python Path (idf.pythonBinPath) /opt/esp/python_env/idf5.1_py3.8_env/bin/python
Serial port (idf.port) /dev/ttyACM0
OpenOCD Configs (idf.openOcdConfigs) board/esp32c6-builtin.cfg
ESP-IDF Tools Path (idf.toolsPath) /opt/esp
Git Path (idf.gitPath) /usr/bin/git
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH true
Access to Ninja in environment PATH true
Access to ESP-IDF Tools Path (idf.toolsPath) true
-------------------------------------------------------- Configurations has spaces -------------------------------------------------------------
Spaces in system environment Path false
Spaces in ESP-ADF Path (idf.espAdfPath) false
Spaces in ESP-IDF Path (idf.espIdfPath) false
Spaces in ESP-MDF Path (idf.espMdfPath) false
Spaces in ESP-Matter Path (idf.espMatterPath) false
Spaces in ESP-IDF Custom extra paths
Spaces in Virtual env Python Path (idf.pythonBinPath) false
Spaces in ESP-IDF Tools Path (idf.toolsPath) false
----------------------------------------------------------- Executables Versions -----------------------------------------------------------
[... 145 lines cut ...]
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000

[OpenOCD]
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections

[OpenOCD]
❌ Error: libusb_open() failed with LIBUSB_ERROR_ACCESS

[OpenOCD]
❌ Error: esp_usb_jtag: could not find or open device!
/opt/esp/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin/../share/openocd/scripts/target/esp_common.cfg:9: Error: 
at file "/opt/esp/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin/../share/openocd/scripts/target/esp_common.cfg", line 9

[OpenOCD]
[Stopped] : OpenOCD Server
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2023-12-22:
For the python part, just update `idf.pythonBinPath` in your .vscode/.devcontainer.json to match your docker file esp-idf python version.

About openOCD, have you configured the USB drivers ? 

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/configure-ft2232h-jtag.html#windows

ag00se, 2023-12-22:
The debugging works in the Windows environment so I assumend the drivers are ok. Do i need to manually add the udev rules in the docker container or adjust any drivers for USB debugging? I assumed, reading the tutorial, that the container is prepared and no configuration is needed.

Thanks for your reply

ag00se, 2024-01-02:
I am still facing this issue:
[image]

When connecting as root to the running container (started using vs code devcontainer) I can start the ```openocd``` server. (Marked as 1 in the image)

I adjusted the default ```Dockerfile``` to install ```udev```, copy the default rules to ```/etc/udev/rules.d``` and restart the ```udev``` service. (Marked as 2 in the image)

When I switch to the ```esp``` user I can`t start the ```openocd``` server. (Marked 3 in the image)
The ```esp``` user is added to the ```dialout``` group by the default ```Dockerfile``` to allow access to the ```tty``` devices. (Marked as 4 in the image)

Is anyone facing a similar problem, or has any idea what I am missing?

ag00se, 2024-01-07:
I am one step further to getting this to work. I can't work using ```udev``` inside the container. An error message is shown that ```udev``` is not supported in containers.

I can start ```openocd``` when I add the user ```esp``` to the ```root``` group and flash onto the device using ```jtag```. This is a suitable solution for me for now.
With this problem solved, now a python module is missing:
```
[Debug Adapter]
    from debug_adapter import cli
  File "/home/esp/.vscode-server/extensions/espressif.esp-idf-extension-1.6.5/esp_debug_adapter/debug_adapter/__init__.py", line 25, in <module>
    from .debug_adapter import DebugAdapter, A2VSC_READY2CONNECT_STRING, A2VSC_STOPPED_STRING, A2VSC_STARTED_STRING
  File "/home/esp/.vscode-server/extensions/espressif.esp-idf-extension-1.6.5/esp_debug_adapter/debug_adapter/debug_adapter.py", line 26, in <module>
    import esp_debug_backend as dbg
ModuleNotFoundError: No module named 'esp_debug_backend'
```

I saw that one should just execute ```ESP-IDF: Install ESP-IDF Python Packages``` (as mentioned [here](https://github.com/espressif/vscode-esp-idf-extension/issues/1006) but this results in the following error inside the container:
```
Installing ESP-IDF extension python packages in /opt/esp/python_env/idf5.3_py3.10_env/bin/python ...

Collecting gcovr (from -r /home/esp/.vscode-server/extensions/espressif.esp-idf-extension-1.6.5/requirements.txt (line 1))
  Downloading gcovr-6.0-py2.py3-none-any.whl (94 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 94.1/94.1 kB 2.2 MB/s eta 0:00:00
Collecting websocket_client (from -r /home/esp/.vscode-server/extensions/espressif.esp-idf-extension-1.6.5/requirements.txt (line 2))
  Downloading websocket_client-1.7.0-py3-none-any.whl.metadata (7.9 kB)
Collecting jinja2 (from gcovr->-r /home/esp/.vscode-server/extensions/espressif.esp-idf-extension-1.6.5/requirements.txt (line 1))
  Downloading Jinja2-3.0.3-py3-none-any.whl (133 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 133.6/133.6 kB 3.4 MB/s eta 0:00:00
Collecting lxml (from gcovr->-r /home/esp/.vscode-server/extensions/espressif.esp-idf-extension-1.6.5/requirements.txt (line 1))
  Downloading lxml-5.0.1-cp310-cp310-manylinux_2_28_x86_64.whl.metadata (3.6 kB)
Requirement already satisfied: pygments in ./python_env/idf5.3_py3.10_env/lib/python3.10/site-packages (from gcovr->-r /home/esp/.vscode-server/extensions/espressif.esp-idf-extension-1.6.5/requirements.txt (line 1)) (2.17.2)
Collecting MarkupSafe>=2.0 (from jinja2->gcovr->-r /home/esp/.vscode-server/extensions/espressif.esp-idf-extension-1.6.5/requirements.txt (line 1))
  Downloading MarkupSafe-2.1.3-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (3.0 kB)
Downloading websocket_client-1.7.0-py3-none-any.whl (58 kB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 58.5/58.5 kB 3.2 MB/s eta 0:00:00
Downloading lxml-5.0.1-cp310-cp310-manylinux_2_28_x86_64.whl (8.0 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 8.0/8.0 MB 5.0 MB/s eta 0:00:00
Downloading MarkupSafe-2.1.3-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (25 kB)
Installing collected packages: websocket_client, MarkupSafe, lxml, jinja2, gcovr
ERROR: Could not install packages due to an OSError: [Errno 13] Permission denied: '/opt/esp/python_env/idf5.3_py3.10_env/lib/python3.10/site-packages/websocket'
Check the permissions.

Command failed: "/opt/esp/python_env/idf5.3_py3.10_env/bin/python" -m pip install --upgrade --constraint "/home/esp/.vscode-server/extensions/espressif.esp-idf-extension-1.6.5/espidf.constraints.txt" --no-warn-script-location  -r "/home/esp/.vscode-server/extensions/espressif.esp-idf-extension-1.6.5/requirements.txt"
ERROR: Could not install packages due to an OSError: [Errno 13] Permission denied: '/opt/esp/python_env/idf5.3_py3.10_env/lib/python3.10/site-packages/websocket'
Check the permissions.
```

Should this command execute normally inside the container? The only thing I changed from the generated ```Dockerfile``` is adding the created user to the root group.

brianignacio5, 2024-01-08:
Normally this command runs in the Dockerfile to install extension debug adapter dependencies. Seems like a write permission is necessary for the current user.

ag00se, 2024-01-08:
I adjusted the permissions and I could debug successfully. 
@brianignacio5 Are you running Windows and can confirm that the generated ```Dockerfile``` for the devcontainer works on your setup? 
I think that this seems to be a general problem with the container and should not only affect me.

The following changes to the ```Dockerfile``` worked for me:

Add the container user to the root group:
```
usermod -a -G root $CONTAINER_USER
```

Set permissions to install python packages:
```
RUN chmod -R 775 /opt/esp/python_env/
```

brianignacio5, 2024-01-08:
I think there was a use case from some users they didn't want to add the container user as root user. Adding the chmod -R 775 makes sense. 

I also just realize we don't install python packages in Dockerfile directly. We could add it in the Dockerfile but the problem is that extension path changes with extension version. For now I think we need to update docs to let user know to install with command you used before.

brianignacio5, 2024-01-23:
I couldn't also find a workaround this so I just update the docker file as you did. PTAL and let me know what you think

ag00se, 2024-01-23:
I reviewed the two commits concerning this ticket: 51abe5971583c064440f94c04fedea8a0b51ade3 and 6baf2c8e36c788333c65d2774bc392f6c7ac5219
The changes in these commits will solve the problem documented in this issue. Thanks and regards.

PR #1109 (2024-01-25): update dockerfile devcontainer
## Description

Update Dockerfile and devcontainer.json to latest IDF and python version.

Add Python Lib directory to Path in python env install.

Fixes #1100
Fixes #1101
Fixes #1106

## Type of change

Please delete options that are not relevant.
- Bug fix (non-breaking change which fixes an issue)

## Steps to test this pull request

Provide a list of steps to test changes in this PR and required output

1. Click on "Configure ESP-IDF extension" command
2. Execute setup action
3. Observe results. Python libraries used in setup should come from embed Python.

- [... cut]
closed 2024-01-25 by brianignacio5